zephyrproject-rtos/zephyr #28154: reel_board:running samples/subsys/usb/console/ failed

Zhaoningx, 2020-09-08:
Describe the bug
after running samples/subsys/usb/console/, there is no output at  ttyACM0, but you change the connection from usb debug to usb nRF by manually , there will be output something at ttyACM0.

To Reproduce
Steps to reproduce the behavior:

west build -b reel_board samples/subsys/usb/console/
west flash
see error

before change USB port(connecting to USB debug)
*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe ***

after change USB port(connecting to USB nRF)
*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe ***
Hello World! arm                                                               
Hello World! arm                                                               
Hello World! arm
Hello World! arm

Is there OK?Or should the case run at daily test?

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g8c3efc7be5fe

jfischer-phytec-iot, 2020-09-08:
> Is there OK?

Yes, it is correct behavior. The sample `samples/subsys/usb/console` route the shell over Zephyr's CDC ACM implementation, so one have to open device assigned to it, for example `/dev/ttyACM1`

Zhaoningx, 2020-09-09:
Hello @jfischer-phytec-iot , Although it is correct behavior, but you know, the daily testing is just testing automatically, it's reported a failure to me everyday, could you provide a solution to solve this failure?

nashif, 2020-09-11:
lets fix the yaml file for that instead

PR #28329 (2020-09-14): tests: usb: add fixture requirement
The test for this sample requires a usb connection to a secondary port,
output is not captured on the default usb port.

Fixes #28154
commit 4fa08dec (2020-09-14): tests: usb: add fixture requirement

The test for this sample requires a usb connection to a secondary port,
output is not captured on the default usb port.

Fixes #28154
closed 2020-09-14 by MaureenHelm