zephyrproject-rtos/zephyr #28155: sam_e70_xplained:running samples/net/sockets/civetweb/ failed

Zhaoningx, 2020-09-08:
Describe the bug
running samples/net/sockets/civetweb/, and the result just keep waiting interface until time out.

To Reproduce
Steps to reproduce the behavior:
west build -b sam_e70_xplained samples/net/sockets/civetweb/
west flash
see error

*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe  ***
Hello World! sam_e70_xplained
[00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00
[00:00:00.000,000] <inf> eth_sam: MAC: 00:00:00:00:00:00
[00:00:00.000,000] <inf> eth_sam: Queue 0 activated
[00:00:00.000,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.000,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.000,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
*** Booting Zephyr OS build v2.4.0-rc1-11-g8c3efc7be5fe  ***
[00:00:00.090,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
[00:00:00.095,000] <inf> net_config: Initializing network
[00:00:00.095,000] <inf> net_config: Waiting interface 1 (0x20420a70) to be up...
[IMPLEMENTATION MISSING : sscanf]
[00:01:33.098,000] <err> net_config: Timeout while waiting network interface
[00:01:33.098,000] <err> net_config: Network initialization failed (-62)

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g8c3efc7be5fe

jukkar, 2020-09-08:
I just flashed some other sample to this board and noticed that the MAC address was all zeros. This caused issues when trying to get IP address. The error I saw was not `Network initialization failed` so the issue described here might be something else.
Anyway, I needed to set `CONFIG_ETH_SAM_GMAC_MAC_I2C_EEPROM=y` in order the network interface work properly.

Zhaoningx, 2020-09-18:
Hi ,how's it going and what to do next step?

closed 2020-12-02 by github-actions[bot]