zephyrproject-rtos/zephyr #28156: twr_kv58f220m: libraries.cmsis_dsp.transform.cf64 test fails

hakehuang, 2020-09-08:
**Describe the bug**
cmsis_dsp transform function error in twr_kv58f220m

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_kv58f220m ..
3. make
4. See error

**Expected behavior**
test pass

**Impact**
cmsis_dsp function

**Logs and console output**

```
======== Target.main.COM98 Output:   ========
*** Booting Zephyr OS version 2.4.0-rc1  ***

Running test suite transform_cf64

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_16

 PASS - test_arm_cfft_f64_cmplx_noisy_16

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_16

 PASS - test_arm_cifft_f64_cmplx_noisy_16

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_32

 PASS - test_arm_cfft_f64_cmplx_noisy_32

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_32

 PASS - test_arm_cifft_f64_cmplx_noisy_32

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_64

 PASS - test_arm_cfft_f64_cmplx_noisy_64

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_64

 PASS - test_arm_cifft_f64_cmplx_noisy_64

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_128

 PASS - test_arm_cfft_f64_cmplx_noisy_128

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_128

 PASS - test_arm_cifft_f64_cmplx_noisy_128

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_256

 PASS - test_arm_cfft_f64_cmplx_noisy_256

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_256

 PASS - test_arm_cifft_f64_cmplx_noisy_256

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_512

 PASS - test_arm_cfft_f64_cmplx_noisy_512

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_512

 PASS - test_arm_cifft_f64_cmplx_noisy_512

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_1024

 PASS - test_arm_cfft_f64_cmplx_noisy_1024

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_1024

 PASS - test_arm_cifft_f64_cmplx_noisy_1024

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_2048

 PASS - test_arm_cfft_f64_cmplx_noisy_2048

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_2048

 PASS - test_arm_cifft_f64_cmplx_noisy_2048

===================================================================

START - test_arm_cfft_f64_cmplx_noisy_4096

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/cf64.c:27: test_arm_cfft_f64_cmplx_backend: (output is NULL)

buffer allocation failed

 FAIL - test_arm_cfft_f64_cmplx_noisy_4096

===================================================================

START - test_arm_cifft_f64_cmplx_noisy_4096

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/cf64.c:27: test_arm_cfft_f64_cmplx_backend: (output is NULL)

buffer allocation failed

 FAIL - test_arm_cifft_f64_cmplx_noisy_4096

===================================================================

START - test_arm_cfft_f64_cmplx_step_16

 PASS - test_arm_cfft_f64_cmplx_step_16

===================================================================

START - test_arm_cifft_f64_cmplx_step_16

 PASS - test_arm_cifft_f64_cmplx_step_16

===================================================================

START - test_arm_cfft_f64_cmplx_step_32

 PASS - test_arm_cfft_f64_cmplx_step_32

===================================================================

START - test_arm_cifft_f64_cmplx_step_32

 PASS - test_arm_cifft_f64_cmplx_step_32

===================================================================

START - test_arm_cfft_f64_cmplx_step_64

 PASS - test_arm_cfft_f64_cmplx_step_64

===================================================================

START - test_arm_cifft_f64_cmplx_step_64

 PASS - test_arm_cifft_f64_cmplx_step_64

=======================================================================================

START - test_arm_cifft_f64_cmplx_step_4096

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/cf64.c:27: test_arm_cfft_f64_cmplx_backend: (output is NULL)

buffer allocation failed

 FAIL - test_arm_cifft_f64_cmplx_step_4096

===================================================================

Test suite transform_cf64 failed.

===================================================================

PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc1

hakehuang, 2020-09-08:
@MaureenHelm @dleach02

dleach02, 2020-09-08:
@hakehuang, This test is not supported on this platform. Please note that there was a recent change to the testcase.yaml to change whitelist to integration_platforms (which is confusing... not sure why something like "supported_platforms" was used). I believe you have a script that may be processing the testcase.yaml so make sure it is using this new keyword for whitelist.

hakehuang, 2020-09-09:
> @hakehuang, ~This test is not supported on this platform. Please note that there was a recent change to the testcase.yaml to change whitelist to integration_platforms (which is confusing... not sure why something like "supported_platforms" was used). I believe you have a script that may be processing the testcase.yaml so make sure it is using this new keyword for whitelist.~
> 
> Discussing this in the release prep meeting it was pointed out to me that I'm confused about the use of the whitelist/integration_platforms keyword. It is not meant as a "supported" board list but actually to limit testing to a smaller set of boards for time reasons (finish testing sooner).

@dleach02 , the recommand platform is FRDM_K64F which is Cortex-m4 platform, and I enable the cmsis-dsp cases for M4/M0+/M7/CM33 platform, as the cmsis-dap is a pure arm core function, it supposed to be supported on all NXP arm platform.

dleach02, 2020-09-15:
fixed by #27850

closed 2020-09-15 by dleach02