zephyrproject-rtos/zephyr #28157: benchmark.data_structures fails(bus error) on mimxrt1020/60/64 platform

hakehuang, 2020-09-08:
**Describe the bug**
meed bus error when do rb testing

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1020_evk ..
3. make
4. See error

**Expected behavior**
test pass

**Impact**
rbtree performance

**Logs and console output**

```
start stack at 0x800019c0 !!!
*** Booting Zephyr OS version 2.4.0-rc1  ***

Running test suite rbtree

===================================================================

START - test_rbtree_container

E: ***** BUS FAULT *****

E:   Imprecise data bus error

E: ***** HARD FAULT *****

E:   Fault escalation (see below)

E: ***** BUS FAULT *****

E:   Precise data bus error

E:   BFAR Address: 0xbd3f92c0

E: r0/a1:  0x80002320  r1/a2:  0xbd3f92c0  r2/a3:  0x00000020

E: r3/a4:  0x80002320 r12/ip:  0xb060f0f8 r14/lr:  0x6000385f

E:  xpsr:  0x81000205

E: Faulting instruction address (r15/pc): 0x6000631a

E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0

E: Fault during interrupt handling

E: Current thread: 0x800000e8 (unknown)

E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain ( Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc1

hakehuang, 2020-09-08:
@MaureenHelm @dleach02

MaureenHelm, 2020-09-16:
Fixed by #28140

closed 2020-09-16 by MaureenHelm