zephyrproject-rtos/zephyr #28159: [Coverity CID :214216] Logically dead code in drivers/pwm/pwm_stm32.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/drivers/pwm/pwm_stm32.c#L179

Category: Control flow issues
Function: `get_tim_clk`
Component: Drivers
CID: [214216](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214216)

Details:

```
173          * Otherwise, they are set to twice (2) the frequency of the
174          * APB domain.
175          */
176         if (apb_psc == 1u) {
177             *tim_clk = bus_clk;
178         } else {
>>>     CID 214216:  Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "*tim_clk = bus_clk * 2U;".
179             *tim_clk = bus_clk * 2u;
180         }
181     #endif
182    
183         return 0;
184     }
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

gmarull, 2020-09-08:
PWM `get_tim_clk` needs to be improved to cover all STM32 cases. Some progress can be found here https://github.com/zephyrproject-rtos/zephyr/pull/26221/files#diff-0321881e45a0440f0ae270f34324a9c6R129. I'll take care after 2.4 release unless it is considered high priority.

gmarull, 2020-10-19:
After looking into more detail, I think the warning is a false positive. `apb_psc` can get an arbitrary value (`CONFIG_CLOCK_STM32_APB1_PRESCALER` or `CONFIG_CLOCK_STM32_APB2_PRESCALER`), so `else` may be executed. It would only be dead code if the Kconfig option used is set to 1.

gmarull, 2020-10-19:
Closing as it seems to be a false positive, details reported in Coverity.

closed 2020-10-19 by gmarull