zephyrproject-rtos/zephyr #28165: [Coverity CID :214215] Out-of-bounds access in subsys/mgmt/mcumgr/smp_shell.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/subsys/mgmt/mcumgr/smp_shell.c#L115

Category: Memory - corruptions
Function: `smp_shell_rx_bytes`
Component: Other
CID: [214215](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214215)

Details:

```
110                 byte == '\n') {
111                 data->mcumgr_buff[data->cur] = '\0';
112                 data->cmd_rdy = true;
113                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_1);
114                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_2);
115                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_1);
>>>     CID 214215:    (ARRAY_VS_SINGLETON)
>>>     Passing "&data->esc_state" to function "atomic_clear_bit" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
116                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_2);
117                 data->cur = 0U;
118             }
119    
120             ++consumed;
121         }
107    
108             /* Newline in payload means complete frame */
109             if (mcumgr_state == SMP_SHELL_MCUMGR_STATE_PAYLOAD &&
110                 byte == '\n') {
111                 data->mcumgr_buff[data->cur] = '\0';
112                 data->cmd_rdy = true;
>>>     CID 214215:    (ARRAY_VS_SINGLETON)
>>>     Passing "&data->esc_state" to function "atomic_clear_bit" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
113                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_1);
114                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_2);
115                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_1);
116                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_2);
117                 data->cur = 0U;
118             }
109             if (mcumgr_state == SMP_SHELL_MCUMGR_STATE_PAYLOAD &&
110                 byte == '\n') {
111                 data->mcumgr_buff[data->cur] = '\0';
112                 data->cmd_rdy = true;
113                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_1);
114                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_PKT_2);
>>>     CID 214215:    (ARRAY_VS_SINGLETON)
>>>     Passing "&data->esc_state" to function "atomic_clear_bit" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
115                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_1);
116                 atomic_clear_bit(&data->esc_state, ESC_MCUMGR_FRAG_2);
117                 data->cur = 0U;
118             }
119    
120             ++consumed;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nvlsianpu, 2020-09-09:
@carlescufi False negative.
Issue appeared as mcumgr was moved to deeper directory.

nvlsianpu, 2020-09-14:
False positive.

closed 2020-09-14 by nvlsianpu