zephyrproject-rtos/zephyr #28167: [Coverity CID :214209] Dereference after null check in subsys/mgmt/osdp/src/osdp.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/subsys/mgmt/osdp/src/osdp.c#L186

Category: Null pointer dereferences
Function: `osdp_init`
Component: Other
CID: [214209](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214209)

Details:

```
180         /* configure uart device to 8N1 */
181         p->dev_config.baudrate = CONFIG_OSDP_UART_BAUD_RATE;
182         p->dev_config.data_bits = UART_CFG_DATA_BITS_8;
183         p->dev_config.parity = UART_CFG_PARITY_NONE;
184         p->dev_config.stop_bits = UART_CFG_STOP_BITS_1;
185         p->dev_config.flow_ctrl = UART_CFG_FLOW_CTRL_NONE;
>>>     CID 214209:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "p->dev" to "uart_configure", which dereferences it.
186         uart_configure(p->dev, &p->dev_config);
187    
188         uart_irq_rx_disable(p->dev);
189         uart_irq_tx_disable(p->dev);
190         uart_irq_callback_user_data_set(p->dev, osdp_uart_isr, p);
191    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2020-09-10:
@cbsiddharth will you please have a look?

cbsiddharth, 2020-09-10:
@MaureenHelm, Sure.

The coverity report link says `401: Unauthorized` can you help?

MaureenHelm, 2020-09-10:
> The coverity report link says `401: Unauthorized` can you help?

I sent a coverity invite to the email you use in your signed-off-by

cbsiddharth, 2020-09-11:
`p->dev` is checked for NULL before and `k_panic()` is called. A silent return here without panic would cause the sample app to crash as it makes the assumption that OSDP is initialized correctly.

Another approach is to expose an API `osdp_is_ready()` and `k_panic()` in sample. To me both looks similar, can anyone help?

jukkar, 2020-09-14:
To me this looks like a false positive. Coverity does not understand that processing ends when `k_panic()` is called.

cbsiddharth, 2020-09-14:
@MaureenHelm, I'd like to add the following comment and mark the issue as "Ignore" in Coverity:

```
p->dev is checked for NULL before and k_panic() is called. k_panic() does not return.
```

I'm not able to submit this comment in the coverity URL (which looks like a permission issue).

jukkar, 2020-09-14:
I can add this to Coverity and close this one.

closed 2020-09-14 by jukkar