zephyrproject-rtos/zephyr #28168: [Coverity CID :214218] Unused value in subsys/mgmt/osdp/src/osdp.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/subsys/mgmt/osdp/src/osdp.c#L62

Category: Code maintainability issues
Function: `osdp_uart_isr`
Component: Other
CID: [214218](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214218)

Details:

```
56    
57             if (uart_irq_tx_ready(dev)) {
58                 len = ring_buf_get(&p->tx_buf, buf, 1);
59                 if (!len) {
60                     uart_irq_tx_disable(dev);
61                 } else {
>>>     CID 214218:  Code maintainability issues  (UNUSED_VALUE)
>>>     Assigning value from "uart_fifo_fill(dev, buf, 1)" to "wrote" here, but that stored value is overwritten before it can be used.
62                     wrote = uart_fifo_fill(dev, buf, 1);
63                 }
64             }
65         }
66     #ifdef CONFIG_OSDP_MODE_PD
67         /* wake osdp_refresh thread */
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2020-09-10:
@cbsiddharth will you please have a look?

cbsiddharth, 2020-09-11:
A commit in PR #28200 implicitly fixes this issue. Will amend that commit message to include this `Coverity-ID`.

commit 23c0c770 (2020-10-21): mgmt/osdp: Skip all incoming bytes till start of new packet

All OSDP packets must start with a mark byte (0xFF) followed by a Start
of Message byte (0x53). Skip all bytes received until such a sequence is
seen to to avoid waking up the refresh thread needlessly.

Also, refactor osdp_uart_isr() to reduce number of local variables.

Fixes: #28168 Coverity-ID: 214218
closed 2020-10-21 by carlescufi