zephyrproject-rtos/zephyr #28169: [Coverity CID :214220] Explicit null dereferenced in subsys/mgmt/hawkbit/hawkbit.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/subsys/mgmt/hawkbit/hawkbit.c#L802

Category: Null pointer dereferences
Function: `response_cb`
Component: Other
CID: [214220](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214220)

Details:

```
796                 if (body_data == NULL) {
797                     body_data = rsp->recv_buf;
798                     body_len = rsp->data_len;
799                 }
800             }
801    
>>>     CID 214220:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "body_data" to "flash_img_buffered_write", which dereferences it.
802             ret = flash_img_buffered_write(&hb_context.flash_ctx, body_data,
803                                body_len,
804                                final_data == HTTP_DATA_FINAL);
805             if (ret < 0) {
806                 LOG_ERR("flash write error");
807                 hb_context.code_status = HAWKBIT_DOWNLOAD_ERROR;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-09-08:
@Navin-Sankar can you check this one, for some reason I could not assign this to you.

MaureenHelm, 2020-09-15:
> @Navin-Sankar can you check this one, for some reason I could not assign this to you.

@Navin-Sankar ping

PR #28425 (2020-09-16): mgmt: hawkbit: Fix NULL pointer dereferences
Coverity spots that null pointer is passing to flash_img_buffered_write.
This patch fixes the problem.
Fixes #28169
closed 2020-09-16 by MaureenHelm