zephyrproject-rtos/zephyr #28170: [Coverity CID :214222] Unrecoverable parse warning in include/ec_host_cmd.h

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/include/ec_host_cmd.h#L195

Category: Parse warnings
Function: ``
Component: Other
CID: [214222](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214222)

Details:

```
189         EC_HOST_CMD_BUS_ERROR = 15,
190         /** System busy. Should retry later. */
191         EC_HOST_CMD_BUSY = 16,
192    
193         EC_HOST_CMD_MAX = UINT16_MAX /* Force enum to be 16 bits */
194     } __packed;
>>>     CID 214222:  Parse warnings  (PARSE_ERROR)
>>>     static assertion failed with ""
195     BUILD_ASSERT(sizeof(enum ec_host_cmd_status) == sizeof(uint16_t));
196    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jettr, 2020-09-08:
I'm setting up access to coverity now, then will send out a fix. Thanks for redirecting this my way!

jettr, 2020-09-10:
I finally have a coverity account (it took over a day for it to work for some reason). When I click on the link to get more information, it doesn't work. Do I need to be added to the coverity Zephyr project to see details?

Also, I cannot immediately see what the problem is with the line in question. It looks okay to me, and there are other examples similar to it in the code base. Maybe the issue is taking sizeof an enum?

When I build locally, I do not see that this assertion fails either.

nashif, 2020-09-11:
@jettr this could be a build issue during coverity run and thus a false positive. Not sure why it fails, works for me locally. Will keep looking.

jettr, 2020-09-15:
Can we conclude this was a false positive?

ceolin, 2020-09-15:
Nope, that is not a false positive IMHO. C standard does not define the enum size, it says that it has to be compatible with char and a signed/unsigned integer type an be capable of represent the values of all members, but this is implementation defined. So, it could be an unsigned 32 integer, that is the case without the packed attribute. For the portability sake we should not have that assert.

jettr, 2020-09-16:
SG. I will just remove the `BUILD_ASSERT`

PR #28648 (2020-10-02): include: remove enum build assert
Remove BUILD_ASSERT that is causing coverity error. This fixes #28170. The enum is always serialized at 16 bits, and it is documented in the comments.
closed 2020-10-02 by carlescufi