zephyrproject-rtos/zephyr #28172: [Coverity CID :214227] Resource leak in subsys/mgmt/hawkbit/hawkbit.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/subsys/mgmt/hawkbit/hawkbit.c#L285

Category: Resource leaks
Function: `start_http_client`
Component: Other
CID: [214227](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214227)

Details:

```
276             return false;
277         }
278     #endif
279    
280         if (connect(hb_context.sock, addr->ai_addr, addr->ai_addrlen) < 0) {
281             LOG_ERR("Failed to connect to Server");
>>>     CID 214227:    (RESOURCE_LEAK)
>>>     Variable "addr" going out of scope leaks the storage it points to.
282             return false;
283         }
284    
285         return true;
286     }
287    
253             return false;
254         }
255    
256         hb_context.sock = socket(addr->ai_family, SOCK_STREAM, protocol);
257         if (hb_context.sock < 0) {
258             LOG_ERR("Failed to create TCP socket");
>>>     CID 214227:    (RESOURCE_LEAK)
>>>     Variable "addr" going out of scope leaks the storage it points to.
259             return false;
260         }
261    
262     #if defined(CONFIG_NET_SOCKETS_SOCKOPT_TLS)
263         sec_tag_t sec_tag_opt[] = {
264             CA_CERTIFICATE_TAG,
279    
280         if (connect(hb_context.sock, addr->ai_addr, addr->ai_addrlen) < 0) {
281             LOG_ERR("Failed to connect to Server");
282             return false;
283         }
284    
>>>     CID 214227:    (RESOURCE_LEAK)
>>>     Variable "addr" going out of scope leaks the storage it points to.
285         return true;
286     }
287    
288     static void cleanup_connection(void)
289     {
290         if (close(hb_context.sock) < 0) {
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-09-08:
@Navin-Sankar can you check this one, for some reason I could not assign this to you.

MaureenHelm, 2020-09-15:
@Navin-Sankar ping

PR #28418 (2020-09-16): mgmt: hawkbit: Fix resources leakage
Coverity spots that memory allocated for addr was leaking, but the
code was also leaking the socket file descriptor in the error path.
This patch fixes both problems.

Fixes #28172
commit e28d3404 (2020-09-16): mgmt: hawkbit: Fix resources leakage

Coverity spots that memory allocated for addr was leaking, but the
code was also leaking the socket file descriptor in the error path.
This patch fixes both problems.

Fixes #28172
closed 2020-09-16 by carlescufi