zephyrproject-rtos/zephyr #28175: [Coverity CID :214214] Uninitialized pointer read in tests/benchmarks/data_structure_perf/rbtree_perf/src/rbtree_perf.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/tests/benchmarks/data_structure_perf/rbtree_perf/src/rbtree_perf.c#L58

Category: Memory - illegal accesses
Function: `test_rbtree_container`
Component: Tests
CID: [214214](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214214)

Details:

```
52         struct rbnode *foreach_node;
53         struct container_node tree_node[10];
54    
55         test_tree_l.lessthan_fn = node_lessthan;
56         for (uint32_t i = 0; i < ARRAY_SIZE(tree_node); i++) {
57             tree_node[i].value = i;
>>>     CID 214214:  Memory - illegal accesses  (UNINIT)
>>>     Using uninitialized value "test_tree_l.max_depth" when calling "rb_insert".
58             rb_insert(&test_tree_l, &tree_node[i].node);
59         }
60    
61         RB_FOR_EACH(&test_tree_l, foreach_node) {
62             zassert_true(CONTAINER_OF(foreach_node, struct container_node,
63             node)->value == count, "RB_FOR_EACH failed");
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

Zhaoningx, 2020-09-09:
I have added the solution to solve this issue.

Zhaoningx, 2020-09-09:
I have added the solution to solve the bug, and I'll close this issue if there is no longer anything wrong.

PR #28212 (2020-10-21): tests: initializing rbtree and its node
add  initializing rbtree and its node to fix the issue https://github.com/zephyrproject-rtos/zephyr/issues/28175

Fixes #28175
closed 2020-10-21 by nashif