zephyrproject-rtos/zephyr #28176: [Coverity CID :214217] Out-of-bounds access in tests/kernel/mem_protect/mem_map/src/main.c

zephyrbot, 2020-09-08:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/27e1fd69f987d3082229b3368a87d59ffdcab5a4/tests/kernel/mem_protect/mem_map/src/main.c#L103

Category: Memory - corruptions
Function: `test_z_mem_map_exec`
Component: Tests
CID: [214217](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214217)

Details:

```
97         expect_fault = false;
98    
99         /* Map with write permissions and copy the function into the page */
100         z_mem_map(&mapped_rw, (uintptr_t)test_page,
101               sizeof(test_page), BASE_FLAGS | K_MEM_PERM_RW);
102    
>>>     CID 214217:  Memory - corruptions  (OVERRUN)
>>>     Overrunning buffer pointed to by "&transplanted_function" of 0 bytes by passing it to a function which accesses it at byte offset 4095 using argument "4096U". [Note: The source code implementation of the function has been overridden by a builtin model.]
103         memcpy(mapped_rw, &transplanted_function, CONFIG_MMU_PAGE_SIZE);
104    
105         /* Now map with execution enabled and try to run the copied fn */
106         z_mem_map(&mapped_exec, (uintptr_t)test_page,
107               sizeof(test_page), BASE_FLAGS | K_MEM_PERM_EXEC);
108    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dcpleung, 2020-10-08:
This is intentional, and marked as such in Coverity. So closing this...

closed 2020-10-08 by dcpleung

aasthagr, 2020-10-08:
Thanku so much @dcpleung