zephyrproject-rtos/zephyr #28181: MQTT not working with MOSQUITTO broker:

mtahirbutt, 2020-09-08:
**Describe the bug**
A clear and concise description of what the bug is.
MQTT is not able to connect to test.mosquitto.org (IP address: 5.196.95.208).
What have you tried to diagnose or workaround this issue?
I tried mqtt.eclipse.org (IP address: 137.135.83.217), that worked and I get publish messages etc. and the client is successfully connected to the broker, but test.mosquitto.org does not seem to work.
I checked MQTT explorer, from there I get connection to even test.mosquitto.org as well as mqtt.eclipse.org but zephyr client is problematic with test.mosquitto.org as I use the IP address 5.196.95.208 in the configuration file.
It should be noted I am using SARA U201 modem shield with disco_l475_iot1.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=disco_l475_iot1
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I should get stable connection as I get with mqtt.eclipse.org (137.135.83.217).
**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I also want to test TLS client which used to work sometimes ago but I do not know what has changed on my side or on broker side.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:15.608,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:19.114,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:22.621,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:26.127,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:29.634,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:33.140,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:36.647,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:40.153,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
[1B][1;32muart: [1B][m[1B][8D[1B][J[1B][1;32muart: [1B][m[1B][8D[1B][J[00:02:43.660,000] [1B][0m<inf> net_mqtt_publisher_sample: mqtt_connect: -60 <ERROR>[1B][0m
**Environment (please complete the following information):**
 - OS: (e.g. Windows)
 - Toolchain (e.g Zephyr SDK 0.11.3, 2.4.0-rc1)
 - Commit SHA or Version used latest

**Additional context**
Add any other context about the problem here.
test.mosquitto.org worked perfectly fine (even with TLS) sometimes with 2.3.0 with one of my custom board, but right now I do not know what has changed either on the broker side or something else.
Besides, I get the IP address of test.mosquitto.org using ping. I do not know what is the usual way of getting correct IP address of test.mosquitto.org

rlubos, 2020-09-09:
I've reproduced the issue with `test.mosquitto.org` (but not only on current master, 2.2.0 or 2.3.0 failed for me as well).

The problem turned out to be the long response time for the CONNECT message from `test.mosquitto.org`. By default we timeout the connection after 500 ms, but for `test.mosquitto.org` I've observed that it took up to 2 seconds to get a response. I think that a valid solution would be to increase the connect timeout in the sample, it should be no harm in general and will improve connectivity to slower brokers, like the aformentioned one. I will submit a fix.

> Besides, I get the IP address of test.mosquitto.org using ping. I do not know what is the usual way of getting correct IP address of test.mosquitto.org

The sample was originally intended to test with a broker running in a local network, hence it uses IP addresses directly. For public servers, it's reasonable to use DNS to obtain the IP address of the server (`getaddrinfo()`). See for instance how `http_get` sample implements this:
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/sockets/http_get/src/http_get.c#L74

mtahirbutt, 2020-09-09:
@rlubos thanks for the fix, I will check it with my board.

PR #28222 (2020-09-09): samples: net: mqtt: Increase connect timeout
The default timeout used (500 ms) turns out to be too small for certain servers (test.mosquitto.org), rendering it impossible to connect (the TCP connection was closed before the actual Connect ACK response arrived). Increase the connect timeout to cover this case.

Additionally, remove some unused defines that were not used anywhere in the sample or tests.

Fixes #28181
commit a6eb5569 (2020-09-09): samples: net: mqtt: Increase connect timeout

The default timeout used (500 ms) turns out to be too small for certain
servers (test.mosquitto.org), rendering it impossible to connect (the
TCP connection was closed before the actual Connect ACK response
arrived). Increase the connect timeout to cover this case.

Fixes #28181
closed 2020-09-09 by jukkar