zephyrproject-rtos/zephyr #28184: tests: drivers: spi: spi_loopback: fails on board nucleo_wb55rg

ABOSTM, 2020-09-08:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. west -p auto -b nucleo_wb55rg tests/drivers/spi/spi_loopback  nucleo_wb55rg
2. west flash
3. See error

**Expected behavior**
test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc1-10-g4d3580d29735  ***
Running test suite test_spi
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x20000000/0x200004f0
I: SPI test slow config
I: Start complete multiple
I: CS control inhibited (no GPIO device)
E: DMA callback channel 8 is not valid.
E: Code -5

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:140: spi_complete_multiple: (ret is true)
SPI transceive failed
 FAIL - test_spi_loopback
===================================================================
Test suite test_spi failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux and Windows
 - Toolchain : Zephyr SD
 - Commit SHA:4d3580d29735bd3d3d7a89e7475ccd9e001cd818

lowlander, 2020-09-08:
@ABOSTM sadly I don't have stm32wb hardware to test, but is this a new failure (after I reworked the DMA driver) or did it fail before?  The 8 not valid seems to be a off by one error (the channel, stream, id numbering is very confusing for the v1, v2, and dmamux) 

I could try with L4R5 hardware that should have the same DMAMUX setup.

ABOSTM, 2020-09-09:
@lowlander , I pushed a PR which solves the issue: #28182.

PR #28182 (2020-09-11): drivers: dma: stm32: fixes spi_loopback on nucleo_wb55rg
2 commits:
* drivers: dma: dma_stm32: can't use device_get_binding() during init
* drivers: dma: dma_stm32_v2: typo in compilation switch

Fixes #28184
closed 2020-09-11 by MaureenHelm