zephyrproject-rtos/zephyr #28185: Problem using SX1276 with nRF52840dk

GABS9718, 2020-09-08:
**Describe the bug**
Hi, I'm trying to use the radio SX1276 in the module RFM95 with the development kit BMD-340 that have the nRF52840. But my issue is before the hardware, as I'm not able to build the project. I'm just trying to build the code that is inside <zephyrproject\zephyr\samples\drivers\lora\send>. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open the cmd
2. go to directory <zephyrproject\zephyr>
3. digit: **west build -p auto -b nrf52840dk_nrf52840 samples\drivers\lora\send**

**Expected behavior**
I was hoping to be able to build the project.

**Impact**
I received many warnings related with macros that begin with DT, what I imagine that is devicetree. More than that, I received error related with some variables that are not declared. 

**Logs and console output**
I have attached one .txt with the log that I got.
[log.txt](https://github.com/zephyrproject-rtos/zephyr/files/5188425/log.txt)
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU ARM Embedded
 - The newest version avaiable on the repository

**Additional context**
I tried to change the variable in line 3, in the file main.c from zephyrproject\zephyr\samples\drivers\lora\send, from 
#define DEFAULT_RADIO_NODE DT_ALIAS(lora0) to #define DEFAULT_RADIO_NODE DT_ALIAS(LORA_SX1276) but nothing changed. I'm new in the context of Zephyr and I saw that the support for the module SX1276 is already done. So, probably I'm doing something wrong in the seetings. Someone can help me, please?

carlescufi, 2020-09-08:
>west build -p auto -b nrf52840dk_nrf52840 samples\drivers\lora\send

Did you add a LoRa radio to the board's .dts file? You need to do that in order for the driver to work. This can be used as a source:

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/96b_wistrio/96b_wistrio.dts#L65

You can also use .overlay files to avoid modifying the board.

Everything is documented here: https://docs.zephyrproject.org/latest/guides/dts/index.html

carlescufi, 2020-09-08:
@GABS9718 I closed this since it seems you have not declared the LoRa device in Devicetree. If that is incorrect please reopen it.

closed 2020-09-08 by carlescufi

GABS9718, 2020-09-09:
@carlescufi Hi! The problem related to the building process was really related with not declaring in the .dts of the board the lora device. Thank you.