zephyrproject-rtos/zephyr #28196: samples/boards/intel_s1000_crb/audio/sample.board.intel_s1000_crb.audio Fails to build

galak, 2020-09-08:
Doing:

`./scripts/sanitycheck -p intel_s1000_crb -s samples/boards/intel_s1000_crb/audio/sample.board.intel_s1000_crb.audio`

Gets:

```
[ 10%] Building C object CMakeFiles/app.dir/src/usb_transport.c.obj
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:34:16: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   34 |  .get_report = usb_transport_get_report,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:34:16: note: (near initialization for 'usb_transport_callbacks.get_report')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:35:14: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   35 |  .get_idle = usb_transport_get_idle,
      |              ^~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:35:14: note: (near initialization for 'usb_transport_callbacks.get_idle')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:36:18: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   36 |  .get_protocol = usb_transport_get_protocol,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:36:18: note: (near initialization for 'usb_transport_callbacks.get_protocol')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:37:16: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   37 |  .set_report = usb_transport_set_report,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:37:16: note: (near initialization for 'usb_transport_callbacks.set_report')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:38:14: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   38 |  .set_idle = usb_transport_set_idle,
      |              ^~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:38:14: note: (near initialization for 'usb_transport_callbacks.set_idle')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:39:18: error: initialization of 'int (*)(const struct device *, struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(const struct device *, struct usb_setup_packet *, int *, unsigned char **)'} from incompatible pointer type 'int (*)(struct usb_setup_packet *, int32_t *, uint8_t **)' {aka 'int (*)(struct usb_setup_packet *, int *, unsigned char **)'} [-Werror=incompatible-pointer-types]
   39 |  .set_protocol = usb_transport_set_protocol,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:39:18: note: (near initialization for 'usb_transport_callbacks.set_protocol')
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:40:18: error: initialization of 'void (*)(const struct device *)' from incompatible pointer type 'void (*)(void)' [-Werror=incompatible-pointer-types]
   40 |  .int_in_ready = usb_transport_host_ready,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/samples/boards/intel_s1000_crb/audio/src/usb_transport.c:40:18: note: (near initialization for 'usb_transport_callbacks.int_in_ready')
cc1: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/app.dir/build.make:135: CMakeFiles/app.dir/src/usb_transport.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2306: CMakeFiles/app.dir/all] Error 2
gmake: *** [Makefile:104: all] Error 2
```

INFO    - 5/8 intel_s1000_crb           samples/boards/intel_s1000_crb/audio/sample.board.intel_s1000_crb.audio FAILED Build failure (build)

jfischer-phytec-iot, 2020-09-08:
Sample was broken by commit 0d8bd579a5a3ca819641fb840de97e76b5647ed0

emob-nordic, 2020-09-09:
PR posted.

PR #28209 (2020-09-09): usb: intel: Fix the sample, update to recent API change.
HID API has changed and now all usb HID callbacks require
pointer to device as a first parameter. This patch is fixing
regression issue.

Fixes: #28196
closed 2020-09-09 by nashif