zephyrproject-rtos/zephyr #28197: samples/net/sockets/echo_client/sample.net.sockets.echo_client.nrf_openthread fails to build

galak, 2020-09-08:
`./scripts/sanitycheck -p nrf52840dk_nrf52840 -s samples/net/sockets/echo_client/sample.net.sockets.echo_client.nrf_openthread`

```
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - not found
-- Looking for stddef.h
-- Looking for stddef.h - not found
-- Check size of unsigned short
-- Check size of unsigned short - failed
-- Check size of unsigned int
-- Check size of unsigned int - failed
-- Check size of unsigned long
-- Check size of unsigned long - failed
CMake Error at /usr/share/cmake/Modules/TestBigEndian.cmake:50 (message):
  no suitable type found
Call Stack (most recent call first):
  /home/galak/git/modules/lib/openthread/CMakeLists.txt:50 (TEST_BIG_ENDIAN)

-- Configuring incomplete, errors occurred!
See also "/home/galak/git/zephyr/sanity-out/nrf52840dk_nrf52840/samples/net/sockets/echo_client/sample.net.sockets.echo_client.nrf_openthread/CMakeFiles/CMakeOutput.log".
See also "/home/galak/git/zephyr/sanity-out/nrf52840dk_nrf52840/samples/net/sockets/echo_client/sample.net.sockets.echo_client.nrf_openthread/CMakeFiles/CMakeError.log".
```

aurel32, 2020-09-08:
@tejlmand  This issue has been introduced by commit c55c64e242ea7f3533f3245edf604d6fe3d3953f.

tejlmand, 2020-09-10:
Reopened. Wating for: https://github.com/zephyrproject-rtos/zephyr/pull/28273 to be merged.

PR #28273 (2020-09-14): Openthread build flags
Fixes: #28197

Update openthread to build flag changes for fixing #28197.

see https://github.com/zephyrproject-rtos/openthread/pull/29

It also introduces `zephyr_get_targets()` that allows fetching of build target from a directory and down.
This is useful when including external modules where only libraries
required by Zephyr should be built, and remove the need of setting build flags on openthread libraries not used by Zephyr.

This has the positive side effect of reducing the number of build steps and build time significantly.

Before, 780 steps:
`time ninja -Cbuild` [... cut]
commit 0328e893 (2020-09-14): west.yml: update openthread module

Fixes: #28197

Update openthread to build flag changes for fixing #28197.

see zephyrproject-rtos/openthread/pull/29
commit 2d2fea87 (2020-09-14): cmake: create zephyr_get_targets() functions

Fixes: #28197

zephyr_get_targets() provides a mechanism for retrieving all CMake
targets for a given directory and below.

This is useful when including external modules where only libraries
required by Zephyr should be built.

For example, in openthread, this reduces the number of build steps and
build time significantly.

Before, 780 steps:
`time ninja -Cbuild`
[708/708] Linking CXX executable zephyr/zephyr.elf
135.08user 30.24system 0:46.21elapsed 357%CPU

After, 480 steps:
`time ninja -Cbuild`
[480/480] Linking CXX executable zephyr/zephyr.elf [... cut]
closed 2020-09-14 by carlescufi