zephyrproject-rtos/zephyr #28202: Adafruit TFT touch shield cap touch flipped sides left-to-right

shileiphys, 2020-09-08:
**Describe the bug**
I am working with nRF52840, Adafruit 2.8 TFT touch shield with capacitor touch, and I am using the **lvgl** for GUI.
1) If a button were put inside the left half screen, only touching the mirrored position in the right half screen can trigger the touch of the button.
-- I have no workaround for the issue.

2) quite often, the touch ability could be totally lost.
-- I have tried reset, button but not work, only a total power down and restart can bring back the touch, (still flipped)

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare the shield, cut/soldering, to make SPI use D13,12 and 11
2. Modify the code inside samples/display/lvgl/main.c, move **hello_word_button** left (-80)
```
	// if (IS_ENABLED(CONFIG_LVGL_POINTER_KSCAN)) {
		lv_obj_t *hello_world_button;

		hello_world_button = lv_btn_create(lv_scr_act(), NULL);
		lv_obj_align(hello_world_button, NULL, LV_ALIGN_CENTER, -80, 0);
		lv_btn_set_fit(hello_world_button, LV_FIT_TIGHT);
		hello_world_label = lv_label_create(hello_world_button, NULL);
	// } else {
	// 	hello_world_label = lv_label_create(lv_scr_act(), NULL);
	// }

	static lv_style_t style_btn;
	lv_style_init(&style_btn);
	lv_style_set_bg_color(&style_btn, LV_STATE_DEFAULT, LV_COLOR_GRAY);
	lv_obj_add_style(hello_world_button, LV_BTN_PART_MAIN, &style_btn);

	lv_label_set_text(hello_world_label, "Hello!");
```
3. west build -b nrf52840dk_nrf52840 samples/display/lvgl -- -DSHIELD=adafruit_2_8_tft_touch_v2
4. west flash

**Expected behavior**
A touch of the "hello" button should show color change indicating the button is touched.

**Impact**
not able to use the touch shield properly.

**Logs and console output**
```
lei@zephyr$ west build -b nrf52840dk_nrf52840 samples/display/lvgl -- -DSHIELD=adafruit_2_8_tft_touch_v2
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/lei/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/lei/zephyrproject/zephyr/samples/display/lvgl
-- Zephyr version: 2.3.99 (/home/lei/zephyrproject/zephyr)
-- Found west: /home/lei/.local/bin/west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Found toolchain: zephyr (/home/lei/zephyr-sdk-0.11.3)
-- Found dtc: /home/lei/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/lei/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /home/lei/zephyrproject/zephyr/boards/shields/adafruit_2_8_tft_touch_v2/adafruit_2_8_tft_touch_v2.overlay
-- Generated zephyr.dts: /home/lei/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/lei/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/lei/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/lei/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/home/lei/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/home/lei/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Cache files will be written to: /home/lei/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lei/zephyrproject/zephyr/build
-- west build: building application
[122/126] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       98744 B         1 MB      9.42%
            SRAM:      177400 B       256 KB     67.67%
        IDT_LIST:         104 B         2 KB      5.08%
[126/126] Linking C executable zephyr/zephyr.elf
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: zephyr-sdk-0.11.3
 - Zephy: latest

jfischer-phytec-iot, 2020-09-11:
I guess @gmarull has a fix for the mirror issue, see #28247. @shileiphys can you test it, please?

shileiphys, 2020-09-11:
Yes, I have tested, it fixed the problem for me. Thanks!

gmarull, 2020-09-14:
@shileiphys I have pushed a slightly different approach on the same PR (in some way inspired by X11 settings).

shileiphys, 2020-09-14:
Thank you! I'll give it a try

PR #28247 (2020-09-16): lib: gui: lvgl: add support for kscan axes swap and inversion
Add new Kconfig options to allow kscan axes swap and inversion. These options are useful to align display and touch frame of reference.

NOTE: If a touch API is ever introduced these options could be moved there (DT).

Fixes #28202
closed 2020-09-16 by carlescufi