zephyrproject-rtos/zephyr #28205: kernel.timer.tickless: frdmk64f failure

hakehuang, 2020-09-09:
**Describe the bug**    
     FAIL - test_timer_duration_period

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
test_timer_duration_period

**Impact**
timer settings

**Logs and console output**

```
*** Booting Zephyr OS version 2.4.0-rc1  ***
    Running test suite timer_api
    ===================================================================
    START - test_time_conversions
     PASS - test_time_conversions
    ===================================================================
    START - test_timer_duration_period
    
        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:94: duration_expire: (interval >= 100) || (((10000 % 1000U) != 0) && (interval == 100 - 1)) is false
    
     FAIL - test_timer_duration_period
    ===================================================================
    START - test_timer_period_0
     PASS - test_timer_period_0
    ===================================================================
    START - test_timer_expirefn_null
     PASS - test_timer_expirefn_null
    ===================================================================
    START - test_timer_periodicity
     PASS - test_timer_periodicity
    ===================================================================
    START - test_timer_status_get
     PASS - test_timer_status_get
    ===================================================================
    START - test_timer_status_get_anytime
     PASS - test_timer_status_get_anytime
    ===================================================================
    START - test_timer_status_sync
     PASS - test_timer_status_sync
    ===================================================================
    START - test_timer_k_define
     PASS - test_timer_k_define
    ===================================================================
    START - test_timer_user_data
     PASS - test_timer_user_data
    ===================================================================
    START - test_timer_remaining
     PASS - test_timer_remaining
    ===================================================================
    START - test_timeout_abs
     PASS - test_timeout_abs
    ===================================================================
    Test suite timer_api failed.
    ===================================================================
    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc1

hakehuang, 2020-09-09:
@MaureenHelm , @dleach02

dleach02, 2020-09-15:
@hakehuang can you retest this? I'm unable to reproduce this particular issue.

ceolin, 2020-09-16:
tested here and it is working fine as well. Couldn't reproduce the problem.

hakehuang, 2020-09-16:
@dleach02 If I rerun it for several times, it is OK, the is a low changes it gets wrong, looks like the test_timer_duration_period case does not give much margin if the code runs slow for same reason. I will close it.

closed 2020-09-16 by hakehuang