zephyrproject-rtos/zephyr #28206: mimxrt685_cm33: many cases has no console output seems hangs in kernel init.

hakehuang, 2020-09-09:
**Describe the bug**
mimxrt685_cm33: many cases has no console output seems hangs in kernel init.

**To Reproduce**
please check attached report for timeout cases

**Expected behavior**
test can pass

**Impact**
some core setting for CM33 is not correct in kernel

**Logs and console output**
NO console output

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc1

hakehuang, 2020-09-09:
@MaureenHelm , @dleach02 , @mmahadevan108

hakehuang, 2020-09-14:
@mmahadevan108  with your fix in [https://github.com/zephyrproject-rtos/hal_nxp/pull/57], the /tests/subsys/debug/coredump case still fails, there is no coredump happen, only print out the first message. see below log
```
*** Booting Zephyr OS build v2.4.0-rc1  ***
Coredump: mimxrt685_evk_cm33
```

the expected log shall be, take FRDM_k64f as example

```
*** Booting Zephyr OS build v2.4.0-rc1  ***
Coredump: frdm_k64f
E: ***** USAGE FAULT *****
E:   Attempt to execute undefined instruction
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000080
E: r3/a4:  0x00000000 r12/ip:  0x0000000f r14/lr:  0x00000713
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00000716
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200001c0 (unknown)
E: #CD:BEGIN#
E: #CD:5a4501000300050000000000
E: #CD:4101002400
E: #CD:000000000000000080000000000000000f000000130700001607000000000061
E: #CD:18070020
E: #CD:4d01000000002090100020
E: #CD:ed0a0000ed0a0000000000209010002000000000000000003118000025180000
E: #CD:19180000a117000000c201000001030000c201000001030000093d0000800000
E: #CD:4000002040000020b43700000c360000e435000000000000ba37000004360000
E: #CD:e435000000000000c0370000fc350000e435000000000000c6370000f4350000
E: #CD:e4350000000000008c37000000000000bc35000000000000a237000000000000
E: #CD:d035000000000000883d000094360000a0360000300000206937000088360000
E: #CD:a036000028000020cc370000000000000000000000000000653d000078360000
E: #CD:14360000940200206c3d0000683600001436000088020020733d000058360000
E: #CD:143600007c0200207a3d0000483600001436000070020020813d000038360000
E: #CD:1436000064020020ff3f00002c0100202c01002000000000000000000f000000
E: #CD:00000000000000000000000001000f0000000000000000000000000000000000
E: #CD:0000000000000000000000000000000070010020700100200000000000000000
E: #CD:0000000000000000000000000000000000000000000000007008002000000000
E: #CD:0000000000000000500700204001000000000000000000000000000000000000
E: #CD:cc020020cc020020000000000180000000000000000000000000000000000000
E: #CD:00000000000000000000000000000000f0010020f00100202920000050070020
E: #CD:c001002040010020500700209010002029200000010000002007002000000000
E: #CD:00000000000000005003002000040000000000000000000000000000f5ffffff
E: #CD:0000000000000000b800002000000000cd30000060ccff000000000018070020
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:80f0fa02000000000000000090100020c0010020400100200000000000000000
E: #CD:c0020020c0020020c0010020c0010020c0010020000000000000000000000000
E: #CD:00000000000000000000000080000000000000000f0000001307000016070000
E: #CD:0000006118070020053030303030303631313830373030323030353330333033
E: #CD:3033303330333033363331333133383330333733303330333333333333333333
E: #CD:33333333333333333300010006b0bde8f08ddae90101b0fbf1f7604279420840
E: #CD:4542a54200d13d44049e22e0a5eb0408b04500d9b0464fea181001900f487844
E: #CD:0068046001200e49c8710198000ac8720198887205984872fff7b6fe8346bbf1
E: #CD:000f01d05846d1e7a6eb080644443d4400bf002edad10020c8e700000e010000
E: #CD:0000024070b504460d460cb9042070bd40200349c87108468571fff795fef6e7
E: #CD:000002402de9ff4d0c4617461d46dde90cb8ddf838a01db9042004b0bde8f08d
E: #CD:04233a4621460098fff71eff06460eb13046f2e722e014487844006804600220
E: #CD:1249c871084680f80bb00f4979441439286809688860fff767fe06465eb1b8f1
E: #CD:000f01d0c8f80040baf1000f02d00020caf8000004e03f1f2d1d241d002fdad1
E: #CD:00bf3046c9e70000740000000000024000000000000008000000100000002000
E: #CD:0000400000008000000000010000000104000240000000000000000000001000
E: #CD:0200000000100000000000000000000000000000000000000000000000010000
E: #CD:c4040020c4040020c4040020000000000000000000000000d8040020d8040020
E: #CD:0100000001000000e8040020e804002000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000100001c0500201c050020
E: #CD:1c05002000000000000000000000000030050020300500200100000001000000
E: #CD:4005002040050020000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000001000074050020740500207405002000000000
E: #CD:0000000000c201000001030000c201000001030000093d000080000048070020
[... 66 lines cut ...]
E: #CD:ba96e72a8ce31c64154e0e16ff88d92d71e27b34f1ae83c0b9c2398706ff086d
E: #CD:2b23309655c6fc62220e39c8ed0543484c0a0020a53c0000300700206c0e0020
E: #CD:ffffffffdb440000300700203007002048070020f9180000e9030000f1170000
E: #CD:560400000000000010000000dd52000048070020d13e0000e8030000ef080000
E: #CD:0000000000ed00e04c0a00200000000000000000d3490000a419000000000061
E: #CD:e803000000000000e80300000100000000000000a505002001000000814a0000
E: #CD:a00a00206c000020000a0020c30800000100000015710000b1700000db260000
E: #CD:b926000000000000000000000000000000000000396100000000000041610000
E: #CD:89a0d3cd6373ecb8649e4c016cd815fb5624c3d2cde51c712e4b3fc3fbf0a65e
E: #CD:44a4039092ef1015b2483683161e35cf7d2f00000a000000480200200b0b0000
E: #CD:ed0a00009c0f0020b9070000ff070000ed0a0000ac0f0020b9070000ff070000
E: #CD:000000000d0000007d2f00005001000040000000400f00200903002009030020
E: #CD:1f3d00000000000001000000230a00001f3d0000480000000903002003290000
E: #CD:1f3d0000940f002009030020ff1700001f3d0000450000000903002000000000
E: #CD:600200200000000000000000000000000600000006580000292000000b170000
E: #CD:0000000000000020901000009b2c0000180700204d0100000000209010002000
E: #CD:410100241000002048100020371700006002002089170000c00100205a450100
E: #CD:0300050000000000c001002000000000c00100204810002000000000a91f0000
E: #CD:0000000048100020800000000000000000000000030000001807002033130000
E: #CD:ffffff7f330c0000000000000000000080000000000000000f00000013070000
E: #CD:1607000000000061883600002920000050070020c00100204001002050070020
E: #CD:90100020850f000090100020fdffffff
E: #CD:END#
E: Halting system
```

mmahadevan108, 2020-09-15:
@hakehuang I debugged this issue on RT600. The test is writing 0 to address 0x0. This write is erasing the debug_console function pointer, hence the coredump is not printed on the console. 
On RT600, 0x0 is mapped to Shared RAM access via CM33 code bus and 0x30000000 is mapped to Shared RAM access CM33 data bus.

closed 2020-09-15 by mmahadevan108

hakehuang, 2020-09-16:
> @hakehuang I debugged this issue on RT600. The test is writing 0 to address 0x0. This write is erasing the debug_console function pointer, hence the coredump is not printed on the console.
> On RT600, 0x0 is mapped to Shared RAM access via CM33 code bus and 0x30000000 is mapped to Shared RAM access CM33 data bus.
> 
> Could this be a test issue?

let me fix this, Thanks @mmahadevan108