zephyrproject-rtos/zephyr #28216: socket: send fails instead of blocking when there are no more net buffers

armandciejak, 2020-09-09:
**Describe the bug**
When trying to send a large amount of data to a TCP socket an application can quickly exhausts the net buffers. The standard behavior of `send` it to block until buffers are released (see man page extract below), but in Zephyr `send` returns an error.

Send man page:

> When the message does not fit into the send buffer of the socket, send() normally blocks, unless the socket has been placed in nonblock‐ing I/O mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply patch in [test-tcp2.zip](https://github.com/zephyrproject-rtos/zephyr/files/5182994/test-tcp2.zip)
2. west build -b mimxrt1050_evk zephyr/samples/net/sockets/dumb_http_server
3. west flash
4. load the page with curl `while true; do time curl http://192.168.1.241:8080 --output test || break; done`

**Expected behavior**
`send` should block until more data can be send and not fail.

**Impact**
Since we need to serve "large" files via HTTP this problem prevents us to deliver stable firmware to our customers.

**Logs and console output**
Here is what is printed when the error occurs:

```
Connection #150 from 192.168.1.146
Err[00:00:55.861,000] <err> net_pkt: Data buffer (1500) allocation failed.
```

**Environment (please complete the following information):**
 - OS: Kubuntu 20.04
 - Toolchain: Zephyr SDK 0.11.4, west 0.7.3
 - Zephyr: v2.4.0-rc1

jukkar, 2020-09-09:
Ping @pfalcon, any opinion how we should act on different error / edge cases?

pfalcon, 2020-09-09:
The approach I always advocated was: follow the POSIX (https://pubs.opengroup.org/onlinepubs/9699919799/) for behavior. When POSIX is not detailed enough (which is often), follow the behavior of Linux.

So yes, in this case, if send() is non-blocking, we should return EAGAIN, and if it's blocking, just wait until send() can get a buffer (waiting indefinitely isn't a good option for unattended apps, that's why there's https://github.com/zephyrproject-rtos/zephyr/issues/11820). The only exception to the above is when we can be sure that send() will never succeed, i.e. if we can detect a deadlock. E.g., if we reserve a TX per open socket, and there're as many sockets as buffers. Then we can return something like ENOBUFS. But we likely would detect that in socket(), not in send(). And it's still questionable, because maybe there're too many open sockets now, but some may get closed, and we will be able to get more buffers.

All in all, I think that cases like this are much more important to deal with, than for example https://github.com/zephyrproject-rtos/zephyr/pull/27054. That PR deals with unrealistic conditions which would never be hit but any reasonable written app (while adds noticeable overhead). Whereas here we hit quite real issues actually happening.

jukkar, 2020-09-09:
Currently the networking samples assume that `send()` can return less bytes that what user wanted to send. So we have this kind of loops in the code:
```
data = content;
len = sizeof(content);
while (len) {
	int sent_len = send(client, data, len, 0);

	if (sent_len == -1) {
		printf("Error sending data to peer\n");
		break;
	}

	data += sent_len;
	len -= sent_len;
}
```
Is the expectation here that we would not need to do this any more? After all the `send()` manual page says `When the message does not fit into the send buffer of the socket, send() normally blocks, unless the socket has been placed in nonblock‐ing I/O mode.` which looks this case here.

pfalcon, 2020-09-09:
> Currently the networking samples assume that send() can return less bytes that what user wanted to send. So we have this kind of loops in the code:
Is the expectation here that we would not need to do this any more?

I'm not sure whom you ask, if myself, then nope, all of send()/recv()/write()/read() can always return with less bytes read/written than requested. That's essence of POSIX API, and why it scales down to even low-memory systems, like our usecase. 

Note that recv() has MSG_WAITALL flag requesting to return the requested number of bytes, and as usual, we have feature request for that (as usual, not backed by anyone) https://github.com/zephyrproject-rtos/zephyr/issues/13091 . But POSIX is inconsistent here, and doesn't provide MSG_WAITALL for send().

(For full disclosure, opinions vary on details of POSIX semantics, I saw people who think that send() doesn't provide MSG_WAITALL, because it's implicit. I personally don't share that opinion, and consider that send(), just as write(), may return with less bytes than was requested, and portable software should account for that.)

jukkar, 2020-09-09:
> all of send()/recv()/write()/read() can always return with less bytes read/written than requested. That's essence of POSIX API, and why it scales down to even low-memory systems, like our usecase.

So how this is then different from the case if send() would return 0 if we could not send any data (like if we ran out of mem). Currently this error is propagated to application.

pfalcon, 2020-09-09:
> So how this is then different from the case if send() would return 0 if we could not send any data (like if we ran out of mem).

Well, 0 return from write()/send() is a grey area at all. For read()/recv(), 0 return means EOF. But there's no such concept for write operations. And POSIX doesn't specify what 0 return from write()/send() means. Thus, reasonable approach is to never return 0 from these calls (as there's no portable way to deal with that, so most applications wouldn't deal with that at all, and others would deal "randomly").

Instead, blocking send() should wait for minimal amount of data which it can send (should be configurable; value of 1 would lead to very poor performance), send as much as it can, and return to app with the number of bytes sent.

I can offer argument why letting an app loop is better than looping in the kernel: with app looping, it usually should come naturally that one app (thread) doesn't suffocate another - they all will have a chance to run in a normal way, per their thread priority. Of course, you probably can achieve the same with kernel looping, but that needs to be done, tested, maintained to avoid priority regressions, etc. And it's only more of system-specific behavior for apps to rely on, thus leading to non-portable apps.

jukkar, 2020-09-11:
@armandciejak I have a fix proposal for this, please try it and report if it works for you.

armandciejak, 2020-09-11:
@jukkar Thanks for the fix. I tested it on the MIMXRT1050 EVK and it works much better than before.

Here are my observations:
- I cannot reproduce the crash described in #28131 anymore
- Send does not fail anymore
- I still observe a problem, sometimes this error `<err> net_pkt: Data buffer (1500) allocation failed.` occurs and in this case the file downloaded by curl is larger than it should be

Here is how to reproduce the problem:
- Configure the dumb_http_server project to serve the 100k file
- Compile the dumb_http_server project with my patch
- Ping the board with `ping -A 192.168.5.10`
- In parallel load the file with curl using `./test_http_download.sh`

[test_http_download.zip](https://github.com/zephyrproject-rtos/zephyr/files/5209341/test_http_download.zip)

The error usually occurs after several thousand successful downloads.

Diff between the original file and the downloaded one:

```diff
--- response_100k.html.bin      2020-09-11 15:07:56.000000000 +0200
+++ test        2020-09-11 15:24:26.291530020 +0200
@@ -1,6 +1,3 @@
-HTTP/1.0 200 OK
-Content-Type: text/html; charset=utf-8
-
 <html>
 <head>
 <style>
@@ -1100,6 +1097,25 @@
 eSBwYXJ0IG9mCiAgICAgICAgICB0aGUgRGVyaXZhdGl2ZSBXb3JrczsgYW5kCgogICAgICAoZCkg
 SWYgdGhlIFdvcmsgaW5jbHVkZXMgYSAiTk9USUNFIiB0ZXh0IGZpbGUgYXMgcGFydCBvZiBpdHMK
 ICAgICAgICAgIGRpc3RyaWJ1dGlvbiwgdGhlbiBhbnkgRGVyaXZhdGl2ZSBXb3JrcyB0aGF0IFlv
+dSBkaXN0cmlidXRlIG11c3QKICAgICAgICAgIGluY2x1ZGUgYSByZWFkYWJsZSBjb3B5IG9mIHRo
+ZSBhdHRyaWJ1dGlvbiBub3RpY2VzIGNvbnRhaW5lZAogICAgICAgICAgd2l0aGluIHN1Y2ggTk9U
+SUNFIGZpbGUsIGV4Y2x1ZGluZyB0aG9zZSBub3RpY2VzIHRoYXQgZG8gbm90CiAgICAgICAgICBw
+ZXJ0YWluIHRvIGFueSBwYXJ0IG9mIHRoZSBEZXJpdmF0aXZlIFdvcmtzLCBpbiBhdCBsZWFzdCBv
+bmUKICAgICAgICAgIG9mIHRoZSBmb2xsb3dpbmcgcGxhY2VzOiB3aXRoaW4gYSBOT1RJQ0UgdGV4
+dCBmaWxlIGRpc3RyaWJ1dGVkCiAgICAgICAgICBhcyBwYXJ0IG9mIHRoZSBEZXJpdmF0aXZlIFdv
+cmtzOyB3aXRoaW4gdGhlIFNvdXJjZSBmb3JtIG9yCiAgICAgICAgICBkb2N1bWVudGF0aW9uLCBp
+ZiBwcm92aWRlZCBhbG9uZyB3aXRoIHRoZSBEZXJpdmF0aXZlIFdvcmtzOyBvciwKICAgICAgICAg
+IHdpdGhpbiBhIGRpc3BsYXkgZ2VuZXJhdGVkIGJ5IHRoZSBEZXJpdmF0aXZlIFdvcmtzLCBpZiBh
+bmQKICAgICAgICAgIHdoZXJldmVyIHN1Y2ggdGhpcmQtcGFydHkgbm90aWNlcyBub3JtYWxseSBh
+cHBlYXIuIFRoZSBjb250ZW50cwogICAgICAgICAgb2YgdGhlIE5PVElDRSBmaWxlIGFyZSBmb3Ig
+aW5mb3JtYXRpb25hbCBwdXJwb3NlcyBvbmx5IGFuZAogICAgICAgICAgZG8gbm90IG1vZGlmeSB0
+aGUgTGljZW5zZS4gWW91IG1heSBhZGQgWW91ciBvd24gYXR0cmlidXRpb24KICAgICAgICAgIG5v
+dGljZXMgd2l0aGluIERlcml2YXRpdmUgV29ya3MgdGhhdCBZb3UgZGlzdHJpYnV0ZSwgYWxvbmdz
+aWRlCiAgICAgICAgICBvciBhcyBhbiBhZGRlbmR1bSB0byB0aGUgTk9USUNFIHRleHQgZnJvbSB0
+aGUgV29yaywgcHJvdmlkZWQKICAgIwZXJ0YWluIHRvIGFu
+eSBwYXJ0IG9mCiAgICAgICAgICB0aGUgRGVyaXZhdGl2ZSBXb3JrczsgYW5kCgogICAgICAoZCkg
+SWYgdGhlIFdvcmsgaW5jbHVkZXMgYSAiTk9USUNFIiB0ZXh0IGZpbGUgYXMgcGFydCBvZiBpdHMK
+ICAgICAgICAgIGRpc3RyaWJ1dGlvbiwgdGhlbiBhbnkgRGVyaXZhdGl2ZSBXb3JrcyB0aGF0IFlv
 dSBkaXN0cmlidXRlIG11c3QKICAgICAgICAgIGluY2x1ZGUgYSByZWFkYWJsZSBjb3B5IG9mIHRo
 ZSBhdHRyaWJ1dGlvbiBub3RpY2VzIGNvbnRhaW5lZAogICAgICAgICAgd2l0aGluIHN1Y2ggTk9U
 SUNFIGZpbGUsIGV4Y2x1ZGluZyB0aG9zZSBub3RpY2VzIHRoYXQgZG8gbm90CiAgICAgICAgICBw
```

jukkar, 2020-09-14:
>     * I still observe a problem, sometimes this error `<err> net_pkt: Data buffer (1500) allocation failed.` occurs and in this case the file downloaded by curl is larger than it should be

I found out this duplicate data issue and there is a new version at #28302, can you re-try it?

jukkar, 2020-09-14:
> I found out this duplicate data issue and there is a new version at #28302, can you re-try it?

Hmm, I seem to have messed something in rebase, the sample app does not work any more. Investigating...

jukkar, 2020-09-14:
> Hmm, I seem to have messed something in rebase, the sample app does not work any more. Investigating...

Fixed the issue in #28302, I am not seeing duplicate bytes any more. @armandciejak can you verify whether this works in your test setup.

armandciejak, 2020-09-14:
@jukkar I tested on the MIMXRT1050 EVK board and do not see the size missmatch problem anymore but I had a crash.

```
Connection #3847 from 192.168.1.135
Connection from 192.168.1.135 closed
Connection #3848 from 192.168.1.135
[00:06:59.153,000] <err> os: ***** BUS FAULT *****
[00:06:59.153,000] <err> os:   Precise data bus error
[00:06:59.153,000] <err> os:   BFAR Address: 0x485db9bc
[00:06:59.153,000] <err> os: r0/a1:  0x00000080  r1/a2:  0x00000001  r2/a3:  0x00000080
[00:06:59.153,000] <err> os: r3/a4:  0x485db9ae r12/ip:  0x00000000 r14/lr:  0x6000e965
[00:06:59.153,000] <err> os:  xpsr:  0x61000200
[00:06:59.153,000] <err> os: Faulting instruction address (r15/pc): 0x6000e8b8
[00:06:59.153,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:06:59.153,000] <err> os: Current thread: 0x80001918 (unknown)
[00:06:59.187,000] <err> os: Halting system
```

```
6000e8b0 <pkt_cursor_advance>:
        if (!cursor->buf) {
6000e8b0:       6943            ldr     r3, [r0, #20]
{
6000e8b2:       b410            push    {r4}
        if (!cursor->buf) {
6000e8b4:       b15b            cbz     r3, 6000e8ce <pkt_cursor_advance+0x1e>
        len = write ? cursor->buf->size : cursor->buf->len;
6000e8b6:       b141            cbz     r1, 6000e8ca <pkt_cursor_advance+0x1a>
6000e8b8:       89dc            ldrh    r4, [r3, #14]
        if ((cursor->pos - cursor->buf->data) == len) {
6000e8ba:       6982            ldr     r2, [r0, #24]
6000e8bc:       689b            ldr     r3, [r3, #8]
6000e8be:       1ad2            subs    r2, r2, r3
6000e8c0:       42a2            cmp     r2, r4
6000e8c2:       d104            bne.n   6000e8ce <pkt_cursor_advance+0x1e>
}
```

jukkar, 2020-09-15:
I am not seeing exactly this backtrace but fixed one null pointer access in the PR.

armandciejak, 2020-09-15:
@jukkar I could run my test longer but still managed to crash it.

```
Connection #5385 from 192.168.1.135
[00:13:20.548,000] <err> os: ***** USAGE FAULT *****
[00:13:20.548,000] <err> os:   Illegal load of EXC_RETURN into PC
[00:13:20.548,000] <err> os: r0/a1:  0x80012308  r1/a2:  0x600213a0  r2/a3:  0x60021410
[00:13:20.548,000] <err> os: r3/a4:  0x80012317 r12/ip:  0x00002710 r14/lr:  0x00000000
[00:13:20.548,000] <err> os:  xpsr:  0x00000200
[00:13:20.548,000] <err> os: Faulting instruction address (r15/pc): 0x00000080
[00:13:20.548,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:13:20.548,000] <err> os: Current thread: 0x80001918 (unknown)
[00:13:20.578,000] <err> os: Halting system
```

armandciejak, 2020-09-16:
I discovered another issue, while testing on MIMXRT1050 EVK board. I enhanced my test script to check the integrity of the received data in addition to checking for the size. Now I see that the data received is sometimes corrupted, see examples below.

The problem occurs systematically after running the test script for a few minutes.

Test script:

```shell
#!/bin/sh

SERVER='192.168.1.241:8080'
EXPECTED_FILE_SIZE=108222

while true; do
    curl http://$SERVER --max-time 10 --output test || exit $?
    FILE_SIZE=$(stat --printf=%s test)
    if [ $FILE_SIZE -ne $EXPECTED_FILE_SIZE ]; then
        echo "wrong size, expected $EXPECTED_FILE_SIZE, got $FILE_SIZE"
        exit 1
    fi
    md5sum -c MD5SUM || exit $?
done
```

Data corruption:

```diff
--- zephyr/samples/net/sockets/dumb_http_server/src/response_100k.html.bin	2020-09-15 17:25:30.968076371 +0200
+++ test_corrupted	2020-09-15 17:56:02.788244941 +0200
@@ -851,7 +851,8 @@
 b3VyIHVzZSwKICAgICAgcmVwcm9kdWN0aW9uLCBhbmQgZGlzdHJpYnV0aW9uIG9mIHRoZSBXb3Jr
 IG90aGVyd2lzZSBjb21wbGllcyB3aXRoCiAgICAgIHRoZSBjb25kaXRpb25zIHN0YXRlZCBpbiB0
 aGlzIExpY2Vuc2UuCgogICA1LiBTdWJtaXNzaW9uIG9mIENvbnRyaWJ1dGlvbnMuIFVubGVzcyBZ
-b3UgZXhwbGljaXRseSBzdGF0ZSBvdGhlcndpc2UsCiAgICAgIGFueSBDb250cmlidXRpb24gaW50
+b3UgZXhwbGljaXRseSBzdGF0ZudHJpYnV0aW9uLiIKCiAg
+IGFueSBDb250cmlidXRpb24gaW50
 ZW50aW9uYWxseSBzdWJtaXR0ZWQgZm9yIGluY2x1c2lvbiBpbiB0aGUgV29yawogICAgICBieSBZ
 b3UgdG8gdGhlIExpY2Vuc29yIHNoYWxsIGJlIHVuZGVyIHRoZSB0ZXJtcyBhbmQgY29uZGl0aW9u
 cyBvZgogICAgICB0aGlzIExpY2Vuc2UsIHdpdGhvdXQgYW55IGFkZGl0aW9uYWwgdGVybXMgb3Ig
```

```diff
--- zephyr/samples/net/sockets/dumb_http_server/src/response_100k.html.bin	2020-09-15 17:25:30.968076371 +0200
+++ test_corrupted2	2020-09-16 09:22:06.151574262 +0200
@@ -1121,7 +1121,7 @@
 dCB0byBZb3VyIG1vZGlmaWNhdGlvbnMgYW5kCiAgICAgIG1heSBwcm92aWRlIGFkZGl0aW9uYWwg
 b3IgZGlmZmVyZW50IGxpY2Vuc2UgdGVybXMgYW5kIGNvbmRpdGlvbnMKICAgICAgZm9yIHVzZSwg
 cmVwcm9kdWN0aW9uLCBvciBkaXN0cmlidXRpb24gb2YgWW91ciBtb2RpZmljYXRpb25zLCBvcgog
-ICAgICBmb3IgYW55IHN1Y2ggRGVyaXZhdGl2ZSBXb3JrcyBhcyBhIHdob2xlLCBwcm92aWRlZCBZ
+ICAgYCBmb3IgYW55IHN1Y2ggRGVyaXZhdGl2ZSBXb3JrcyBhcyBhIHdob2xlLCBwcm92aWRlZCBZ
 b3VyIHVzZSwKICAgICAgcmVwcm9kdWN0aW9uLCBhbmQgZGlzdHJpYnV0aW9uIG9mIHRoZSBXb3Jr
 IG90aGVyd2lzZSBjb21wbGllcyB3aXRoCiAgICAgIHRoZSBjb25kaXRpb25zIHN0YXRlZCBpbiB0
 aGlzIExpY2Vuc2UuCgogICA1LiBTdWJtaXNzaW9uIG9mIENvbnRyaWJ1dGlvbnMuIFVubGVzcyBZ
```

```diff
--- zephyr/samples/net/sockets/dumb_http_server/src/response_100k.html.bin	2020-09-15 17:25:30.968076371 +0200
+++ test_corrupted3	2020-09-16 09:34:51.169839237 +0200
@@ -790,8 +790,7 @@
 ICAgIHdvcmxkd2lkZSwgbm9uLWV4Y2x1c2l2ZSwgbm8tY2hhcmdlLCByb3lhbHR5LWZyZWUsIGly
 cmV2b2NhYmxlCiAgICAgIGNvcHlyaWdodCBsaWNlbnNlIHRvIHJlcHJvZHVjZSwgcHJlcGFyZSBE
 ZXJpdmF0aXZlIFdvcmtzIG9mLAogICAgICBwdWJsaWNseSBkaXNwbGF5LCBwdWJsaWNseSBwZXJm
-b3JtLCBzdWJsaWNlbnNlLCBhbmQgZGlzdHJpYnV0ZSB0aGUKICAgICAgV29yayBhbmQgc3VjaCBE
-ZXJpdmF0aXZlIFdvcmtzIGluIFNvdXJjZSBvciBPYmplY3QgZm9ybS4KCiAgIDMuIEdyYW50IG9m
+b3JtLCBzdWJsaWNlbnNlLCBhbmQgZGlzdHJpYnV0ZSB0aGUKICAgICAgV29yayBhbmQgc3VjaCBmcgZnJvbSBtZWNoYW5pY2FsCiAgIFNvdXJjZSBvciBPYmplY3QgZm9ybS4KCiAgIDMuIEdyYW50IG9m
 IFBhdGVudCBMaWNlbnNlLiBTdWJqZWN0IHRvIHRoZSB0ZXJtcyBhbmQgY29uZGl0aW9ucyBvZgog
 ICAgICB0aGlzIExpY2Vuc2UsIGVhY2ggQ29udHJpYnV0b3IgaGVyZWJ5IGdyYW50cyB0byBZb3Ug
 YSBwZXJwZXR1YWwsCiAgICAgIHdvcmxkd2lkZSwgbm9uLWV4Y2x1c2l2ZSwgbm8tY2hhcmdlLCBy
```

armandciejak, 2020-09-16:
I run the same test on STM32 Nucleo-144 F767ZI board and got the following errors:

```
Connection from 192.168.1.135 closed
Connection #26 from 192.168.1.135
Connection from 192.168.1.135 closed
Connection #27 from 192.168.1.135
Connection from 192.168.1.135 closed
Connection #28 from 192.168.1.135
[00:03:55.091,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:55.091,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:55.292,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:55.292,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:55.492,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:55.492,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:55.693,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:55.693,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:55.895,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:55.895,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:56.095,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:56.095,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:56.270,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:03:56.272,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:03:56.272,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:03:56.276,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:03:56.296,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:56.296,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
[00:03:56.497,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:03:56.497,000] <err> net_tcp: conn: 0x20028a68 packet allocation failed, len=1452
```

I cannot download the file via HTTP more than a few times (less than 10).

jukkar, 2020-09-16:
I managed to source the mimxrt1050_evk board and I am not seeing these issues. I am currently running a script that checks that the file was downloaded properly and so far the test has downloaded 15000 times the 100kb file. Just make sure that the main stack size is larger than 1024, so `CONFIG_MAIN_STACK_SIZE=2048` for example. The "kernel stacks" command revealed that the stack usage was 1052 in my tests.

I have also the same stm32 board, I can try that next.

jukkar, 2020-09-16:
The issue in stm32 is that the driver returned incorrect value if it could not allocate buffers. Fixed by #28431

armandciejak, 2020-09-16:
> I managed to source the mimxrt1050_evk board and I am not seeing these issues. I am currently running a script that checks that the file was downloaded properly and so far the test has downloaded [...]

I ran all my tests with:

```
CONFIG_MAIN_STACK_SIZE=8192

CONFIG_NET_BUF_TX_COUNT=256
CONFIG_NET_MAX_CONN=20
CONFIG_NET_MAX_CONTEXTS=10
CONFIG_NET_SOCKETS_POLL_MAX=5
CONFIG_NET_TCP_BACKLOG_SIZE=30
```

armandciejak, 2020-09-17:
@jukkar Here is patch which allows catching NULL pointer access on i.MXRT SoC. It configures the MPU to prevent any access at address 0, which is the base address of the ITCM (not used in the sample application). With this patch any access to NULL triggers an MPU fault and therefore allows to figure out which instruction caused the problem.

```diff
diff --git a/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h b/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h
index 4bcebed4e6..e68f3a8d81 100644
--- a/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h
+++ b/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h
@@ -130,6 +130,11 @@
 #define REGION_PPB_ATTR(size) { (STRONGLY_ORDERED_SHAREABLE | size | \
                P_RW_U_NA_Msk) }
 #define REGION_IO_ATTR(size) { (DEVICE_NON_SHAREABLE | size | P_RW_U_NA_Msk) }
+#define REGION_NA_ATTR(size) \
+{ \
+       (NORMAL_OUTER_INNER_WRITE_BACK_WRITE_READ_ALLOCATE_NON_SHAREABLE | \
+        MPU_RASR_XN_Msk | size | NO_ACCESS_Msk) \
+}
 
 struct arm_mpu_region_attr {
        /* Attributes belonging to RASR (including the encoded region size) */
diff --git a/soc/arm/common/cortex_m/arm_mpu_regions.c b/soc/arm/common/cortex_m/arm_mpu_regions.c
index 409546b49b..6d711d0c37 100644
--- a/soc/arm/common/cortex_m/arm_mpu_regions.c
+++ b/soc/arm/common/cortex_m/arm_mpu_regions.c
@@ -28,6 +28,12 @@ static const struct arm_mpu_region mpu_regions[] = {
 #else
                         REGION_RAM_ATTR(REGION_SRAM_SIZE)),
 #endif
+#if defined(CONFIG_SOC_FAMILY_IMX)
+       /* Region 2 - ITCM for NULL pointer detection */
+       MPU_REGION_ENTRY("NULL",
+                        0x0,
+                        REGION_NA_ATTR(REGION_16K)),
+#endif /* CONFIG_SOC_MIMXRT1062 */
 };
 
 const struct arm_mpu_config mpu_config = {
```

I had an occurrence while testing today with a 10 M-bit hub and MIMXRT1050-EVK.

```
Connection #1211 from 192.168.1.135
Connection from 192.168.1.135 closed
Connection #1212 from 192.168.1.135
Connection from 192.168.1.135 closed
Connection #1213 from 192.168.1.135
[00:03:46.274,000] <err> os: ***** MPU FAULT *****
[00:03:46.274,000] <err> os:   Data Access Violation
[00:03:46.274,000] <err> os:   MMFAR Address: 0x94
[00:03:46.274,000] <err> os: r0/a1:  0x80006e50  r1/a2:  0x00000001  r2/a3:  0x00000080
[00:03:46.274,000] <err> os: r3/a4:  0x8000c208 r12/ip:  0x00000000 r14/lr:  0x6000ea53
[00:03:46.274,000] <err> os:  xpsr:  0x21000000
[00:03:46.274,000] <err> os: Faulting instruction address (r15/pc): 0x6000e9cc
[00:03:46.274,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:03:46.274,000] <err> os: Current thread: 0x80001918 (unknown)
[00:03:46.307,000] <err> os: Halting system
```

```
6000e9bc <net_pkt_cursor_operate>:
{
6000e9bc:       e92d 43f8       stmdb   sp!, {r3, r4, r5, r6, r7, r8, r9, lr}
6000e9c0:       4606            mov     r6, r0
6000e9c2:       460f            mov     r7, r1
6000e9c4:       4615            mov     r5, r2
6000e9c6:       4699            mov     r9, r3
6000e9c8:       f89d 8020       ldrb.w  r8, [sp, #32]
        while (c_op->buf && length) {
6000e9cc:       6973            ldr     r3, [r6, #20]
6000e9ce:       2b00            cmp     r3, #0
6000e9d0:       d051            beq.n   6000ea76 <net_pkt_cursor_operate+0xba>
6000e9d2:       b90d            cbnz    r5, 6000e9d8 <net_pkt_cursor_operate+0x1c>
        return 0;
6000e9d4:       4628            mov     r0, r5
6000e9d6:       e00d            b.n     6000e9f4 <net_pkt_cursor_operate+0x38>
6000e9d8:       f896 303d       ldrb.w  r3, [r6, #61]   ; 0x3d
                pkt_cursor_advance(pkt, net_pkt_is_being_overwritten(pkt) ?
6000e9dc:       4630            mov     r0, r6
6000e9de:       f013 0f01       tst.w   r3, #1
6000e9e2:       bf0c            ite     eq
6000e9e4:       4641            moveq   r1, r8
6000e9e6:       2100            movne   r1, #0
6000e9e8:       f7ff ffa6       bl      6000e938 <pkt_cursor_advance>
                if (c_op->buf == NULL) {
6000e9ec:       6973            ldr     r3, [r6, #20]
6000e9ee:       b91b            cbnz    r3, 6000e9f8 <net_pkt_cursor_operate+0x3c>
                return -ENOBUFS;
6000e9f0:       f06f 0068       mvn.w   r0, #104        ; 0x68
}
```

jukkar, 2020-09-18:
@armandciejak I just sent a new version of the PR, seems to work ok in my testing but please try it too.

closed 2020-09-19 by nashif

dleach02, 2020-09-20:
@armandciejak, the PR was merged to allow more test coverage in the current release cycle. Can you test with the release to see how it affects your report? We can reopen this bug or create a new one if you still have problems.

armandciejak, 2020-09-21:
It works better but I still experiment crashes, I'll report them in #28121 since the functionality requested in this issue is now working fine.

armandciejak, 2020-09-21:
@jukkar I just got a new occurrence of data corruption with v2.4.0-rc2. Shall I open a new issue specific to this problem?

jukkar, 2020-09-21:
> Shall I open a new issue specific to this problem?

Yes please.

armandciejak, 2020-09-22:
@jukkar I opened #28587

commit 69d9eb6c (2020-09-19): net: socket: Wait new buffers when sending

If we run out of network buffers and cannot send data, and if
we have a blocking socket, then wait until new buffers are
available before returning. As this might lead to deadlock,
wait only max 10 seconds and return ENOMEM if we cannot get
buffers in a reasonable amount of time.

Fixes #28216