zephyrproject-rtos/zephyr #28220: flash: revise API to remove block restrictions on write operations

pabigot, 2020-09-09:
#28192 exists in part because of [my comment](https://github.com/zephyrproject-rtos/zephyr/pull/23628#discussion_r465795677).

I had  what seemed like a reasonable expectation: that users of the generic flash API (like file systems) should not have to independently provide a solution to satisfying arbitrary block size requirements imposed by specific flash drivers that they are given to use.

I still believe this is the only reasonable solution from a system perspective.  Certainly different flash technologies may have write size restrictions, but exposing them to downstream users is not a clean interface.

So either the flash API should ensure that unaligned writes of sub-block length are supported, or there should be an adaptation layer that makes it appear like they're supported.  I don't see how the second approach could work, but it's a potential solution.

nvlsianpu, 2020-09-28:
Due to memory technology limitation software which implements a data storage need to be aware of memory instance characteristic. Also technically not each memory instance allow multiple writes to write-block-size entity at all.

I would like a solution which will allow to support requested feature for each driver selectively.

Second solution i can imagine as a software-flash-driver implementation which uses hardware-flash-driver (we have) as hardware access layer.

pabigot, 2020-09-29:
2020-09-29 API: Limitations on flash devices make it difficult to promise anything about sub-block writes, e.g. whether they are even allowed, and their impact on write endurance calculations.  @d3zd3z was identified as somebody who should provide input.

Absent a clear need I expect to close this as not worth dealing with: any infrastructure that needs to use a flash driver for write operations will have to be prepared to deal with the write block size restrictions imposed by the flash.

pabigot: will close 2020-10-06 if no further activity.

d3zd3z, 2020-10-07:
I believe this should be closed, and should not be implemented. As @nvlsianpu mentions, in order to be able to use a flash device in a way that can survive untimely powerdowns, the code needs to understand the write and erase restrictions of the underlying hardware. Faking that in a driver, by, for example, reading, erasing and writing will result in code that causes corruption or loss when power is lost during the operation.

closed 2020-10-07 by pabigot