zephyrproject-rtos/zephyr #28223: LEDs in the board nRF52840dk_nRF52840 dont work with Lora

GABS9718, 2020-09-09:
**Describe the bug**
Hi, I'm trying to integrate the LoRa library with the LEDs in the development kit nRF52840. I notice that when I change the file prj.conf to the settings of the LoRa, the sketch from \samples\basic\blinky does not work anymore. I can build and flash the program without errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to <zephyr\samples\basic\blinky>
2. Open the file prj.conf
3. Change it to 

CONFIG_LOG=y
CONFIG_SPI=y
CONFIG_GPIO=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
CONFIG_PRINTK=y

4. Build and flash

**Expected behavior**
I think that the LED should keep working as normal.

**Impact**
The LED does not work

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU ARM Embedded
 - The newst version avaiable on the repository
- Board: nrf52840dk_nrf52840
- DeviceTree that I'm using for this board is attached, as I made some changes for use LoRa (**it is .txt format**) and I dont know if those changes could have some impact on this problem. This file is in \zephyr\boards\arm\nrf52840dk_nrf52840
[nrf52840dk_nrf52840.txt](https://github.com/zephyrproject-rtos/zephyr/files/5194880/nrf52840dk_nrf52840.txt)

**Additional context**
I could not understand how changing just the file .conf could have impact in the LEDs working as the main.c is the same.

Mani-Sadhasivam, 2020-09-14:
@GABS9718 Did you see any error in serial console?

pabigot, 2020-09-15:
The difference is with `CONFIG_LORA_SX12XX=y` the sx1276 driver is being initialized and started.

I can replicate the behavior using your DTS patch without having LoRa hardware attached (I don't have any).  The `k_msleep()` in the main routine is never waking.

For some reason the SX1276 driver successfully initializes, but there's no way it can work.  Please see [the hardware connection data](https://infocenter.nordicsemi.com/topic/ug_nrf52840_dk/UG/nrf52840_DK/connector_if.html) for this board.  Many of the pins you're using are already in use for other purposes.

In particular, you've assigned P0.01 to DIO0, but that pin is connected to the 32 KiHz crystal that drives the system clock.  So the timer isn't advancing, and the sleep doesn't wake, and the LED isn't toggled.

You should probably also use `CONFIG_GPIO_AS_PINRESET=n` because the default behavior is to use P0.18 as a hardware reset (it's connected to the reset button).  You're currently using P0.18 as the `RESETn` signal to the SX1276.  That is not the cause of this behavior.

Also P0.5 (DIO3) is the RTS signal for UART0.

Any or all of those would cause problems, but it's P0.1 that's preventing the blink from working.  When I change all DIOs to P0.2 and the RESETn to P0.3 the blinking occurs.

closed 2020-09-15 by pabigot

GABS9718, 2020-09-16:
Hi @pabigot. You are right. Thank you so much for the answer. Best regards.