zephyrproject-rtos/zephyr #28229: Possible NULL dereference in subsys/net/ip/net_context.c.

tluio, 2020-09-09:
**Describe the bug**
In function context_sendto(), **msghdr**  and **dst_addr** are checked:
    if (!msghdr && !dst_addr &&
	    !(IS_ENABLED(CONFIG_NET_SOCKETS_CAN) &&
	      net_context_get_ip_proto(context) == CAN_RAW)) {
		return -EDESTADDRREQ;
   }
If both of them are NULL, some variables such as **ll_addr** on line 1454 and **can_addr** on line 1488 are NULL, too. However, they are dereferenced through **ll_addr->sll_ifindex** and **can_addr->can_ifindex**.

jukkar, 2020-09-10:
The **ll_addr** access on line 1454 will not happen because the the check at line 1372 (that you copied above) will catch that (AF_PACKET socket is not handling CAN packets).
The **can_addr** check on line 1488 could happen, I will send a fix.

jukkar, 2020-09-10:
> The **ll_addr** access on line 1454 will not happen because the the check at line 1372 (that you copied above) will catch that (AF_PACKET socket is not handling CAN packets).

Hmm, after better look at the code, we might have same protocol numbers depending on the family. We need to add family check to line 1372.

tluio, 2020-09-10:
Ok. I'm not sure if **addr6**  on line 1401 and **addr4** on line 1427 have the same problem. Could you please have a look at them.

jukkar, 2020-09-10:
I simplified the address checks at line 1372, which then required that the sendmsg() handling needed to be added to AF_CAN sockets. This should now work uniformly for all currently supported socket families.

PR #28271 (2020-09-10): Fix sendmsg() for AF_CAN sockets
Fixes #28229
commit 0c7d2bcd (2020-09-10): net: context: Simplify the param check when sending data

No need to specifically check CAN sockets here. This also means
that SocketCAN will need to support sendmsg() call too. This will
come in following commit.

Fixes #28229
closed 2020-09-10 by MaureenHelm