zephyrproject-rtos/zephyr #28230: "make zephyr_generated_headers" produces incorrect result (SHELL:")

pfalcon, 2020-09-09:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

pfalcon, 2020-09-09:
(Sorry, I initially mis-pressed Enter and incomplete ticket was submitted, I edited description afterwards, so it might not go into email notification.)

tejlmand, 2020-09-09:
Fixed: https://github.com/zephyrproject-rtos/zephyr/pull/28232

PR #28232 (2020-09-10): cmake: export build flags to external build systems excluding SHELL tag
Fixes: #28230

The introduction of CMake SHELL prefix in compile options to avoid
symbol de-duplication caused the build flag export feature to fail.

The export itself works, but unfortunately CMake only strips `SHELL`
prefix for its own compiler invocation, and not when exporting the build
flags.

This causes `SHELL` to be present in the Makefile.exports.\<language\>
files, leading to problems in external build systems.

This is fixed by using `zephyr_get_compile_options_for_lang()` that
strips `SHELL` and use the return [... cut]
commit a5917f0a (2020-09-10): cmake: export build flags to external build systems excluding SHELL tag

Fixes: #28230

The introduction of CMake SHELL prefix in compile options to avoid
symbol de-duplication caused the build flag export feature to fail.

The export itself works, but unfortunately CMake only strips `SHELL`
prefix for its own compiler invocation, and not when exporting the build
flags.

This causes `SHELL` to be present in the Makefile.exports.<language>
files, leading to problems in external build systems.

This is fixed by using `zephyr_get_compile_options_for_lang()` that
strips `SHELL` and use the returne [... cut]
closed 2020-09-10 by carlescufi