zephyrproject-rtos/zephyr #28270: Errors in HL7800.c file

is-none, 2020-09-10:
**Describe the bug**
I have a Pinnacle 100 dvk board that I integrated to Platformio to be used with Zephyr 2.3. I need to use the HL7800 semiconductor that is not available in the current version of Zephyr but may be for the 2.4 version. So I decided to add what is missing manually, so far so good. I end up with these errors:

**1)**

> ~/.platformio/packages/framework-zephyr/drivers/modem/hl7800.c: In function 'mdm_gpio6_callback_isr':
 ~/.platformio/packages/framework-zephyr/drivers/modem/hl7800.c:3607:2: error: #endif without #if
 #endif
  ^~~~~

**2)** I tried to fix this one but I am missing something...

> ~/.platformio/packages/framework-zephyr/drivers/modem/hl7800.c: In function 'hl7800_init':
 ~/.platformio/packages/framework-zephyr/drivers/modem/hl7800.c:4862:15: error: 'struct mdm_receiver_context' has no member named 'data_imei'; did you mean 'data_iccid'?
  ictx.mdm_ctx.data_imei = ictx.mdm_imei;
               ^~~~~~~~~
               data_iccid

**3)**

> At top level:
 ~.platformio/packages/framework-zephyr/drivers/modem/hl7800.c:322:19: warning: 'TIME_STRING_FORMAT' defined but not used [-Wunused-const-variable=]
 static const char TIME_STRING_FORMAT[] = "\"yy/MM/dd,hh:mm:ss?zz\"";

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK

rerickson1, 2020-09-10:
This is fixed in https://github.com/zephyrproject-rtos/zephyr/pull/27755. There are other fixes/improvements as well.

rerickson1, 2020-09-16:
Fix via https://github.com/zephyrproject-rtos/zephyr/pull/27755

closed 2020-09-16 by rerickson1