zephyrproject-rtos/zephyr #28282: Slave host auto-initiate stalls if master does not support extended reject indications, and procedure collision occurs

mtpr-ot, 2020-09-10:
**Problem**
If an LLCP collision occurs during auto-initiate, e.g. with PHY_UPDATE, and master sends a REJECT_IND, auto-initatiate is not continued, as no notification is sent to host from LL.

**Impact**
Other host- or app initiated LLCP requests are never sent.

**Additional context**
In function `ctrl_rx` `(ull_conn.c)`, `PDU_DATA_LLCTRL_TYPE_REJECT_IND` is not handled properly. The case needs to handle the packet similar to `reject_ind_phy_upd_recv` in `reject_ext_ind_recv`.

NOTE: This was found on the 2.2.1 branch, but as far as I can tell, it hasn't been fixed on master.

cvinayak, 2020-09-11:
@mtpr-ot will a PR be upstream soon? I will be available to review/test the same.

mtpr-ot, 2020-09-11:
> @mtpr-ot will a PR be upstream soon? I will be available to review/test the same.

I'm not sure I can provide the full functionality - I have only fixed locally for PHY_UPDATE. Perhaps I can have Thomas (@thoh-ot) make the full implementation.

mtpr-ot, 2020-09-14:
The following patch implements REJECT_IND handling for PHY collision only:
[phy_reject_ind_fix.zip](https://github.com/zephyrproject-rtos/zephyr/files/5217145/phy_reject_ind_fix.zip)

PR #28550 (2020-09-21): Bluetooth: controller: Fix REJECT_IND PDU handling
Fix for handling REJECT_IND PDU received for PHY Update,
Connection Parameter Request and Data Length Update control
procedures.

If a link layer control procedure collision occurs, example
with local initiated PHY Update Procedure, and peer sends a
REJECT_IND PDU, then the PHY Update Procedure is stalled.

Fixes #28282.
commit a8ffc03f (2020-09-21): Bluetooth: controller: Fix REJECT_IND PDU handling

Fix for handling REJECT_IND PDU received for PHY Update,
Connection Parameter Request and Data Length Update control
procedures.

If a link layer control procedure collision occurs, example
with local initiated PHY Update Procedure, and peer sends a
REJECT_IND PDU, then the PHY Update Procedure is stalled.

Fixes #28282.
closed 2020-09-21 by carlescufi