zephyrproject-rtos/zephyr #28289: tests: arch: arm: arm_sw_vector_relay: fails on nucleo_f091rc

ABOSTM, 2020-09-10:
**Describe the bug**
tests: arch: arm: arm_sw_vector_relay: fails on nucleo_f091rc

**To Reproduce**
Steps to reproduce the behavior:
1. west -p auto -b nucleo_f091rc
2. west flash
3. See error

**Expected behavior**
Test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc1-22-gf60a11368632  ***
Running test suite arm_sw_vector_relay
===================================================================
START - test_arm_sw_vector_relay

    Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_sw_vector_relay/src/arm_sw_vector_relay.c:64: test_arm_sw_vector_relay: (_vector_table_pointer == (uint32_t)_vector_start is false)
vector table pointer not pointing to vector start, 0xffffffff, 0x80000bc

 FAIL - test_arm_sw_vector_relay
===================================================================
Test suite arm_sw_vector_relay failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS:  Linux, and Windows
 - Toolchain Zephyr SDK
 - Commit SHA: d6fa062a7d3fe322dd50d4e4371f49b6283a0185

ABOSTM, 2020-09-10:
I started the analysis, and it seems to me that issue is linked to the fact that nucleo_f091rc is based on cortex-M0 and thus without VTOR register.
Also there is a specific implementation for stm32f0 serie of function `relocate_vector_table()`  in zephyr\soc\arm\st_stm32\stm32f0\soc.c
This specific implementation is anterior to feature SW_VECTOR_RELAY, and I think I can get ride of this specific implementation of `relocate_vector_table()` and use SW_VECTOR_RELAY instead.

But test is still failed when I remove the specific implementation:
In the generic implementation
```
void __weak relocate_vector_table(void)
{
#if defined(CONFIG_XIP) && (CONFIG_FLASH_BASE_ADDRESS != 0) || \
    !defined(CONFIG_XIP) && (CONFIG_SRAM_BASE_ADDRESS != 0)
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	(void)memcpy(VECTOR_ADDRESS, _vector_start, vector_size);
#elif defined(CONFIG_SW_VECTOR_RELAY) || defined(CONFIG_SW_VECTOR_RELAY_CLIENT)
	_vector_table_pointer = _vector_start;
#endif
}
```
code in #if is executed because CONFIG_XIP is true and CONFIG_FLASH_BASE_ADDRESS=0x8000000.
In other words, even if CONFIG_SW_VECTOR_RELAY is set, the corresponding code (#else) is not effective.
And of course test is failed.

If I revert the order of both  conditions
```
void __weak relocate_vector_table(void)
{
#if defined(CONFIG_SW_VECTOR_RELAY) || defined(CONFIG_SW_VECTOR_RELAY_CLIENT)
	_vector_table_pointer = _vector_start;
#elif defined(CONFIG_XIP) && (CONFIG_FLASH_BASE_ADDRESS != 0) || \
    !defined(CONFIG_XIP) && (CONFIG_SRAM_BASE_ADDRESS != 0)
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	(void)memcpy(VECTOR_ADDRESS, _vector_start, vector_size);
#endif
}
```
then test is passed for nucleo_f091rc.

But I don't know whether CONFIG_SW_VECTOR_RELAY should take precedence over CONFIG_XIP ...

ABOSTM, 2020-09-10:
^^ @ioannisg

ioannisg, 2020-09-14:
Yes, this might be a test issue; the test does not account for platforms that concurrently enable the code relocation feature. Will take a look as well.

b0661, 2020-09-15:
The _vector_table_pointer is always 0. In case of relocation the RAM is additionally mapped to 0; without relocation the FLASH is additionally mapped to 0.

- _vector_start is in FLASH, only if the FLASH is also mapped to address 0 (no relocation) it is the "real" _vector_table_pointer.
- VECTOR_ADDRESS is in RAM, only if the RAM is also mapped to address 0 (relocation) it is the "real" _vector_table_pointer.

I don´t know about the CONFIG_SW_VECTOR_RELAY feature. The relocation is basically used for bootloader operation. A bootloader (any application started without bootloader) starts with the the FLASH mapped to address 0. When the bootloader starts the application, the application maps the RAM to address 0 and relocates it´ s vector table to the start of RAM to make the relocated application vector table the one used by the processor. If the application is started without bootloader it may use the vector table from FLASH or do the mapping to RAM to make the vector table mutable. 

In case you want to replace the whole mimic by CONFIG_SW_VECTOR_RELAY, the feature  has to account for the two scenarios - running as a bootloader like application (can live with the vector table in FLASH or relocates to RAM) and running as a bootloader started application (always needs the vector table relocated to RAM). It has also to provide the dependent mapping of RAM/ FLASH to address 0.

ABOSTM, 2020-09-15:
@b0661, I am a bit confused:
> The _vector_table_pointer is always 0

_vector_table_pointer variable is only defined under CONFIG_SW_VECTOR_RELAY feature

> _vector_start is in FLASH, only if the FLASH is also mapped to address 0 (no relocation) it is the "real" _vector_table_pointer.

On nucleo_f091rc (and most STM32 chip), Flash is mapped to address 0x08000000, and `CONFIG_FLASH_BASE_ADDRESS=0x8000000`.
(even if there is some capability to also access it from 0, depending on BOOT Option Bytes) 
so `_vector_start ` is also 0x08000000 and this is the real vector table when there is no relocation.

As per the 2 scenarios you talked about:
Both can be covered using CONFIG_SW_VECTOR_RELAY by default for such board (boards which don't have VTOR), so we don't have to distinguish between bootloader like application and bootloader started application.

b0661, 2020-09-15:
@ABOSTM Sorry for the confusing text.

>>   The _vector_table_pointer is always 0

> _vector_table_pointer variable is only defined under CONFIG_SW_VECTOR_RELAY feature

Should read "the real vector table pointer (that is used by the processor) is always 0". You may have a vector table at start of RAM and another vector table at start of FLASH. Depending on BOOT options and later on mapping one of these will be mapped to address 0.

> As per the 2 scenarios you talked about:
Both can be covered using CONFIG_SW_VECTOR_RELAY by default for such board (boards which don't have VTOR), so we don't have to distinguish between bootloader like application and bootloader started application.

My only concern is to  avoid the extra RAM space occupied by an unused vector table in case my application is fine with the FLASH vector table. So replacing everything with the CONFIG_SW_VECTOR_RELAY mechanism should in one way allow to avoid the RAM vector table and have the FLASH vector table only. If this is the case I´m fine with it.

ioannisg, 2020-09-15:
@ABOSTM thanks for your input. I have some comments:

> Also there is a specific implementation for stm32f0 serie of function relocate_vector_table() in zephyr\soc\arm\st_stm32\stm32f0\soc.c

This forces the SoC to copy the vector table from ROM to RAM and use the SoC specific HW Relocation mechanism to switch to using the RAM table. This is only happening if the Zephyr image is _not_ Bootloader. 

So I am wondering: if Zephyr _is_ a bootloader how does the vector table "work" directly from ROM? If it does work, then I think this should also be the case for _standalone_ non-bootloader images (i.e. images that are _not_ loaded by mcuboot or other bootloader). This would save you the space on RAM.

Now, if you really want to keep the current implementation, with the SoC-specific relocate_vector_table function, we have the following options:

1. Do not allow neither SW_VECTOR_RELAY nor SW_VECTOR_RELAY_CLIENT Kconfig options to be enabled on STM43F0: we can block them using `depends on !CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP`. This is clean, but I'd say "unfortunate" since the SW_VECTOR_RELAY feature could in-theory work on this platform. We have recently made SW vector table forwarding work evn on Main-line Cortex-M as well (i.e. on SoCs with VTOR) for some corner use-cases, so it would be nice that is works unconditionally for any Cortex-M processor. But, anyway, if you insist on not having this on SoCs with HW-mechanism, this is an option...

2. Allow SW_VECTOR_RELAY and SW_VECTOR_RELAY_CLIENT to be enabled unconditionally. Then we need to make sure that when any of these options are set, the STM32F0 relocate_vector_table function does not do anything and let the SW table forwarding to do the job.

I am open to both options, let me know what you prefer.
Copying @erwango

ABOSTM, 2020-09-15:
> So I am wondering: if Zephyr is a bootloader how does the vector table "work" directly from ROM?
yes that is the way it works.

I prefer option 2) so that SW_VECTOR_RELAY and SW_VECTOR_RELAY_CLIENT are available on this platform.

And so to save space in RAM, I kept the original specific redefinition of `relocate_vector_table()` of stm32f3.
Just add the case of SW_VECTOR_RELAY or SW_VECTOR_RELAY_CLIENT activation.
see #28405

I am still wondering whether the generic weak implementation should also apply the inversion of conditions ?
**prep_c.c**
```
void __weak relocate_vector_table(void)
{
#if defined(CONFIG_SW_VECTOR_RELAY) || defined(CONFIG_SW_VECTOR_RELAY_CLIENT)
	_vector_table_pointer = _vector_start;
#elif defined(CONFIG_XIP) && (CONFIG_FLASH_BASE_ADDRESS != 0) || \
    !defined(CONFIG_XIP) && (CONFIG_SRAM_BASE_ADDRESS != 0)
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	(void)memcpy(VECTOR_ADDRESS, _vector_start, vector_size);
#endif
}
```

ioannisg, 2020-09-15:
> yes that is the way it works.

Then, how about standalone zephyr images (without bootloader)? why do they need to use the HW re-map mechanism? Wouldn't it be better to "behave" as a bootloader, then?

ioannisg, 2020-09-15:
>I am still wondering whether the generic weak implementation should also apply the inversion of conditions ?
**prep_c.c**
```
void __weak relocate_vector_table(void)
{
#if defined(CONFIG_SW_VECTOR_RELAY) || defined(CONFIG_SW_VECTOR_RELAY_CLIENT)
	_vector_table_pointer = _vector_start;
#elif defined(CONFIG_XIP) && (CONFIG_FLASH_BASE_ADDRESS != 0) || \
    !defined(CONFIG_XIP) && (CONFIG_SRAM_BASE_ADDRESS != 0)
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	(void)memcpy(VECTOR_ADDRESS, _vector_start, vector_size);
#endif
}
```

This looks weird, actually. Unless I am missing something,
if we have `XIP && FLASH_BASE_ADDRESS != 0` then the memcpy will copy the table to address 0, which probably is outside flash, so that should give a fault, actually.

@ABOSTM could you file a ticket for this, and we can take a look outside the scope of this bug report?

ABOSTM, 2020-09-16:
@ioannisg 
> This looks weird, actually. Unless I am missing something,
> if we have `XIP && FLASH_BASE_ADDRESS != 0` then the memcpy will copy the table to address 0, which probably is outside flash, so that should give a fault, actually.

If `XIP && FLASH_BASE_ADDRESS != 0`  it may expect to have RAM at 0x0 and then make a copy/relocation of vector to RAM ?
(assuming processor is always working with 0x0 address vector table)
This could explain why F0 need a custom function  `relocate_vector_table()`, to add dedicated mechanism to remap SRAM at 0x0.

PR #28405 (2020-09-21): soc: arm: st_stm32: stm32f0: take into account SW_VECTOR_RELAY
soc: arm: st_stm32: stm32f0: take into account SW_VECTOR_RELAY

This STM32 serie redefines function relocate_vector_table()
It should take into account features:
SW_VECTOR_RELAY and SW_VECTOR_RELAY_CLIENT

fixes #28289
commit 2e1499eb (2020-09-21): soc: arm: st_stm32: stm32f0: take into account SW_VECTOR_RELAY

This STM32 serie redefines function relocate_vector_table()
It should take into account features:
SW_VECTOR_RELAY and SW_VECTOR_RELAY_CLIENT

fixes #28289
closed 2020-09-21 by carlescufi