zephyrproject-rtos/zephyr #28294: arch.interrupt.gen_isr_table.arm_mainline: lpcxpresso55s16_ns failed

hakehuang, 2020-09-11:
**Describe the bug**
interrupt 57 didn't run once, ran 0 times

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s16_ns ..
3. make
4. See error

**Expected behavior**
test pass

**Impact**
core interrupt settings

**Logs and console output**

```
 *** Booting Zephyr OS version 2.4.0-rc1  ***

START - Test gen_isr_tables

IRQ configuration (total lines 60):

isr1 isr=0x000004b9 irq=59

isr2 isr=0x000004e1 irq=58

Checking _irq_vector_table entry 59 for irq 59

triggering irq 59

isr1 ran

Checking _irq_vector_table entry 58 for irq 58

triggering irq 58

isr2 ran

_sw_isr_table at location 0x20000018

isr3 isr=0x00000429 irq=57 param=0xb01dface

Checking _sw_isr_table entry 57 for irq 57

triggering irq 57

interrupt 57 didn't run once, ran 0 times

 FAIL - main

===================================================================

===================================================================

PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0-rc1

hakehuang, 2020-09-11:
@MaureenHelm @dleach02

galak, 2020-09-15:
What are you using for the trusted firmware / secure side?

closed 2020-09-15 by MaureenHelm

hakehuang, 2020-09-16:
> What are you using for the trusted firmware / secure side?

No, looks like the case shall not run for this ns platform