zephyrproject-rtos/zephyr #28295: kernel.common.stack_protection_arm_fpu_sharing: lpcxpresso55s16_ns test failure

hakehuang, 2020-09-11:
**Describe the bug**
ESF could not be retrieved successfully. Shall never occur.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
test pass

**Impact**
kernel common function

**Logs and console output**

```
 *** Booting Zephyr OS version 2.4.0-rc1  ***

Running test suite fatal

===================================================================

START - test_fatal

test alt thread 1: generic CPU exception

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:955

	ESF could not be retrieved successfully. Shall never occur.

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:955

	ESF could not be retrieved successfully. Shall never occur.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.4.0

hakehuang, 2020-09-11:
@MaureenHelm @dleach02

galak, 2020-09-15:
Is this being run with some TFM since this is a `non-secure` platform?

hakehuang, 2020-09-16:
> Is this being run with some TFM since this is a `non-secure` platform?

@galak, no TFM is running.

ioannisg, 2020-09-17:
@hakehuang @MaureenHelm  I am executing all test suites on nRF5340dk_nrf5340cpuapp**ns**, which is also a TrustZone-M enabled image running at Non-Secure state, and all tests are currently passing (except for tests relying on ISRs; those won't ever pass reliably because non-secure domain does not fully control Interrupt handling).

To be able to do the above, I have flashed a "TF-M" image in the secure version of the board that allows the non-secure zephyr image to execute and run the tests. Otherwise, the nRF "**ns**" platforms won't even boot, because the image base address is set  (by default) to slot_0_nonsecure. In other words a Zephyr image for an "**ns**" board should not be a standalone image.

So why is lpcxpresso55s16_**ns** able to boot and run these tests?

This is probably because the base address of the generated image is 0x00000000. This allows Cortex-M33 to boot this image as a standalone image. Not surprisingly, the image runs in Secure mode.

However, the board _does_ enable the ARM_NONSECURE_FIRMWARE Kconfig switch (because it thinks it runs in non-secure mode). In fault.c there is a code block that is only executed if ARM_NONSECURE_FIRMWARE is defined. The failures you see are there because of the platform trying to execute this code block.

In summary, the problem is that we cannot expect boards with TRUSTED_EXECUTION_NONSECURE set to be standalone images and run the Zephyr test suites as such. Sanity check could filter that out, unless we run these boards together with TF-M integration (FYI @microbuilder ). We need to work on that, but in the mean time, I 'd say we ignore failures on "**ns**" boards.

hakehuang, 2020-09-18:
@ioannisg  thanks for detailed explaination, I will skip those failures

dleach02, 2020-09-23:
Do we want to keep this as a bug then?

closed 2020-09-24 by hakehuang