zephyrproject-rtos/zephyr #28298: Deep sleep(system off) is not working with LVGL and display driver

honestech74, 2020-09-11:
**Describe the bug**
I am going to write a deep sleep in nRF52840-DK. Also I am using the LVGL to display the graph on the SPI LCD screen. But the application was halted at the beginning in zephyr. I can see only one line on debug output.

**To Reproduce**
Steps to reproduce the behavior:
1. Merge ```samples/boards/nrf/system_off``` to ```samples/display/lvgl```.
2. add the SPI LCD on the nrf52840dk_nrf52840.overlay

**Expected output**
```*** Booting Zephyr OS build v2.3.0-rc1-ncs1  ***                       
                                                                                
nrf52840dk_nrf52840 system off demo                                             
Busy-wait 2 s                                                                   
Sleep 2 s                                                                                       
Sleep 2000 ms (deep sleep minimum)                                              
Entering system off; press BUTTON1 to restart ```

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```*** Booting Zephyr OS build v2.3.0-rc1-ncs1  ***```

**Environment (please complete the following information):**
 - OS: Ubuntu 18
 - Toolchain Zephyr SDK-11
 - Zephyr version: v2.3.0

**Additional context**
I have merged ```samples/boards/nrf/system_off``` to ```samples/drivers/display```, then it works well.
I guess that LVGL or SPI conflicts with power management.

pabigot, 2020-09-11:
As I explained on slack: as long as you're using the Nordic fork which apparently doesn't have #26580 enabling deep sleep won't work.  You need to confirm this fails to work on current master, or file this as a bug against the downstream fork, or integrate #26580 and #25490 into your environment.

MaureenHelm, 2020-09-15:
@honestech74 can you reproduce this issue in upstream master?

pabigot, 2020-09-15:
Marking as low as this is believed to be fixed; close or reprioritize if that changes.

closed 2020-11-29 by github-actions[bot]

RuslanSpartak, 2024-03-16:
What I found on that. Seems that after system off zephyr display driver is not ready, but it is not true. If not adding return statement after unsuccesseful checking if it is ready - all is working as expexted.