zephyrproject-rtos/zephyr #28305: Device not found (SX1276 with nRF52840)

GABS9718, 2020-09-11:
**Describe the bug**
Hi, I have been trying to set up the LoRa with the nRF52840, but I'm having some problems that I dont know how to fix it up. I always get the same error message in the output **00> [00:00:00.257,202] <err> lora_send: sx1276 Device not found**

**To Reproduce**
Steps to reproduce the behavior:
1. Add the line _lora0 = &lora;_ in the aliases options from the .dts related to the board nrf52840dk_nrf52840 
2. Add 
	_lora: sx1276@0 {
		status = "okay";
		compatible = "semtech,sx1276";
		reg = <0>;
		label = "sx1276";
		reset-gpios = <&gpio0 13 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
		dio-gpios = <&gpio0 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 4 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
		spi-max-frequency = <1000000>;
		power-amplifier-output = "pa-boost";		
	};_
to the spi1 options in the same file.
3. Change the prf.conf from samples\drivers\lora\send to 

CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

CONFIG_LOG=y
CONFIG_SPI=y
CONFIG_SPI_1=y
CONFIG_SPI_NRFX=y
CONFIG_GPIO=y
CONFIG_PRINTK=y

CONFIG_LORA=y
CONFIG_LORA_SHELL=y
CONFIG_LORA_SX1276=y
CONFIG_HAS_SEMTECH_RADIO_DRIVERS=y
CONFIG_HAS_SEMTECH_LORAMAC=y

**Expected behavior**
Find the SX1276 in the device tree

**Impact**
I always get the same error 
00> [00:00:00.257,202] <err> lora_send: sx1276 Device not found

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU ARM Embedded
 - The newest one from the repository

**Additional context**
I'm new in the context of LoRa, so I'm not pretty sure if I should have to do something else in order to make the SX1276 work with the board nRF52840. Every help is appreciated. Thanks.

Mani-Sadhasivam, 2020-09-11:
@GABS9718 The SPI controllers in nrf52 chipsets can function in multiple modes. So you need to add the relevant compatible in the spi1 node board dts for proper working.

compatible = "nordic,nrf-spi"

GABS9718, 2020-09-11:
I have this line 

&spi1 {
	compatible = "nordic,nrf-spi";
	status = "okay";
	sck-pin = <31>;
	mosi-pin = <30>;
	miso-pin = <25>;	
	cs-gpios = <&arduino_header 13 GPIO_ACTIVE_LOW>;

	lora: sx1276@0 {
		compatible = "semtech,sx1276";
		reg = <0>;
		label = "sx1276";
		reset-gpios = <&gpio0 6 GPIO_ACTIVE_LOW>;
		dio-gpios = <&gpio0 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 4 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio0 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
		spi-max-frequency = <1000000>;
		power-amplifier-output = "pa-boost";		
	};
};

GABS9718, 2020-09-14:
Hi @Mani-Sadhasivam. I saw in one topic that you were able to send the data and receive it in TTN using the OTAA.  I can get the message ''Data sent'' in the log output, but I dont receive in my gateway. I think that this is related with the settings of OTAA protocol, ''Device EUI'', ''Application EUI'' and ''App key''. I dont know where I should make this settings for the board that I'm using nRF52840dk. Can you help me, please?

Mani-Sadhasivam, 2020-09-14:
> I have this line
>
> &spi1 {
> compatible = "nordic,nrf-spi";
> status = "okay";
> sck-pin = <31>;
> mosi-pin = <30>;
> miso-pin = <25>;
> cs-gpios = <&arduino_header 13 GPIO_ACTIVE_LOW>;
>
> ```
> lora: sx1276@0 { [...]

DTS and Config look sane. Were you able to get any other SPI samples working under the same SPI instance? Had you made sure that SPI controller is enabled and able to use?

Mani-Sadhasivam, 2020-09-14:
> Hi @Mani-Sadhasivam. I saw in one topic that you were able to send the data and receive it in TTN using the OTAA. I can get the message ''Data sent'' in the log output, but I dont receive in my [...]

That's from the LoRaWAN PR: https://github.com/zephyrproject-rtos/zephyr/pull/23664

With this PR you should be able to send data to TTN server by modifying the keys in provided sample application.

GABS9718, 2020-09-14:
I dont 

> DTS and Config look sane. Were you able to get any other SPI samples working under the same SPI instance? Had you made sure that SPI controller is enabled and able to use?

I dont know what caused that error, but it is not happening anymore. Thank you for the answer.

Mani-Sadhasivam, 2020-09-14:
> I dont know what caused that error, but it is not happening anymore. Thank you for the answer.

Okay, closing the issue now!

closed 2020-09-14 by Mani-Sadhasivam