zephyrproject-rtos/zephyr #28309: Sample/subsys/fs/littlefs with nucleo_f429zi

Cimex97, 2020-09-11:
**Describe the bug**
The sample project **subsys/Ifs/littlefs** don't work on a nucleo_f429zi board. I always get this error: `littlefs: format failed (LFS -28)`
The configuration option **CONFIG_APP_WIPE_STORAGE=y** don't change the behavior.

**To Reproduce**
cd zepyr/samples/subsys/fs/littlefs
west build -b nucleo_f429zi .
west flash

**Expected behavior**
Those output is generated by the sample project on the board **nrf52840dk_nrf52840**:
```*** Booting Zephyr OS build zephyr-v2.3.0  ***
Area 4 at 0x0 on MX25R64 for 65536 bytes
[00:00:00.006,683] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:00.006,744] <inf> littlefs: FS at MX25R64:0x0 is 16 0x1000-byte blocks with 512 cycle
[00:00:00.006,744] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.007,080] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:998: Corrupted dir pair at {0x0, 0x1}
[00:00:00.007,110] <wrn> littlefs: can't mount (LFS -84); formatting
/lfs mount: 0
/lfs: bsize = 16 ; frsize = 4096 ; blocks = 16 ; bfree = 14
/lfs/boot_count stat: -2
/lfs/boot_count read count 0: 0
/lfs/boot_count seek start: 0
/lfs/boot_count write new boot count 1: 4
/lfs/boot_count close: 0
/lfs opendir: 0
  F 4 boot_count
End of files
/lfs unmount: 0
[00:00:00.104,095] <inf> littlefs: /lfs mounted
[00:00:00.111,572] <err> fs: failed get file or dir stat (-2)
[00:00:00.132,232] <inf> littlefs: /lfs unmounted

```

**Logs and console output**
Those output is generated by the sample project on the board **nucleo_f429zi**:
```
*** Booting Zephyr OS build zephyr-v2.3.0  ***
Area 1 at 0x10000 on FLASH_CTRL for 65536 bytes
FAIL: mount id 1 at /lfs: -28
[00:00:00.008,000] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:00.008,000] <inf> littlefs: FS at FLASH_CTRL:0x10000 is 1 0x10000-byte blocks with 512 cycle
[00:00:00.008,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.008,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:00.008,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:495: No more free space 1
[00:00:00.008,000] <err> littlefs: format failed (LFS -28)
[00:00:00.008,000] <err> fs: fs mount error (-28)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU ARM EMB 8.3.1
 - Zephyr Version: v2.3.0

erwango, 2020-09-14:
Will look into it in details but generally flash based samples do not play well on nucleo_f429zi due to unusual flash map (cf https://github.com/zephyrproject-rtos/zephyr/issues/19895).

erwango, 2020-09-15:
@Cimex97 Following discussion during review of #28387, it appears that LittleFS is not made to work with few STM32 series such as F2/F4/F7.
Reason is that LittleFS (similarly to other FS solutions) will use one flash page per file to be saved (plus one for the superblock).
Since STM32F2/F4/F7 flash hardware block is made of quite little number of big pages, it would be a waste of memory to use 256KB to save only 1 file.
If you're looking for an alternative storage solution, you can check #19895 that exposes solution to use nvs. Though it requires careful flash partition configuration to respect nvs constraints (expect min 2 blocks of memory with total size < 64kB).
Btw, https://github.com/zephyrproject-rtos/zephyr/pull/23876 should help relaxing those constraints.

pabigot, 2020-09-15:
Marking as low as littlefs is not easily supported on this platform, nothing Zephyr can do about that.  Alternative solutions identified above.  This could also be closed.

pabigot, 2020-09-15:
From Triage the concern is why is this being run on a platform that doesn't support it.  The problem is that the test/sample requires a "storage" partition.  Only boards where that partition has at least four blocks will pass this test.

I don't know of a way to filter it other than the existing "platform_allow", which does not include this platform.

@nashif what should I do?

erwango, 2020-09-16:
Removing the "STM32" tag since the remaining point to discuss is not STM32 specific.

Meanwhile, it wouild be good to have @Cimex97's feedback on ticket evolution.

Cimex97, 2020-09-16:
I tried to use LittleFS in combination with the SMP, that comes with the MCUBoot bootloader to transfer configuration files onto the nucleo_f429zi board. 

I used now the [memory map](https://github.com/zephyrproject-rtos/zephyr/issues/19895#issuecomment-545449820) example from @erwangoc and it works fine, in the sample and my own application. But is it a good idea, when I waste a lot of flash for the LittleFS?

pabigot, 2020-09-16:
I would probably look for a different solution, but offhand I have no suggestions (the ideas I have in mind would be a bit of work, e.g. faked file systems).  If it meets your application needs and you have no other use for the flash it should be fine to use littlefs this way.

Cimex97, 2020-09-16:
I need to talk to my colleagues if we can realize it with a different solution.
Maybe we can help you, to implement your Idea.

closed 2020-11-30 by github-actions[bot]