zephyrproject-rtos/zephyr #28312: Add option to enable cache on STM32 FLASH controller

lowlander, 2020-09-11:
The FLASH controller on the STM32F407 has a cache, this is disabled after reset. 

By enabling it with  `FLASH->ACR |= FLASH_ACR_ICEN | FLASH_ACR_DCEN;` the performance dramatically improves, some places up to 2x .

The idea is to add a Kconfig option somewhere that lets the user sellect if the Icache and/or Dcache should be enabled.

ABOSTM, 2020-11-30:
Hi @lowlander,
As you know, activating cache may be easy, but it is not that easy to guarantee cache coherency across zephyr drivers (see issue https://github.com/zephyrproject-rtos/zephyr/issues/29798 and your PR https://github.com/zephyrproject-rtos/zephyr/pull/27911)
So adding such switch need to study all F4 drivers to identify and fix any cache coherency potential issue.

lowlander, 2020-12-22:
> Hi @lowlander,
> As you know, activating cache may be easy, but it is not that easy to guarantee cache coherency across zephyr drivers (see issue #29798 and your PR #27911)
> So adding such switch need to study all F4 drivers to identify and fix any cache coherency potential issue.

@ABOSTM  This is not the general D/I cache (the F4 has no cache), but the special cache of the internal flash controller only. See `3.5.2 Adaptive real-time memory accelerator (ART Accelerator™)` 

This should not cause any of the DMA problems like on the F7 with a real data cache.

ABOSTM, 2021-01-06:
@lowlander,
I really had in mind ART accelerator.
But F7 also have ART and it suffers from cache coherency, because ART is valid only on ITCM bus.
Also on L4 and G4 there is a note in reference manual about ART: 
> Note: The D-Cache is active only when data is requested by the CPU (not by DMA1 and DMA2).

This note is not present in F4 reference manual, but I have some doubts.
I will investigate.

lowlander, 2021-01-06:
@ABOSTM  since the ART-cache is only valid on the ITCM bus and only for the FLASH.

From the F7 manual, page 92;
>  The ART accelerator is available only for flash access on ITCM interface

After reading some more I found out that on the F7 the FLASH is mapped into the memory map twice, one time on `0x0800_0000 ` via the AXIM bus, and one time on `0x0020_0000` via the ITCM bus. 

The ART-cache in only used when using the `0x0020_0000` address, and the "normal" I/D-Cache is only used when using the `0x08000_0000` address. 

There is some more detail on page 87. 

So when executing code/reading data from `0x0020_0000` the ART-cache is in play, and when executing code/reading data from `0x0800_0000` the I/D cache is on play. 

And any DMA will always have to be on `0x0800_0000` since it can not access the ITCM bus. 

This makes things rather confusing, to when to flush/invalidate/disable which cache. 

When you aren't writing to the FLASH and only execute from `0x0020_0000` the ART-Cache should not cause any problems. 

If you are writing data/code to `0x08000_0000` and than reading from `0x0020_0000` you have both the D-cache (for the write) and the ART-Cache (for the read), that sounds like trouble.

If you do DMA write to `0x08000_0000` and than want to read via `0x08000_0000` you have to D-cache to deal with, and when you want to read from `0x0020_0000` you have the ART-Cache to deal with.

If you want to DMA read from `0x08000_0000` you and than write to `0x08000_0000` you have the D-cache to deal with. Since you can't write to `0x0020_0000` the ART-Cache is fine. 

The only good thing is, I think, that writing to FLASH (or any non-volatile storage) is a bit more of an explicit action than writing to RAM. So it will probably be more easy to find a place to add cache flushes than with other drivers. 

Of course if the user knows (like in my case) that he is not going to write to FLASH enabling the ART-cache should be fine.

ABOSTM, 2021-01-07:
@lowlander, 
warning: on STM32F7, 
> The ART accelerator is available only for flash access on ITCM interface

means that only code execution is concerned, and not data read (which go through DTCM bus instead of ITCM bus).
And this is the reason why on F7, there is only 1 bit to activate (FLASH_ACR_ARTEN)

But now I am a bit confused: original request is to activate ART on F4,
now with your detailed analysis on F7 case, it seems request is larger ...

lowlander, 2021-01-07:
> means that only code execution is concerned, and not data read (which go through DTCM bus instead of ITCM bus).
> And this is the reason why on F7, there is only 1 bit to activate (FLASH_ACR_ARTEN)

If it is just code or if it depends on which address map is used is not 100% clear to me, since the datasheet talks about instructions and data, but at the same time also talks about the ITCM bus (which should be for instructions only)  

> But now I am a bit confused: original request is to activate ART on F4,
> now with your detailed analysis on F7 case, it seems request is larger ...

You were the one that brought up the F7, so I looked into that a bit more to see how it works in combination with the D/I-caches.

ABOSTM, 2021-01-08:
> If it is just code or if it depends on which address map is used is not 100% clear to me, since the datasheet talks about instructions and data, but at the same time also talks about the ITCM bus (which should be for instructions only)

I think that generally code is in ITCM, data in DTCM. 
But that is right that nothing prevent user to put data in ITCM,
as well as nothing prevent user to put instruction in DTCM, as mentioned in F7 reference manual:

      2.1.5 DTCM bus
          This bus is used by the Cortex®-M7 for data access on the DTCM RAM. It can be also used
          for instruction fetches.

And in any case we should follow ARM Cortex M7 architecture which match address to interfaces. 
For example:
* ITCM at 0x00000000  (Max 16MB)
* DTCM at 0x20000000 (Max 16MB)

So the truth is that from hardware point of view address determined the bus used.

> You were the one that brought up the F7, so I looked into that a bit more to see how it works in combination with the D/I-caches.

Ok

lowlander, 2021-01-08:
@ABOSTM so what would be a "good" way to add ART support? Just add an kconfig menu option with a big warning like "only enable when you know what you are doing" ?

aurel32, 2021-01-08:
> @lowlander,
> I really had in mind ART accelerator.
> But F7 also have ART and it suffers from cache coherency, because ART is valid only on ITCM bus.

The main reason behind the cache coherency issues on the F7 is the D-Cache integrated in the Cortex-M7, just like other Cortex-M7 from other manufacturers. It caches both accesses to the flash and to the RAM.

On the F4, the ART only caches flash reads, so it's not an issue to enable it. The only case it happens to be an issue is when part of the flash is rewritten, but it's already taken care of in the STM32 flash driver.

> @ABOSTM so what would be a "good" way to add ART support? Just add an kconfig menu option with a big warning like "only enable when you know what you are doing" ?

I do not think we should add an option, it should just be enabled by default in `soc/arm/st_stm32/stm32f4/soc.c`.

aurel32, 2021-01-08:
Also note that `FLASH_ACR_ICEN` and `FLASH_ACR_DCEN` are enabled by default on STM32L4.

lowlander, 2021-01-15:
@aurel32 @ABOSTM should I make a PR that just enables the ART cache for F4 cores?

aurel32, 2021-01-16:
> @aurel32 @ABOSTM should I make a PR that just enables the ART cache for F4 cores?

I think that would be a good idea to move forward. I guess you want to use the `LL_FLASH_EnableDataCache()` and `LL_FLASH_EnableInstCache()` for that.

ABOSTM, 2021-01-18:
> @aurel32 @ABOSTM should I make a PR that just enables the ART cache for F4 cores?

Yes would be great. Thanks.

PR #36277 (2021-06-28): STM32: Enable ART flash cache
Enable ART Flash accelerator for STM32 series supporting it.
Depending on capabilities, enable both Instruction and Data cache.
When necessary, manage cache coherency in flash driver (Similarly to #32218 and #34032).
* stm32f4
* stm32f7
* stm32h7
* stm32l5
* ~~stm32f2~~  --> #35549

Fixes #28312
closed 2021-06-28 by nashif