zephyrproject-rtos/zephyr #28319: tests: kernel: context: fails because timer expiration is shorter than excepted

ABOSTM, 2020-09-11:
**Describe the bug**
tests/kernel/context/ fails on nucleo_f091rc board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f091rc tests/kernel/context/
2. west flash
3. See error

**Expected behavior**
Test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc1-57-g6c910c021e32  ***
Running test suite context
===================================================================
START - test_kernel_interrupts
 PASS - test_kernel_interrupts
===================================================================
START - test_kernel_timer_interrupts
 SKIP - test_kernel_timer_interrupts
===================================================================
START - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
 PASS - test_kernel_ctx_thread
===================================================================
START - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)
 PASS - test_busy_wait
===================================================================
START - test_k_sleep
 thread sleeping for 50 milliseconds
 thread back from sleep
Testing k_thread_create() without cancellation
 thread (q order: 2, t/o: 500) is running
 got thread (q order: 2, t/o: 500) as expected
 thread (q order: 3, t/o: 750) is running
 got thread (q order: 3, t/o: 750) as expected
 thread (q order: 0, t/o: 1000) is running
 got thread (q order: 0, t/o: 1000) as expected
 thread (q order: 6, t/o: 1250) is running
 got thread (q order: 6, t/o: 1250) as expected
 thread (q order: 1, t/o: 1500) is running
 got thread (q order: 1, t/o: 1500) as expected
 thread (q order: 4, t/o: 1750) is running
 got thread (q order: 4, t/o: 1750) as expected
 thread (q order: 5, t/o: 2000) is running
 got thread (q order: 5, t/o: 2000) as expected
Testing k_thread_create() with cancellations
 cancelling [q order: 0, t/o: 1000, t/o order: 0]
 thread (q order: 3, t/o: 750) is running
 got (q order: 3, t/o: 750, t/o order 1) as expected
 thread (q order: 0, t/o: 1000) is running
 got (q order: 0, t/o: 1000, t/o order 2) as expected
 cancelling [q order: 3, t/o: 750, t/o order: 3]
 cancelling [q order: 4, t/o: 1750, t/o order: 4]
 thread (q order: 4, t/o: 1750) is running
 got (q order: 4, t/o: 1750, t/o order 5) as expected
 cancelling [q order: 6, t/o: 1250, t/o order: 6]
 PASS - test_k_sleep
===================================================================
START - test_kernel_cpu_idle_atomic
 SKIP - test_kernel_cpu_idle_atomic
===================================================================
START - test_kernel_cpu_idle

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/context/src/main.c:272: _test_kernel_cpu_idle: (tms2 < tms is true)
Bad ms value computed,got 5485 which is less than 5486

 FAIL - test_kernel_cpu_idle
===================================================================
START - test_k_yield
 PASS - test_k_yield
===================================================================
START - test_kernel_thread
 PASS - test_kernel_thread
===================================================================
Test suite context failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux and Windows)
 - Toolchain: Zephyr SDK
 - Commit SHA: 6c910c021e32c981e10426ae3a243b5ae3277406

**Analysis:**
I inserted some gpio toggle to measure effective time with logical analyser:
```
static void _test_kernel_cpu_idle(int atomic)
{
	int tms, tms2;
	int i;

	/* Set up a time to trigger events to exit idle mode */
	k_timer_init(&idle_timer, idle_timer_expiry_function, NULL);

	for (i = 0; i < 5; i++) { /* Repeat the test five times */
gpio_pin_set(dev, 0, 1);
		k_timer_start(&idle_timer, K_MSEC(1), K_NO_WAIT);

		tms = k_uptime_get_32();

		if (atomic) {
			unsigned int key = irq_lock();

			k_cpu_atomic_idle(key);
		} else {
			k_cpu_idle();
		}
		tms += 1;
gpio_pin_set(dev, 0, 0);
		tms2 = k_uptime_get_32();
		zassert_false(tms2 < tms, "Bad ms value computed,"
	      "got %d which is less than %d\n",
	      tms2, tms);
	}
} 
```

And the measured time between both gpio edge is 963us which is slightly less than the 1ms excepted.
10 Ticks are requested but measurement is between 9 Ticks and 10 Ticks (1 Tick = 100us).
Looking at function `z_impl_k_timer_start()` there is the following comment:
```
	/* z_add_timeout() always adds one to the incoming tick count
	 * to round up to the next tick (by convention it waits for
	 * "at least as long as the specified timeout"), but the
	 * period interval is always guaranteed to be reset from
	 * within the timer ISR, so no round up is desired.  Subtract
	 * one.
	 *
	 * Note that the duration (!) value gets the same treatment
	 * for backwards compatibility.  This is unfortunate
	 * (i.e. k_timer_start() doesn't treat its initial sleep
	 * argument the same way k_sleep() does), but historical.  The
	 * timer_api test relies on this behavior.
	 */
```
I agree that "period interval is always guaranteed to be reset from within the timer ISR", 
But the duration for the 1st timer expiration is not guarantee, because timer didn't start from ISR.
In other words, because tick is partially started when staring timer, and because no round up is apply, the 1st timer expiration will occurs between N-1 and N tick (N =number of ticks requested).
In my case 10 tick are requested , but timer expires between 9 and 10 ticks (between 900us and 1000ms).
This is not fully in line with Zephyr documentation:

> **Timer Limitations**
> Since timers are based on the system clock, the delay values specified when using a timer are minimum values. (See Clock Limitations.)

Please note that I am probably in a corner case (tight timing window) because problem vanish when inserting temporisation at the early beginning of the function (before the timer start)

ABOSTM, 2020-09-11:
^^ @erwango @andyross @andrewboie

andyross, 2020-09-17:
Looking at this seriously for the first time.  Yeah, this test looks wrong.  The  first "duration" argument to the k_timer_start() API is, as explained in the comment you found, implemented with "round down" semantics vs. "round up" as is the standard for every other blocking timeout API.  And I agree that's crazy, but I left it that way because we have tests that actually depend on it.

One suggestion would be to try to add a "tick align" step (e.g. "k_usleep(1);") to the beginning of the test, to ensure that the k_timer_start() call happens as close to the start of a tick as possible.  That will make the results deterministic.

But my bigger question is what is it about this platform that causes this test to fail there and not everywhere.   Looking at the kconfig, this is just the SysTick timer driver running at 10khz, right?  Why don't we see this on all the other (non-nRF) ARM boards?

andyross, 2020-09-17:
Oh wait, no.  What I wrote above is fine, but that's not where the bug is.  The problem reported isn't that the idle took zero time when it expected to sleep for 1ms.  It's that the **TIME WENT BACKWARDS**.  The second call to k_uptime_get_32() returns a smaller number than the first.

I don't have a theory there at all.  The only way to make that happen is to stuff a corrupt timeout object into the queue and/or have the timer driver (again, this is just systick, right?  It's used everywhere) return a bad value for z_clock_elapsed().

andyross, 2020-09-17:
Heh, no.  I confused myself.  The test code modifies the first result before the comparison and I missed that.  This is just a zero time when 1ms was expected condition.

Can you try this (actually it sounds like you might have already) and verify that it fixes the problem?  If so feel free to submit it, or I can do it:

```c
diff --git a/tests/kernel/context/src/main.c b/tests/kernel/context/src/main.c
index bb4859cb92..6ac2149cfc 100644
--- a/tests/kernel/context/src/main.c
+++ b/tests/kernel/context/src/main.c
@@ -247,6 +247,12 @@ static void _test_kernel_cpu_idle(int atomic)
        int tms, tms2;
        int i;
 
+       /* Align to ticks so the first iteration sleeps long enough
+        * (k_timer_start() rounds its duration argument down, not up,
+        * to a tick boundary)
+        */
+       k_usleep(1);
+
        /* Set up a time to trigger events to exit idle mode */
        k_timer_init(&idle_timer, idle_timer_expiry_function, NULL);
 
```

ABOSTM, 2020-09-18:
@andyross ,
I  tested your proposal and it fixes the issue. Thanks.
I pushed the PR: #28495

PR #28495 (2020-09-18): tests: kernel: context: align timer init on ticks
tests: kernel: context: align timer init on ticks

Align to ticks so the first iteration sleeps long enough
(k_timer_start() rounds its duration argument down, not up,
to a tick boundary)

Fixes #28319
commit 8afcffc2 (2020-09-18): tests: kernel: context: align timer init on ticks

Align to ticks so the first iteration sleeps long enough
(k_timer_start() rounds its duration argument down, not up,
to a tick boundary)

Fixes #28319
closed 2020-09-18 by MaureenHelm