zephyrproject-rtos/zephyr #28326: Sample boards nrf mesh onoff not working

Ripitchip, 2020-09-11:
**Describe the bug**
Trying to run the mesh sample described here : [sample on/off](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/nrf/mesh/onoff-app.)

We have the nRF52840 DK.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr
2. west build -p auto -b nrf52840dk_nrf52840 samples/boards/nrf/mesh/onoff-app/
3. west flash

**Expected behavior**
As the device is unprovisioned, I expect the buttons to turn off/off the leds. That's not the case.

**Logs and console output**
As console output, I get this : [img](https://imgur.com/sCH9JzK)
Were do the errors come from ? Are they normal ?

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 LTS (VM)
 - Zephyr SDK (0.11.3)
 - Zephyr version 2.4.0-rc1

trond-snekvik, 2020-09-17:
Hi,

In your screenshot, the settings load procedure is complaining that it can't find any vendor models, which isn't surprising, as this sample doesn't declare any. 

It looks like this device has been provisioned with another application (or at least another composition data layout), and not fully erased before being flashed with this sample. 
Try fully erasing your board by passing `--erase` to west flash: `west flash --erase`. The board should now start acting as a switch for its own leds, as described.

Ripitchip, 2020-09-20:
Yes, you were right. With the full erase, I could run the sample as expected.

Thank you very much for your help :)

closed 2020-09-20 by Ripitchip