zephyrproject-rtos/zephyr #28331: CDC ACM stopped working on an STM32L432 board after PR #24873

eugmes, 2020-09-11:
I have a custom board that is based on `nucleo_l432kc` with a USB interface. A software that is running a shell
on CDC ACM interface used to work before #24873 got merged.
The last working revision was d92b199173d7cce41f4837cffa93637fe50dfcf2.

**To Reproduce**
Reproducing requires a custom board, however it should be compatible with `nucleo_l432kc` if one manages to connect
a USB cable to it.

1. git clone https://github.com/eugmes/ais-recv-fw.git
2. cd ais-recv-fw.git
3. Change revision in `application/west.yml`
3. west init application
4. west build
5. west flash
6. Connect to the board via CDC ACM serial port using `minicom`, try to press any buttons, observer that there is no visible activity. Note that the USB device is correctly discovered by the OS, and `minicom` does not report any errors. `lsusb -v` output looks fine and no different than the one from a working version.

**Expected behavior**
After connecting to the board via CDC ACM serial port one is able to use shell.

**Impact**
It is not possible to use current versions of Zephyr with the board. Old ones work.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
 - Commit SHA: d92b199173d7cce41f4837cffa93637fe50dfcf2

**Additional context**
I understand that it is hard to diagnose the problem without access to my custom hardware and there my be a problem in my code. I had no success when trying to use USB samples on `stm32l476g_disco` or `stm32f429i_disc1` (before or after the mentioned PR). I may try to use `nucleo_l432kc` if I manage to connect a USB cable to it.

@tbursztyka: maybe you have any ideas what to look at?

jfischer-phytec-iot, 2020-09-11:
You can quckly modify _samples/subsys/usb/console_ and reproduce/test the issue with nrf52840dk board for example :smile:
```diff
diff --git a/samples/subsys/usb/console/prj.conf b/samples/subsys/usb/console/prj.conf
index e51d82323b..6d62c70092 100644
--- a/samples/subsys/usb/console/prj.conf
+++ b/samples/subsys/usb/console/prj.conf
@@ -8,3 +8,8 @@ CONFIG_USB_UART_CONSOLE=y
 CONFIG_UART_INTERRUPT_DRIVEN=y
 CONFIG_UART_LINE_CTRL=y
 CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
+
+CONFIG_SHELL=y
+CONFIG_LOG=y
+CONFIG_SHELL=y
+CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"
diff --git a/samples/subsys/usb/console/src/main.c b/samples/subsys/usb/console/src/main.c
index ecad1f6315..e68d0c78ef 100644
--- a/samples/subsys/usb/console/src/main.c
+++ b/samples/subsys/usb/console/src/main.c
@@ -36,7 +36,7 @@ void main(void)
        }
 
        while (1) {
-               printk("Hello World! %s\n", CONFIG_ARCH);
+               /* printk("Hello World! %s\n", CONFIG_ARCH); */
                k_sleep(K_SECONDS(1));
        }
 }
```

The problem is initialization priority of uart backend _shell_uart.c_, unfortunattely there is no config option to change priority, but you can simple change it to 99 as workaround:

```diff
diff --git a/subsys/shell/shell_uart.c b/subsys/shell/shell_uart.c
index 6af79b392b..5802b0bf9b 100644
--- a/subsys/shell/shell_uart.c
+++ b/subsys/shell/shell_uart.c
@@ -297,7 +297,7 @@ static int enable_shell_uart(const struct device *arg)
 
        return 0;
 }
-SYS_INIT(enable_shell_uart, POST_KERNEL, 0);
+SYS_INIT(enable_shell_uart, POST_KERNEL, 99);
 
 const struct shell *shell_backend_uart_get_ptr(void)
 {
```

eugmes, 2020-09-11:
@jfischer-phytec-iot I've got to the same conclusion right now. It is too bad that there is no sample with shell on USB CDC ACM to test things quickly.

eugmes, 2020-09-12:
This seems to be related to #27985.

PR #28339 (2020-09-15): shell: Make shell UART backend initialization priority configurable
Closes: #28331

Configurable initialization priority is useful when using CDC ACM
serial device, for example.
closed 2020-09-15 by MaureenHelm