zephyrproject-rtos/zephyr #28333: usb: stm32: testusb fails and often hangs forever

mbittan, 2020-09-11:
**Describe the bug**

On the Nucleo F767zi and Nucleo F070rb boards, many test cases from the testusb Linux kernel module fail. Failure rates and their consequences can take various forms: most failures only result in a timeout for a specific tranfer, but some of the failures can render the device unusable, forcing the user to reset the MCU in order to use the USB device again.

The following table summarize the status of each test case for the two boards.
| Test case | nucleo_f767zi | Failure Rate | nucleo_f070rb | Failure Rate |
|:---------:|:--------------------:|:------------:|:--------------------:|:------------:|
|     0     |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     1     |          :x:         |    10-20%    |  :heavy_check_mark:  |      N/A     |
|     2     |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     3     |          :x:         |    10-20%    |  :heavy_check_mark:  |      N/A     |
|     4     |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     5     |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     6     |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     7     |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     8     |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     9     |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     10    |  :heavy_check_mark:  |      N/A     |          :x:         |  100%  |
|     11    |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     12    |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     13    |          :x:         |     100%     |          :x:         |     100%     |
|     14    |          :x:         |     < 10%    |          :x:         |     100%     |
|     17    |          :x:         |    10-20%    |  :heavy_check_mark:  |      N/A     |
|     18    |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     19    |          :x:         |    10-20%    |  :heavy_check_mark:  |      N/A     |
|     20    |  :heavy_check_mark:  |      N/A     |          :x:         |     100%     |
|     21    |          :x:         |     < 10%    |          :x:         |     100%     |
|     24    |          :x:         |     100%     |  :heavy_check_mark:  |      N/A     |
|     27    |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |
|     28    |  :heavy_check_mark:  |      N/A     |          :x:         |  100% |
|     29    |  :heavy_check_mark:  |      N/A     |  :heavy_check_mark:  |      N/A     |

I used the following command line for all the tests:
```
sudo ./testusb -D /dev/usb/xxx/yyy -v 512 -t <test_case>
```

On the nucleo_f767zi board:
- test cases 1, 3, 17 and 19 fail between 10 and 20% of the time. When it fails, the test end with "Connection timed out". The device is still usable after that failure.
- test case 13 always fail with "Invalid argument". It doesn't have any other impact.
- test cases 14 and 21 seem to trigger issue #27992 . Indeed, after enough time, these tests fail and the control endpoints become unresponsive. Unplugging the cable and plugging it again won't solve the issue: the device is unable to enumerate and the MCU has to be reset in order to have a functional USB device.
- test case 24 always fails and is unable to complete. The testusb program stays stuck in the kernel in an uninterruptible sleep state. The device is still operational after plugging it again.

On the nucleo_f070rb board:
- test cases 2, 4, 6, 8, 14, 18, 20, 21 always fail and return "Connection timed out". The device can operate normally after that.
- test case 13 always fail with "Invalid argument".
- test cases 10 and 28 always fail and are stuck in the kernel indefinitely. The device needs to plugged again in order to 

On both boards, whenever a test fails, the log of the device shows:
```
<err> usb_dc_stm32: Unable to get write lock (-16)
<err> usb_dc_stm32: Unable to get write lock (-16)
``` 

Changing the value of `USB_NUMOF_EP_WRITE_RETRIES` from 3 to 30 doesn't seem to have any impact on the tests failures.

**To Reproduce**

For the nucleo_f767zi board:
- Compile and flash the testusb sample
- On the Linux host, follow the instructions from `sample/subsys/usb/testusb/README.rst`
- For each test case, run the following command:
```
sudo ./testusb -D /dev/usb/xxx/yyy -v 512 -t <test_case>
```
Since some test cases only fail around 10% of the time, this command may need to be executed multiple times before the failure occurs.

For the nucleo_f070rb board:
- Since that board doesn't have a USB connector, you'll need an external micro USB connector (the PA11 and PA12 pins can be used as USB D+/D-)
- Once you are able to connect a Linux host to the board, you need to set the status of the "usb" node of the device tree to "okay":
```
diff --git a/boards/arm/nucleo_f070rb/nucleo_f070rb.dts b/boards/arm/nucleo_f070rb/nucleo_f070rb.dts
index 9f5923d36d57..dcdf20912124 100644
--- a/boards/arm/nucleo_f070rb/nucleo_f070rb.dts
+++ b/boards/arm/nucleo_f070rb/nucleo_f070rb.dts
@@ -41,6 +41,10 @@
        };
 };
 
+&usb {
+       status = "okay";
+};
+
 &usart1 {
        current-speed = <115200>;
        status = "okay";
```
- Compile and flash the testusb sample
- On the Linux host, follow the instructions from `sample/subsys/usb/testusb/README.rst`
- Use the command:
```
sudo ./testusb -D /dev/usb/xxx/yyy -v 512 -t <test_case>
```

**Expected behavior**

All tests should pass.

**Impact**

There are multiple consequences to these failures:
- The USB device might not pass USB certification
- Some transfers randomly time out which compromise the communication with a host.
- In some cases, USB is completely unusable without a MCU hardware reset, which is not acceptable or even possible in production.

**Environment (please complete the following information):**
 - Host: Ubuntu 19.10 with  Linux kernel 5.3.0-64-generic
 - Toolchain : Zephyr SDK
 - Commit SHA: c8cca16c204543b8f6dd7261c643b48ba8c95a94
 - Boards: nucleo_f767zi and nucleo_f070rb (with external usb connector)

jfischer-phytec-iot, 2020-09-11:
> Since some test cases only fail around 10% of the time, this command may need to be executed multiple times before the failure occurs.

There is option to set number of iterations  `-c iterations		default 1000`

jfischer-phytec-iot, 2020-09-11:
> For the nucleo_f070rb board:
> 
>     * Since that board doesn't have a USB connector, you'll need an external micro USB connector (the PA11 and PA12 pins can be used as USB D+/D-)

Depending on how this connection was made (simply wires ?), it is probably not reliable and results should not be taken so seriously.

simonguinot, 2020-09-13:
[image]

Hi @jfischer-phytec-iot. Thanks a lot for looking into this issues. See the attached picture to check how the connection is made. IMHO issues are more likely in the driver or in the HAL than in the wires. For each test we made the results are not erratics but systematics and reproducibles.

jfischer-phytec-iot, 2020-09-14:
> IMHO issues are more likely in the driver or in the HAL than in the wires.

The driver certainly has issues/bugs. It is just that the higher number of errors for nucleo_f070rb could be influenced by _bad_ connection.

erwango, 2020-09-14:
^^ @loicpoulain, @ydamigos Would one of you have time to look at this issue?

loicpoulain, 2020-09-15:
I do not have this nucleo board, but can have a look on disco-iot and carbon one.

erwango, 2020-09-15:
@loicpoulain would be great, txs!

loicpoulain, 2020-09-15:
Result is a bit different on disco-iot:

carlescufi, 2020-09-16:
@erwango this seems related to the USB device driver for STM32.

erwango, 2020-09-18:
@loicpoulain So Maybe 13/18/19/20 would be common to all these.
Also, 1, 3, 14, 17 21 have low reproduction rate, so it is not excluded they have the same issue.

I had a try on my side, but I'm stuck.
I'm seeing this:
```
[17780.127010] usb 1-4.2.2: new full-speed USB device number 14 using xhci_hcd
[17780.229296] usb 1-4.2.2: New USB device found, idVendor=2fe3, idProduct=0009, bcdDevice= 2.04
[17780.229302] usb 1-4.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17780.229306] usb 1-4.2.2: Product: Zephyr testusb sample
[17780.229309] usb 1-4.2.2: Manufacturer: ZEPHYR
[17780.229312] usb 1-4.2.2: SerialNumber: 203633413752500C
[17856.701958] usbtest 1-4.2.2:1.0: matched module params, vend=0x2fe3 prod=0x0009
[17856.701962] usbtest 1-4.2.2:1.0: Generic USB device
[17856.701964] usbtest 1-4.2.2:1.0: full-speed {control} tests
[17856.702019] usbcore: registered new interface driver usbtest
```

But the following fails:
```
$ sudo ./testusb -D /dev/bus/usb/009/016
./testusb: /dev/bus/usb/009/016 may see only control tests
can't open dev file r/w: No such file or directory

```
even after:
`$ sudo sh -c "echo 0x2fe3 0x0009 0 0x0525 0xa4a0 > /sys/bus/usb/drivers/usbtest/new_id"`

Am I missing something?

mbittan, 2020-09-18:
It seems you're not using the right path for the USB device. According to your logs, the correct path is probably `/dev/bus/usb/001/014` (the device number 14 on the USB bus 1).

You should unplug the device and plug it again after that command:
```
$ sudo sh -c "echo 0x2fe3 0x0009 0 0x0525 0xa4a0 > /sys/bus/usb/drivers/usbtest/new_id"
```

erwango, 2020-09-21:
So I've done a couple of tests (one run/board) on various boards from various vendors: disco_l475_iot (ST), frdm_k64f (NXP), reel_board (nRF).
Result  is as follows (each time I had to end up the test after > 5 min w/o life sign):
```
disco_l475_iot1

$ sudo ./testusb -D /dev/bus/usb/001/024
./testusb: /dev/bus/usb/001/024 may see only control tests
/dev/bus/usb/001/024 test 0,    0.000010 secs
/dev/bus/usb/001/024 test 1,    2.012529 secs
/dev/bus/usb/001/024 test 2,    2.113732 secs
/dev/bus/usb/001/024 test 3,    2.015036 secs
/dev/bus/usb/001/024 test 4,    2.179339 secs
/dev/bus/usb/001/024 test 5,   54.174164 secs
/dev/bus/usb/001/024 test 6,   60.895902 secs
/dev/bus/usb/001/024 test 7,   54.430262 secs
/dev/bus/usb/001/024 test 8,   61.258191 secs
/dev/bus/usb/001/024 test 9,    1.239666 secs
/dev/bus/usb/001/024 test 10,    5.571183 secs
/dev/bus/usb/001/024 test 11,   16.551997 secs
/dev/bus/usb/001/024 test 12,   16.502997 secs
/dev/bus/usb/001/024 test 13 --> 22 (Invalid argument)
/dev/bus/usb/001/024 test 14 --> 22 (Invalid argument)
/dev/bus/usb/001/024 test 17,    1.912823 secs
/dev/bus/usb/001/024 test 18 --> 110 (Connection timed out)
/dev/bus/usb/001/024 test 19,    2.151868 secs
/dev/bus/usb/001/024 test 20 --> 110 (Connection timed out)
/dev/bus/usb/001/024 test 21 --> 22 (Invalid argument)

frdm_k64f

$ sudo ./testusb -D /dev/bus/usb/001/026
[sudo] password for frq07517:
./testusb: /dev/bus/usb/001/026 may see only control tests
/dev/bus/usb/001/026 test 0,    0.000015 secs
/dev/bus/usb/001/026 test 1,    1.534650 secs
/dev/bus/usb/001/026 test 2 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 3 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 4 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 5 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 6 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 7 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 8 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 9 --> 32 (Broken pipe)
/dev/bus/usb/001/026 test 10 --> 32 (Broken pipe)
/dev/bus/usb/001/026 test 11,   16.931834 secs
/dev/bus/usb/001/026 test 12,   16.918699 secs
/dev/bus/usb/001/026 test 13 --> 5 (Input/output error)
/dev/bus/usb/001/026 test 14 --> 22 (Invalid argument)
/dev/bus/usb/001/026 test 17 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 18 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 19 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 20 --> 110 (Connection timed out)
/dev/bus/usb/001/026 test 21 --> 22 (Invalid argument)

reel_board

$ sudo ./testusb -D /dev/bus/usb/001/028
./testusb: /dev/bus/usb/001/028 may see only control tests
/dev/bus/usb/001/028 test 0,    0.000012 secs
/dev/bus/usb/001/028 test 1,    3.348177 secs
/dev/bus/usb/001/028 test 2,    2.991246 secs
/dev/bus/usb/001/028 test 3,    3.299371 secs
/dev/bus/usb/001/028 test 4,    2.957255 secs
/dev/bus/usb/001/028 test 5,  100.615993 secs
/dev/bus/usb/001/028 test 6,   91.922076 secs
/dev/bus/usb/001/028 test 7,  101.164655 secs
/dev/bus/usb/001/028 test 8,   92.064393 secs
/dev/bus/usb/001/028 test 9,    2.012352 secs
/dev/bus/usb/001/028 test 10,    8.241837 secs
/dev/bus/usb/001/028 test 11,   16.969417 secs
/dev/bus/usb/001/028 test 12,   16.606636 secs
/dev/bus/usb/001/028 test 13 --> 22 (Invalid argument)
```

So it appears that at least 1 test is not working whatever the USB driver:
`/dev/bus/usb/001/028 test 13 --> 22 (Invalid argument)`
According to dmesg:
disco_l475_iot1
```
[48288.082259] usbtest 1-4.2.2:1.0: TEST 13:  set/clear 1000 halts
[48298.265505] usb 1-4.2.2: verify_not_halted failed, iterations left 0, status -110 (not 0)
[48298.265524] usbtest 1-4.2.2:1.0: halts failed, iterations left 997
```
frdm_k64f
```
[50279.767589] usbtest 1-4.2.3:1.0: TEST 13:  set/clear 1000 halts
[50279.767843] usbtest 1-4.2.3:1.0: ep 81 couldn't get no-halt status, -5
[50279.767855] usbtest 1-4.2.3:1.0: halts failed, iterations left 999
```
reel_board:
```
[51054.501486] usbtest 1-4.2.2:1.0: TEST 13:  set/clear 1000 halts
[51065.618531] usb 1-4.2.2: verify_not_halted failed, iterations left 0, status -110 (not 0)
[51065.618547] usbtest 1-4.2.2:1.0: halts failed, iterations left 927
```

So I think this whole issue would deserve an initial comparative evaluation, test per test. Likely this will issue several tickets in various drivers.
@jfischer-phytec-iot do you have preference on how you want to deal with this?

jfischer-phytec-iot, 2020-09-21:
> So it appears that at least 1 test is not working whatever the USB driver:
> `/dev/bus/usb/001/028 test 13 --> 22 (Invalid argument)`

I can not confirm that. On the reel board all the tests pass (14, 21 needs additional parameter, e.g. -v 64).
I see regressions on frdm_k64f, investigating it.

jfischer-phytec-iot, 2020-09-23:
The reason for the failures on frdm_k64f is described in #28634, I think it does not affect usb_dc_stm32 driver directly.

erwango, 2020-09-24:
Still in the characterization process of this issue. 

Test 24 is systematically failed and sometime get testusb crashing.

For the other reported faulty tests (1, 3, 4, 17, 21), I've run loop testing on nucleo_f476zg (closest I have from nucleo_f767zi: same USB hardware block, same clock config):
```
for i in {1..20}
do
   for j in 1 3 14 17 19 21
   do
       echo "./testusb  -t $j round $i"
       sudo ./testusb  -t $j -D /dev/bus/usb/001/042 -v 512 -c 100000
   done
done
```
All tests passed (20 x 100000 iterations) except 2 issues on test 3: `test 3 --> 110 (Connection timed out)`. Test log:
```
[143030.160090] usbtest 1-4.2.3:1.0: TEST 3:  write/512 0..1024 bytes 100000 times
[143047.512832] usb 1-4.2.3: test3 failed, iterations left 92380, status -110 (not 0)
```

jfischer-phytec-iot, 2020-09-24:
@erwango Please note that the test or the host may have its own issues. I have a setup that I regularly check with two boards, reel board and frdm_k64f (frdm_k64f with a modified driver that I do not want to upstream yet).

If a test fails one should check what happens, for example, nucleo_f413zh with latest master:

_Mostly I have Wireshark running, if the targen does not have an exclusive host controller, you can filter the target in wireshark: `usb.device_address == 29`_

Test 13, `testusb -t 13 -D /dev/bus/usb/001/029` fails with unhelpfully output
`/dev/bus/usb/001/029 test 13 --> 22 (Invalid argument)`

But dmesg output offers more information:
```
[523198.761525] usb 1-6: new full-speed USB device number 29 using xhci_hcd
[523198.911162] usb 1-6: New USB device found, idVendor=2fe3, idProduct=0009, bcdDevice= 2.04
[523198.911166] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[523198.911169] usb 1-6: Product: Zephyr testusb sample
[523198.911171] usb 1-6: Manufacturer: ZEPHYR
[523198.911172] usb 1-6: SerialNumber: 3630363830365111
[523198.913186] usbtest 1-6:1.0: Linux gadget zero
[523198.913192] usbtest 1-6:1.0: full-speed {control in/out bulk-in bulk-out} tests (+alt)
[523290.492158] usbtest 1-6:1.0: TEST 13:  set/clear 1000 halts
[523300.701536] usb 1-6: verify_not_halted failed, iterations left 0, status -110 (not 0)
[523300.701550] usbtest 1-6:1.0: halts failed, iterations left 997
```

Wireshark:
I can see the host requests Set Feature ENDPOINT HALT, verifies it by Get Status, and verifies it by a transfer (e.g. IN endpint (0x81) should fail), requests Clear Feature ENDPOINT HALT, verifies it by Get Status, and verifies it by a transfer (IN (0x81) should work). The host also verifies by Get Status and by an OUT transfer that the OUT endpoint (0x01) has not been affected . Then the section is repeated with exchanged endpoints. I see that the test is aborted quickly because halting the IN endpoint affects OUT endpoint or vice versa.
[image]

Only when you have more information, you can start looking for errors. The problem is also how the driver is. It should be rewritten for each controller type, based on the lowest level offered by STM32 HAL.

erwango, 2020-09-24:
> The problem is also how the driver is.

I don't claim the drive is bug free

> It should be rewritten for each controller type, 

ok

> based on the lowest level offered by STM32 HAL.

This is already the case

carlescufi, 2021-02-09:
@erwango there seems to be some consensus that the STM32 USB driver needs some serious rework. Is this something that could be done by you or other ST engineers?

erwango, 2021-02-09:
@carlescufi This is in my radar and action has been taken. This is taking more time than expected, but this should eventually be solved.

closed 2022-01-16 by github-actions[bot]