zephyrproject-rtos/zephyr #28343: Bluetooth peripheral sample auto disconnects "ST B_L4S5I_IOT01A Discovery kit"

parthitce, 2020-09-12:
**Describe the bug**
Bluetooth [Peripheral sample](https://docs.zephyrproject.org/latest/samples/bluetooth/peripheral/README.html) failed to work in [ST B_L4S5I_IOT01A Discovery kit](https://docs.zephyrproject.org/latest/boards/arm/b_l4s5i_iot01a/doc/index.html) board.

Disconnection with the host after,
`[00:00:29.020,000] <err> bt_hci_core: packets count mismatch`

**To Reproduce**
Steps to reproduce the behavior:

**Zephyr steps:**
1. `west build -b disco_l475_iot1 -s samples/bluetooth/peripheral`
2. `west flash`

**Linux bluetoothctl:**
```
[bluetooth]# scan on
Discovery started
[CHG] Controller AC:67:5D:3C:4F:AF Discovering: yes
[NEW] Device 50:5F:38:1B:FB:C5 Zephyr Peripheral Sample Long
[bluetooth]# scan off
Discovery stopped
[bluetooth]# trust 50:5F:38:1B:FB:C5 
[CHG] Device 50:5F:38:1B:FB:C5 Trusted: yes
Changing 50:5F:38:1B:FB:C5 trust succeeded
[bluetooth]# connect 50:5F:38:1B:FB:C5 
Attempting to connect to 50:5F:38:1B:FB:C5
[CHG] Device 50:5F:38:1B:FB:C5 Connected: yes
Connection successful
[CHG] Device 50:5F:38:1B:FB:C5 Connected: no
[DEL] Device 50:5F:38:1B:FB:C5 Zephyr Peripheral Sample Long
```

**Expected behavior**
Connection established and should be able to view the profiles and examine GATT attributes from bluetoothctl.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-2807-gbc8832f6c879  ***
[00:00:00.007,000] <inf> fs_nvs: 8 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fsBluetooth initialized
_nvs: alloc wra: 0, 7Advertising successfully started
90
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 90
[00:00:00.011,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
[00:00:00.028,000] <wrn> bt_hci_core: ECC HCI commands not available
[00:00:00.028,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.031,000] <inf> bt_hci_core: Identity: c6:e8:81:02:d1:58 (random)
[00:00:00.031,000] <inf> bt_hci_core: HCI: version 4.1 (0x07) revision 0x3107, manufacturer 0x0030
[00:00:00.031,000] <inf> bt_hci_core: LMP: version 4.1 (0x07) subver 0x0023
Connected
Disconnected (reason 0x08)
[00:00:29.020,000] <err> bt_hci_core: packets count mismatch
```

**Environment (please complete the following information):**
 - OS: Linux with bluez Version 5.54
 - bc8832f6c8792a541e3000fc42a914955beaf989

erwango, 2020-09-14:
@parthitce can you clarify board you're using ?

> Bluetooth Peripheral sample failed to work in ST B_L4S5I_IOT01A Discovery kit board.
> west build -b disco_l475_iot1 -s samples/bluetooth/peripheral

These aren't the same board. ST B_L4S5I_IOT01A is `b_l4s5i_iot01a`.

Btw, I've tested `samples/bluetooth/peripheral` successfully on both boards, with 2 mobile app (ST BLE profile and nRF Connect). Will try to do some more testing using bluetoothctl.

parthitce, 2020-09-14:
> @parthitce can you clarify board you're using ?

I wasn't aware there is different revisions available. Mine is `B-L475E-IOT01A1` (printed on the PCB)

erwango, 2020-09-14:
> I wasn't aware there is different revisions available. Mine is B-L475E-IOT01A1 (printed on the PCB)

Ok, so this is indeed disco_l475_iot1.
ST B_L4S5I_IOT01A is based on STM32L4S5I. I agree this could be confising.

erwango, 2020-09-14:
@parthitce I've been able to test succesfuly:

```
$ scan on
[NEW] Device 75:F8:5B:01:8C:6D 75-F8-5B-01-8C-6D
[NEW] Device 79:5F:5D:7F:4D:77 Zephyr Peripheral Sample Long
[NEW] Device 5B:36:9A:38:DF:CF 5B-36-9A-38-DF-CF
$ scan off
Discovery stopped
$ connect 79:5F:5D:7F:4D:77
Attempting to connect to 79:5F:5D:7F:4D:77
[CHG] Device 79:5F:5D:7F:4D:77 Connected: yes
[...]
[NEW] Characteristic
	/org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0023
	00002a38-0000-1000-8000-00805f9b34fb
	Body Sensor Location
[NEW] Characteristic
	/org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0025
	00002a39-0000-1000-8000-00805f9b34fb
	Heart Rate Control Point
[...]
[Zephyr Peripheral Sample Long Name]# menu gatt
[Zephyr Peripheral Sample Long Name]# select-attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0025
[Zephyr Peripheral Sample Long Name]# notify on
[CHG] Attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0020 Notifying: yes
Notify started
[CHG] Attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0020 Value:
  06 79                                            .y              
[CHG] Attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0020 Value:
  06 7a                                            .z              
[CHG] Attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0020 Value:
  06 7b                                            .{              
[CHG] Attribute /org/bluez/hci0/dev_79_5F_5D_7F_4D_77/service001f/char0020 Value:
  06 7c                                            .|              
```

On board:
```
*** Booting Zephyr OS build v2.4.0-rc1-76-geb2a70549739  ***                    
[00:00:00.007,000] <inf> fs_nvs: 4 Sectors of 4096 bytes                        
[00:00:00.007,000] <inf> fs_nvs: aBluetooth initialized                         
lloc wra: 0, ff0                                                                
[Advertising successfully started                                               
00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0                                  
[00:00:00.012,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)    
[00:00:00.029,000] <wrn> bt_hci_core: ECC HCI commands not available            
[00:00:00.029,000] <inf> bt_hci_core: No ID address. App must call settings_loa)
[00:00:00.035,000] <inf> bt_hci_core: Identity: fb:d3:b5:8a:3d:8a (random)      
[00:00:00.035,000] <inf> bt_hci_core: HCI: version 4.1 (0x07) revision 0x3107, 0
[00:00:00.035,000] <inf> bt_hci_core: LMP: version 4.1 (0x07) subver 0x0023     
[00:00:00.035,000] <inf> bt_settings: Saving ID                                                                                
Connected                                                                       
[00:04:26.023,000] <inf> hrs: HRS notifications enabled                                                                  
```

Environment (please complete the following information):

OS: Linux with bluez Version 5.48
Zephyr: v2.4.0-rc1-76-geb2a705497

parthitce, 2020-09-14:
> @parthitce I've been able to test succesfuly:

@erwango Thank you very much for the tests. I will re-do the tests today and update my results. I was sure that the problem was persistent with constant disconnection and device deletion in bluetoothctl end.

Also there was another test which was failing, nrf52840dk acting as central and disco acting as peripheral. Will confirm my tests again. Thanks again.

parthitce, 2020-09-18:
@erwango It's working fine now. I couldn't reproduce the problem anymore. Sorry for the noise, closing it.

closed 2020-09-18 by parthitce