zephyrproject-rtos/zephyr #28363: ssd16xx: off-by-one with non-multiple-of-eight heights

weinholtendian, 2020-09-14:
**Describe the bug**
I'm using the ssd16xx driver with a 122x250 pixel display with all settings in OTP. The image is rendered with an offset, such that the top of the screen is some uninitialized data and the rest of the image is shifted downwards.

**To Reproduce**
I'm not sure how to reproduce without an actual 122x250 pixel display. One could use `west build -b nrf52840dk_nrf52840 samples/display/lvgl -- -DSHIELD=waveshare_epaper_gdeh0213b72`, but the display uses a different pinout.

**Expected behavior**
The whole 122x250 area of the display should be usable and rendered without offsets.

**Impact**
Annoyance; I have implemented a workaround.

**Additional context**
My workaround is to set the display to 128x250 in the device tree and change one line in the driver:

```diff
        case SSD16XX_DATA_ENTRY_XDYIY:
-               x_start = (panel_h - 1 - y) / SSD16XX_PIXELS_PER_BYTE;
+               x_start = (panel_h - 1 - y + (SSD16XX_PIXELS_PER_BYTE - 1)) /
+                       SSD16XX_PIXELS_PER_BYTE;
                x_end = (panel_h - 1 - (y + desc->height - 1)) /
```

The intent is to have the division round upwards. I didn't do the analysis to see if this is the correct thing to do, but the image is no longer shifted. The image shifting appears to happen because `x_start` is off-by-one, so it starts writing into RAM at one byte off from where it should.

There are a few places in the driver that takes the panel height and rounds it *down* to a multiple of eight:

```c
        uint16_t panel_h = EPD_PANEL_HEIGHT -
                        EPD_PANEL_HEIGHT % EPD_PANEL_NUMOF_ROWS_PER_PAGE;
```

I tried fixing it to round upwards instead, since I can get a good image with 128 for `EPD_PANEL_HEIGHT`, but the obvious thing didn't work and I ran out of time. With the above and a 122x250 pixel display, `panel_h = 122  - 122 % 8 = 120`, so what about the two missing lines? (It could very well be that the image is shifted down six pixels, it looks that way, but I didn't measure it).

jfischer-phytec-iot, 2020-09-14:
> The whole 122x250 area of the display should be usable and rendered without offsets.

No, that is not always possible because 122 is not multiple of 8

I see that I forgot to update shields properties like in 14ef0a7ccb0a593c1de54be96b920bfdee0b4e14. I will open a PR.

weinholtendian, 2020-09-14:
> No, that is not always possible because 122 is not multiple of 8.

Hmm, I must be misunderstanding something. I'm half sure that I do get 122 pixels in height on my display and all lines are rendered now, thanks to the workaround. If I set the height to 120 then the two last lines are blank.

I have some reference code from the manufacturer and that sends 250*128 bytes. Isn't it just the case that you should send 128 lines and the display itself ignores the six extra bits per line?

jfischer-phytec-iot, 2020-09-14:
> > No, that is not always possible because 122 is not multiple of 8.
> 
> Hmm, I must be misunderstanding something. I'm half sure that I do get 122 pixels in height on my display and all lines are rendered now, thanks to the workaround. If I set the height to 120 then the two last lines are blank.
> 

Correct, two line (first or last, depending on configuration) are unused. 

> I have some reference code from the manufacturer and that sends 250*128 bytes. Isn't it just the case that you should send 128 lines and the display itself ignores the six extra bits?

You are free to use the code from the manufacturer :-). Depending on scanning configuration the last six lines are not displayed, or worst case, the first six lines. I have decided that it is better to leave the two lines unused than to pretend to have a higher resolution that cannot be displayed at all.

weinholtendian, 2020-09-15:
(In the previous comment I wrote "250\*128 bytes", but I meant to write 250\*16 bytes or 250\*128 bits).

> Depending on scanning configuration the last six lines are not displayed, or worst case, the first six lines. I have decided that it is better to leave the two lines unused than to pretend to have a higher resolution that cannot be displayed at all.

Okay, I see what you mean, that makes sense if it can't really show all 122 lines.

I will take some time during the week to prove whether or not the display has the ability to display all 122 lines or not. Hopefully I will be back with some photos to demonstrate that it's possible. :)

weinholtendian, 2020-09-15:
I used this test pattern: 
[image]

The driver can render it either with the image shifted down:

[image]

Or with the image shifted up:

[image]

When the image is shifted down it uses garbage data from the display's RAM for the first two lines. If the display is not power cycled then it will keep the old image in RAM. The following shows what happens if the image is rendered so it's shifted up, and then rendered so it's shifted down, without power cycling the display between renderings:

[image]

I believe that the last image shows that it's possible to use all lines on the display, but that the driver is doing something that prevents that from working properly. It might be that the pixel format is wrong, maybe every byte should be shifted two bits? I'm not sure, and I don't have time to investigate this further.

If this is not convincing enough that the display can use all 122 lines then I'm out of ideas, so feel free to close the issue. Could very well be that it doesn't work as advertised. It works well enough for my current application.

closed 2020-11-29 by github-actions[bot]