zephyrproject-rtos/zephyr #28371: gcc 10.x compile warning/error for array subscript 0 is outside the bounds in subsys/bluetooth/mesh/prov.c

galak, 2020-09-14:
When building the following test with gcc 10.2 we get:

`tests/bluetooth/mesh_shell/bluetooth.mesh.mesh_shell`

```
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c: In function 'prov_link_opened':
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:267:46: error: array subscript 0 is outside the bounds of an interior zero-length array 'struct provisioner_link[0]' [-Werror=zero-length-bounds]
  267 |  net_buf_simple_add_u8(&inv, link.provisioner->attention_duration);
      |                              ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:102:26: note: while referencing 'provisioner'
  102 |  struct provisioner_link provisioner[PROVISIONER_LINK];
      |                          ^~~~~~~~~~~ /home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:126:25: note: defined here 'link'
  126 | static struct prov_link link;
      |                         ^~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:269:40: error: array subscript 0 is outside the bounds of an interior zero-length array 'struct provisioner_link[0]' [-Werror=zero-length-bounds]
  269 |  link.conf_inputs[0] = link.provisioner->attention_duration;
      |                        ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:102:26: note: while referencing 'provisioner'
  102 |  struct provisioner_link provisioner[PROVISIONER_LINK];
      |                          ^~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:126:25: note: defined here 'link'
  126 | static struct prov_link link;
      |                         ^~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c: In function 'bt_mesh_pb_adv_open':
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:1150:18: error: array subscript 0 is outside the bounds of an interior zero-length array 'struct provisioner_link[0]' [-Werror=zero-length-bounds]
 1150 |  link.provisioner->addr = addr;
      |  ~~~~~~~~~~~~~~~~^~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:102:26: note: while referencing 'provisioner'
  102 |  struct provisioner_link provisioner[PROVISIONER_LINK];
      |                          ^~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:126:25: note: defined here 'link'
  126 | static struct prov_link link;
      |                         ^~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:1151:18: error: array subscript 0 is outside the bounds of an interior zero-length array 'struct provisioner_link[0]' [-Werror=zero-length-bounds]
 1151 |  link.provisioner->net_idx = net_idx;
      |  ~~~~~~~~~~~~~~~~^~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:102:26: note: while referencing 'provisioner'
  102 |  struct provisioner_link provisioner[PROVISIONER_LINK];
      |                          ^~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:126:25: note: defined here 'link'
  126 | static struct prov_link link;
      |                         ^~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:1152:18: error: array subscript 0 is outside the bounds of an interior zero-length array 'struct provisioner_link[0]' [-Werror=zero-length-bounds]
 1152 |  link.provisioner->attention_duration = attention_duration;
      |  ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:102:26: note: while referencing 'provisioner'
  102 |  struct provisioner_link provisioner[PROVISIONER_LINK];
      |                          ^~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/mesh/prov.c:126:25: note: defined here 'link'
  126 | static struct prov_link link;
      |                         ^~~~
cc1: all warnings being treated as errors
gmake[2]: *** [zephyr/subsys/bluetooth/mesh/CMakeFiles/subsys__bluetooth__mesh.dir/build.make:239: zephyr/subsys/bluetooth/mesh/CMakeFiles/subsys__bluetooth__mesh.dir/prov.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3254: zephyr/subsys/bluetooth/mesh/CMakeFiles/subsys__bluetooth__mesh.dir/all] Error 2
gmake: *** [Makefile:104: all] Error 2

```

galak, 2020-09-14:
@carlescufi @jhedberg not sure what we want to do here.

jhedberg, 2020-09-14:
@tsvehagen FYI

jhedberg, 2020-09-14:
I suspect "brute force" solution would be to convert `IS_ENABLED(CONFIG_BT_MESH_PROVISIONER)` usage to `#ifdefs`, though this whole trickery with a zero-length array was supposed to allow the usage of `IS_ENABLED` to begin with, I believe.

trond-snekvik, 2020-09-15:
There aren't _that_ many places we'd have to add #ifdefs to to get around this. I'll make a PR, so we can get this fixed for 2.4.

PR #28388 (2020-09-17): Bluetooth: Mesh: Use #ifdef for provisioner build
Replaces the current zero length array for provisioner data with
ifdefs, to silence new gcc 10 warnings about array element access.

Fixes #28371.
commit d7e0d813 (2020-09-17): Bluetooth: Mesh: Use #ifdef for provisioner build

Replaces the current zero length array for provisioner data with
ifdefs, to silence new gcc 10 warnings about array element access.

Fixes #28371.
closed 2020-09-17 by jhedberg