zephyrproject-rtos/zephyr #28383: bq274xx sample is not working

jrananth1032, 2020-09-15:
**Describe the bug**
I am trying to get bq27421 up and running on our developement board(soc:nrf52840).But sample is not working, because in sample trying to fetch samples for all channels(SENSOR_CHAN_ALL) ,but driver does not handling SENSOR_CHAN_ALL
**To Reproduce**
Steps to reproduce the behavior:
1. west build 
2. Opened minicom
4. Seeing values as zero for all the channels

**Environment (please complete the following information):**
 - OS: Linux
 - gcc-arm-none-eabi-9-2020-q2-update
 - revision: 0742a45ac9f51d45f7b1fe148ac7d9bbfecb2f83

MaureenHelm, 2020-09-15:
@parthitce will you please take a look?

jrananth1032, 2020-09-15:
It is not a big issue , just need to read each channel one by one otherwise ,SENSOR_CHAN_ALL should be handled in driver.

parthitce, 2020-09-15:
> @parthitce will you please take a look?

@MaureenHelm Sure I will.

parthitce, 2020-11-18:
PR [28429](https://github.com/zephyrproject-rtos/zephyr/pull/28429) fixed this issue. Closing this issue @jrananth1032. Please re-open if anything else needs to be added.

PR #28429 (2021-01-11): samples: sensor: bq274xx: Read Gauge Channel Individually.
Read Gauge Channel Individually.
Fixes #28383.
commit 4c0b3240 (2021-01-11): samples: sensor: bq274xx: Read Gauge Channel Individually

Read sensor channel individually.
Fixes #28383.
closed 2020-11-18 by parthitce