zephyrproject-rtos/zephyr #28384: Bluetooth: L2CAP: Bad CoC SDU segment handling

jori-nordic, 2020-09-15:
**Bug description**
This is a bug happening when handling a fragmented SDU, and when not using a custom allocator function/memory pool for CoC.

When l2cap_chan_send() is called with a netbuf allocated from a pool with its buffer size equal to the exact data payload length, l2cap_chan_send() will segment the SDU and create an additional 2-byte PDU (in order to send the sdu_len).

On the receiving side, if there is not a custom allocator function defined (.alloc callback in chan_ops):

- l2cap will not mark the first PDU as a segment in a longer SDU
- on the second PDU (SDU's last 2 bytes), l2cap (in the same function) will try to read the sdu_len field, which will be garbage (or zero), and will disconnect the l2cap CoC channel.

In this case I got garbage for the second sdu_len read in my private repo, but could only reproduce getting zero when modifying a Zephyr sample.

**To Reproduce**
Check out https://github.com/jori-nordic/zephyr/commits/sdu_len_rx_bug

- Run bluetooth central sample on one device (I used nRF52840 DK)
- Run bluetooth peripheral_hr on another device
- Check the peripheral logs for warnings with sdu_len=0. e.g.
`[00:00:16.904,479] [1;33m<wrn> bt_l2cap: chan 0x20000890 len 15 sdu_len 0[0m`

To trigger the bug in another way, set TRIGGER_SDU_BUG_AGAIN to 0 and TRIGGER_SDU_BUG to 1 in samples/bluetooth/central/src/main.c 

**Expected behavior**
Correct fragmented SDU handling, i.e. not treating the second PDU as its own SDU, and not re-reading the sdu_len field.

**Impact**
Annoyance.

**Logs and console output**
[central.log.1.txt](https://github.com/zephyrproject-rtos/zephyr/files/5222922/central.log.1.txt)
[periph.log.1.txt](https://github.com/zephyrproject-rtos/zephyr/files/5222924/periph.log.1.txt)

[central.log.txt](https://github.com/zephyrproject-rtos/zephyr/files/5222923/central.log.txt)
[periph.log.txt](https://github.com/zephyrproject-rtos/zephyr/files/5222926/periph.log.txt)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Parent commit SHA: c64e0ce923cde54415d203e447a1dc62894ed2b0

Vudentz, 2020-09-17:
Weird, why is there are no logs of l2cap_chan_le_recv or l2cap_chan_le_recv_seg?

Vudentz, 2020-09-17:
When testing with shell everything looks fine:

Peripheral: https://gist.github.com/Vudentz/4ddb92569e80c60773a362f5767e1ecd

```
[00:02:56.470,000] <dbg> bt_l2cap.l2cap_chan_le_recv: chan 0x00270724 len 63 sdu_len 221
[00:02:56.470,000] <dbg> bt_l2cap.l2cap_chan_le_recv_seg: chan 0x00270724 seg 1 len 63
[00:02:56.480,000] <dbg> bt_l2cap.bt_l2cap_recv: Packet for CID 64 len 65
[00:02:56.480,000] <dbg> bt_l2cap.l2cap_rx_process: ch 0x00270724 buf 0x0026da20
[00:02:56.480,000] <dbg> bt_l2cap.l2cap_chan_le_recv_seg: chan 0x00270724 seg 2 len 65
[00:02:56.510,000] <dbg> bt_l2cap.bt_l2cap_recv: Packet for CID 64 len 65
[00:02:56.510,000] <dbg> bt_l2cap.l2cap_rx_process: ch 0x00270724 buf 0x0026da20
[00:02:56.510,000] <dbg> bt_l2cap.l2cap_chan_le_recv_seg: chan 0x00270724 seg 3 len 65
```

I will be trying to test with arbitrary data but look like there is some corruption going on with .sdu/.sdu_len, perhaps some of the memset endup clearing while receiving.

Vudentz, 2020-09-17:
Actually it is misconfiguration problem, you cannot disable alloc_buf if the MTU is bigger than the MPS, so I will likely have to add an assert when that happens or have the application reassemble but that is kind pointless since it does indeed has a buffer to reassemble the segments it should be able to support alloc_buf in the first place.

jori-nordic, 2020-09-18:
All right, thanks for the extra documentation, it was not clear before. 
I thought in the second example (central.1.log/peripheral.1.log) we did not exceed the MPS as the payload is 27 bytes and still managed to segment the SDU on the tx side, and failed to reassemble it on the RX (due to the missing buffer as you said).
I know it's an unusual use case, but wouldn't this still trigger the bug, even after #28488 ?

Vudentz, 2020-09-18:
We are going to truncate the MTU with #28488 so you would not be able to send any packet that would require segmentation.

jori-nordic, 2020-09-21:
In the second example, we have an MTU of 31, MPS of 33:
`[00:01:32.516,052] [0m<dbg> bt_l2cap.le_conn_req: psm 0x80 scid 0x0040 mtu 31 mps 33 credits 2[0m`

But we still truncate on the TX side, because the current behavior right now when sending is to allocate from the same buffer pool as the buffer that is passed to bt_l2cap_chan_send(). 
The problem is that buffer pool is misconfigured (in order to trigger segmentation) to have exactly the same size as the data payload passed to bt_l2cap_chan_send().

So L2CAP on the TX side:
- checks if it can allocate more bytes for the header from the same netbuf, but cannot
- checks if more buffers are available in the same pool, and sees that there are
- segments the SDU into 2 PDUs: one 27-bytes with the 2-byte length field, one with 2 bytes for the remaining 2 bytes of payload data

On the RX side, since there is no allocation function, the reassembly fails.

I am not sure if that is a behavior that can happen with another vendor's stack, or ok spec-wise (fragmenting with the payload length < MPS), but that was what I wanted to test on the receiving side.

In any case, I think that your changes are enough, since it's now documented that we need an alloc function to support fragmentation.

joerchan, 2020-09-21:
> On the RX side, since there is no allocation function, the reassembly fails.

@jori-nordic With the suggested fix it should not be possible to establish a channel with mps/mtu that would result in segmentation.
Did you try the fix?

jori-nordic, 2020-09-21:
@joerchan I just tried, it does disallow sending a segmented packet when payload > MPS, but the situation I described in my previous comment still happens (see attached logs). 
I was just wondering if that would be possible, that a peer might fragment packets even if the actual data size is < MPS.

[central.log](https://github.com/zephyrproject-rtos/zephyr/files/5255710/central.log)
[periph.log](https://github.com/zephyrproject-rtos/zephyr/files/5255711/periph.log)

PR #28488 (2020-09-22): Bluetooth: L2CAP: Document behavior of alloc_buf
Channel that requires segmentation must set alloc_buf in order for the
stack to be able to reassemble segments of an SDU.

If this happens at runtime warn the user and truncate the MTU to
force segmentation to be disabled.

Fixed #28384
closed 2020-09-22 by jhedberg