zephyrproject-rtos/zephyr #28385: drivers.clock.nrf_lf_clock_start_xtal_no_wait.wait_in_thread fails on nrf9160dk_nrf9160

PerMac, 2020-09-15:
**Describe the bug**
The test case `drivers.clock.nrf_lf_clock_start_xtal_no_wait.wait_in_thread fails on nrf9160dk_nrf9160` fails on nrf9160

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. have nrf9160 connected (here at ttyACM0)
3. run `./scripts/sanitycheck --device-testing -T tests/drivers/clock_control/nrf_lf_clock_start/ -p nrf9160dk_nrf9160 --device-serial /dev/ttyACM0 -v -v`
4. See error

**Expected behavior**
The test passes

**Impact**
NA

**Logs and console output**
```
<failure message="failed" type="failure">
CLOCK_CONTROL_NRF_K32SRC=XTAL SYSTEM_CLOCK_NO_WAIT=y SYSTEM_CLOCK_WAIT_FOR_AVAILABILITY=n SYSTEM_CLOCK_WAIT_FOR_STABILITY=n Running test suite test_nrf_lf_clock_start =================================================================== START - test_clock_check PASS - test_clock_check =================================================================== START - test_wait_in_thread Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/clock_control/nrf_lf_clock_start/src/main.c:81: test_wait_in_thread: (t == NRF_CLOCK_LFCLK_Xtal) && o is true FAIL - test_wait_in_thread =================================================================== Test suite test_nrf_lf_clock_start failed. =================================================================== PROJECT EXECUTION FAILED
</failure>
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK 0.11.3
 - Commit SHA 6c53e93318

carlescufi, 2020-09-16:
@ioannisg and @nordic-krch will you fix this for 2.4?

nordic-krch, 2020-09-17:
@carlescufi yes, i will look into this

PR #28497 (2020-09-22): drivers: clock_control: nrf: Fix waiting for LF clock
Algorithm for waiting for clock stabilization was failing in case
when it was waiting for clock availablity and clock was already
available before function was called. That is because nrfx_clock_is_running
was returning false because XTAL was already started but not yet
running.

Added a check for current LF source, if XTAL is picked that indicates
that RC is already running because of two stage startup procedure.

Fixes #28385.
closed 2020-09-22 by carlescufi