zephyrproject-rtos/zephyr #28390: drivers: sensor: hts221 and lsm6dsl compilation issue when sensor defined in board (I2C) and in test (SPI)

ABOSTM, 2020-09-15:
**Describe the bug**
While pushing PR #27013, Buildkite highlight a compilation issue in hts221 and lsm6dsl.

#27753 already fixes a similar issue with lis2dh (on stm32f3_disco).
Now it merged, and new issue raised up on disco_l475_iot1 with compilation of hts221 and lsm6dsl.
Both sensor are embedded in disco_l475_iot1 baord (on I2C), but ther is conflict with test which define the same sensor on SPI:
-> error: 'DT_N_S_soc_S_**i2c**_40005800_S_lsm6dsl_6a_P_**spi_max_frequency**

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b disco_l475_iot1 tests/drivers/build_all/ -- -DCONF_FILE=sensors_trigger_i_z.conf
2. See error

**Expected behavior**
Compilation successful

**Logs and console output**
FAILED: zephyr/drivers/sensor/lsm6dsl/CMakeFiles/drivers__sensor__lsm6dsl.dir/lsm6dsl_spi.c.obj
C:\gnuarmemb\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=v2.4.0-rc1-122-gb3ae394ac314 -DCORE_CM4 -DKERNEL -DSTM32L475xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/st_stm32/stm32l4 -I../drivers -I../soc/arm/st_stm32/common -I../subsys/testsuite/include -IC:/zephyr/zephyr_project/modules/hal/cmsis/CMSIS/Core/Include -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/iis3dhhc_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/lis2mdl_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/lsm6dso_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l4xx/soc -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -isystem ../lib/libc/minimal/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -imacros C:/zephyr/zephyr_project/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros C:/zephyr/zephyr_project/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr/tests/drivers/build_all=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/zephyr/zephyr_project=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/sensor/lsm6dsl/CMakeFiles/drivers__sensor__lsm6dsl.dir/lsm6dsl_spi.c.obj -MF zephyr\drivers\sensor\lsm6dsl\CMakeFiles\drivers__sensor__lsm6dsl.dir\lsm6dsl_spi.c.obj.d -o zephyr/drivers/sensor/lsm6dsl/CMakeFiles/drivers__sensor__lsm6dsl.dir/lsm6dsl_spi.c.obj   -c C:/zephyr/zephyr_project/zephyr/drivers/sensor/lsm6dsl/lsm6dsl_spi.c
In file included from ../soc/arm/st_stm32/stm32l4/soc.h:30,
                 from ../include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from ../include/arch/arm/aarch32/syscall.h:29,
                 from ../include/arch/syscall.h:19,
                 from ../include/syscall.h:12,
                 from ../include/kernel_includes.h:36,
                 from ../include/kernel.h:17,
                 from ../include/init.h:11,
                 from ../include/device.h:22,
                 from ../include/drivers/spi.h:24,
                 from C:/zephyr/zephyr_project/zephyr/drivers/sensor/lsm6dsl/lsm6dsl_spi.c:13:
zephyr/include/generated/devicetree_unfixed.h:3865:32: error: 'DT_N_S_soc_S_i2c_40005800_S_lsm6dsl_6a_P_spi_max_frequency' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40003c00_S_ism43362_1_P_spi_max_frequency'?
 3865 | #define DT_N_INST_0_st_lsm6dsl DT_N_S_soc_S_i2c_40005800_S_lsm6dsl_6a
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/devicetree.h:1911:38: note: in definition of macro 'DT_CAT'
 1911 | #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
      |                                      ^~~~~~~
../include/devicetree.h:1448:34: note: in expansion of macro 'DT_PROP'
 1448 | #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
      |                                  ^~~~~~~
../include/sys/util_internal.h:106:36: note: in expansion of macro 'DT_N_INST_0_st_lsm6dsl'
  106 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
../include/sys/util_internal.h:105:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  105 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
../include/devicetree.h:297:31: note: in expansion of macro 'UTIL_CAT'
  297 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
../include/devicetree.h:1426:27: note: in expansion of macro 'DT_INST'
 1426 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
../include/devicetree.h:1448:42: note: in expansion of macro 'DT_DRV_INST'
 1448 | #define DT_INST_PROP(inst, prop) DT_PROP(DT_DRV_INST(inst), prop)
      |                                          ^~~~~~~~~~~
C:/zephyr/zephyr_project/zephyr/drivers/sensor/lsm6dsl/lsm6dsl_spi.c:30:15: note: in expansion of macro 'DT_INST_PROP'
   30 |  .frequency = DT_INST_PROP(0, spi_max_frequency),
      |               ^~~~~~~~~~~~

**Environment (please complete the following information):**
 - OS:  Linux and Windows
 - Toolchain Zephyr SDK
 - Commit SHA: 6c53e9331854a3485112685c47b77852f3576224

avisconti, 2020-09-16:
OK, I'm scheduling it for the next jira sprint (our internal sprint).
Thanks for reporting it!

avisconti, 2020-09-16:
@ABOSTM 
It seems to me that this bug is only related to lsm6dsl, isn't it?

ABOSTM, 2020-09-16:
@avisconti ,
Looking at buildkite details of #27013, there is also an error for hts221.
I didn't look carefully the 1st time, and I think error was the same than lis2dh and lsm6dsl.
But in fact error is different for hts221. I submitted a new issue #28443
And I updated the title and description of this issue.

PR #28514 (2020-09-19): drivers: sensor: lsm6dsl: make the driver multi-instance
Make this driver multi-instance and use the new API.
Fixes #28390

Tested this commit on nucleo_f401re in two different ways:

1. Using x_nucleo_iks01a3 shield and DIL24 configured in SPI mode with triggers enabled
2. Using x_nucleo_iks01a2 shield and DIL24 configured in I2C mode with triggers enabled.
    In this second case, as a LSM6DSL device is already present on the shield, we have effectively
    two instance working at the same time.
commit ce0cc3a7 (2020-09-19): drivers: sensor: lsm6dsl: make the driver multi-instance

Make this driver multi-instance and use the new API.
Fixes #28390
closed 2020-09-19 by nashif