zephyrproject-rtos/zephyr #28394: nanosleep test failed on ARC series targets

YuguoWH, 2020-09-15:
**Describe the bug**
tests/posix/common test_nanosecond_0_1 failed on ARC series targets including iotdk, emsdp, nsim_em, nsim_hs, etc. 

**To Reproduce**
Steps to reproduce the behavior:
scripts/sanitycheck -p nsim_em -T tests/posix/common

**Expected behavior**
All tests should pass

**Logs and console output**

```
START - test_nanosleep_0_1

    Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:214: common: ((delta_ns < 0) ? ((uint64_t)-delta_ns <= tolerance_ns) : ((uint64_t)delta_ns <= tolerance_ns) is false)
error -52799 beyond tolerance 200 for 1 vs 52800
 FAIL - test_nanosleep_0_1
```

**Additional context**
Related to #27065

ruuddw, 2020-09-15:
@pabigot  @carlescufi
The problem doesn't seem to be related to simulation instruction count vs timer count as mentioned in the PR, but to the fact that a 1 nanosec delay is expected to not take more than 1 cycle (which is hard on targets running at few Mhz and not GHz).
       uint64_t tolerance_ns = MAX(NSEC_PER_CYCLE,  (TOLERANCE_PPC * exp_ns) / 100U);

pabigot, 2020-09-15:
I'm not convinced by that diagnosis.  AFAICT POSIX nanosleep on Zephyr uses `k_busy_wait()` which is not bound to ticks (or isn't supposed to be, at least), but has a resolution of 1 us.   The implementation is probably wrong, since it delays for the floor of ns/1000 rather than the ceiling, so the test for 1 ns is effectively `k_busy_wait(0)`.

The configured cycle rate in the application is 5 MHz, so a 1 cycle sleep would be 200 ns as the test expects.  The output above shows that the cycle count advanced 52800 for a 1 ns sleep.  That would imply a 52.8 THz cycle clock if a 1 ns sleep occurred; I have no idea how that number comes out of `k_busy_wait(0)`.

There's obviously something wrong here, but it's not obvious what.  Somebody who has hardware available should pursue this; when I run the reproducing command above it passes (it also doesn't execute the test since there's no device testing enabled).

ruuddw, 2020-09-15:
No, it is not 52800 cycles that the busy wait call took, it is 52800 nsec = 264 cycles @5Mhz (which is a realistic minimum overhead for a function call that checks a timer and concludes 1 nsec has passed, see the default implementation).
The assert, however, expects 0 or 1 cycle.

```
       uint64_t exp_ns = s * NSEC_PER_SEC + ns;                                         // exp_ns=1
       uint64_t tolerance_ns = MAX(NSEC_PER_CYCLE,                                // tolerance_ns=NSEC_PER_CYCLE, for nsim_em NSEC_PER_CYCLE=1e9/5e6=200
                                (TOLERANCE_PPC * exp_ns) / 100U);
       uint64_t tck_ns = k_cyc_to_ns_ceil64((now == then)                           // tck_ns=(now-then)*NSEC_PER_CYCLE
                                        ? 1U
                                        : (now - then));
       int64_t delta_ns = (int64_t)(exp_ns - tck_ns);

       zassert_true((delta_ns < 0)                                                                 // assert       abs(tck_ns-exp_ns) <= tolerance_ns
                   ? ((uint64_t)-delta_ns <= tolerance_ns)                                 // equals to  abs( (now-then)*NSEC_PER_CYCLE -1) <= NSEC_PER_CYCLE
                   : ((uint64_t)delta_ns <= tolerance_ns),                                   // or             abs(cycles* NSEC_PER_CYCLE -1) <= NSEC_PER_CYCLE
              "error %lld beyond tolerance %llu for %llu vs %llu",                       // only true when cycles=0, 1
                   delta_ns, tolerance_ns, exp_ns, tck_ns);

```

pabigot, 2020-09-15:
@ruuddw Thanks for explaining; that does make sense.

I'm going to solve this with #28400 by removing the test, which was wrong to start.  When I reworked it to support tolerance I unwittingly (witlessly) removed a comment:
```
               /* k_cycle_get_32() does not seem to be completely accurate on
                * some virtual platforms in CI (some function calls to
                * nanosleep reportedly take 0ns).
                */
```

which should have been a clue that the implementation was wrong due to inappropriate truncation.  Also tests will never be within a proportional tolerance for values that are not exactly represented in microseconds.

@cfriedt if you'd like to propose an updated test, along with a fix for nanosleep to round up rather than down, that'd be fine.  At this stage in the release cycle I'm going with tests that pass with the existing behavior.

ABOSTM, 2020-09-15:
I face the same issue on all STM32 boards (disco_l475_iot1, nucleo_f207zg, nucleo_f429zi, nucleo_f746zg, nucleo_g474re, nucleo_h743zi, nucleo_l152re, nucleo_l4r5zi, nucleo_wb55rg ).
Regression introduced by #27065 tests: posix: fix nanosleep tests for clock slew 
```
START - test_nanosleep_0_1

    Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:210: common: ((delta_ns < 0) ? ((uint64_t)-delta_ns <= tolerance_ns) : ((uint64_t)delta_ns <= tolerance_ns) is false)
error -24218 beyond tolerance 31 for 1 vs 24219
 FAIL - test_nanosleep_0_1
```

pabigot, 2020-09-15:
Medium since affects multiple platforms; PR scheduled for 2.4.0

PR #28400 (2020-09-16): tests: posix: common: remove sub-microsecond nanosleep test
Beyond the complexities of tick resolution on QEMU, this test is invalid because POSIX nanosleep as currently implemented uses k_busy_wait(ns / 1000) which means it's measuring the duration of k_busy_wait(0) in cycles, which has no reasonable relation to 1 ns regardless of tolerance.

Fixes #28394.
closed 2020-09-16 by carlescufi