zephyrproject-rtos/zephyr #28411: [Coverity CID :214281] Unchecked return value in subsys/mgmt/osdp/src/osdp.c

zephyrbot, 2020-09-15:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/6fe1fde9dcb11cda20644142e71eebbc5fc6c24b/subsys/mgmt/osdp/src/osdp.c#L194

Category: Error handling issues
Function: `osdp_init`
Component: Other
CID: [214281](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214281)

Details:

```
188         uart_irq_rx_disable(p->dev);
189         uart_irq_tx_disable(p->dev);
190         uart_irq_callback_user_data_set(p->dev, osdp_uart_isr, p);
191    
192         /* Drain UART fifo */
193         while (uart_irq_rx_ready(p->dev)) {
>>>     CID 214281:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "uart_fifo_read" without checking return value (as is done elsewhere 14 out of 17 times).
194             uart_fifo_read(p->dev, &c, 1);
195         }
196    
197         /* Both TX and RX are interrupt driven */
198         uart_irq_rx_enable(p->dev);
199    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cbsiddharth, 2020-09-16:
This is a false positive. My coverity access has some issue, can someone add the following comment and close this issue?

```
uart_fifo_read() is called to flush the FIFO. As data from the FIFO is discarded, a check is not needed.
```

Thanks.

de-nordic, 2020-09-16:
@cbsiddharth I have marked it as requested.

closed 2020-09-16 by de-nordic