zephyrproject-rtos/zephyr #28412: [Coverity CID :214279] 'Constant' variable guards dead code in tests/drivers/clock_control/nrf_lf_clock_start/src/main.c

zephyrbot, 2020-09-15:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/6fe1fde9dcb11cda20644142e71eebbc5fc6c24b/tests/drivers/clock_control/nrf_lf_clock_start/src/main.c#L61

Category: Possible Control flow issues
Function: `test_clock_check`
Component: Tests
CID: [214279](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214279)

Details:

```
55    
56         xtal = IS_ENABLED(CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL) |
57             IS_ENABLED(CONFIG_CLOCK_CONTROL_NRF_K32SRC_EXT_LOW_SWING) |
58             IS_ENABLED(CONFIG_CLOCK_CONTROL_NRF_K32SRC_EXT_FULL_SWING);
59    
60         if (xtal) {
>>>     CID 214279:  Possible Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "xtal_check(on, type);".
61             xtal_check(on, type);
62         } else if (IS_ENABLED(CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC)) {
63             rc_check(on, type);
64         } else {
65             synth_check(on, type);
66         }
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

maksimmasalski, 2020-09-17:
@anangl 
I'm just wondering, why it cannot reach the statement `xtal_check(on, type);` ?
Seems that if one of the configs is enabled, everything will be fine. Why Coverity classified that it is an issue?

anangl, 2020-09-21:
> @anangl
> I'm just wondering, why it cannot reach the statement `xtal_check(on, type);` ?
> Seems that if one of the configs is enabled, everything will be fine. Why Coverity classified that it is an issue?

Exactly. In some configurations this part of code will be executed, in others it is supposed to be optimized out as a dead code (similarly as the calls to `rc_check()` and `synth_check()` that follow).
I marked this as intentional and to be ignored.

closed 2020-09-21 by anangl