zephyrproject-rtos/zephyr #28414: kernel/timeout: next_timeout() is returning negative number of ticks

vanti, 2020-09-15:
**Describe the bug**
next_timeout() in kernel/timeout.c has the following line of code:

`int32_t ret = to == NULL ? MAX_WAIT : MAX(0, to->dticks - ticks_elapsed);`

The intent seems to be if the requested timeout has already elapsed and `to->dticks - ticks_elapsed` is negative, it should set `ret` to 0 (if `to` is not NULL).
However, this is not what happens in practice, given `to->dticks` is an unsigned number. MAX() would internally do an unsigned comparison, which means the result of `to->dticks - ticks_elapsed` would be returned, and casted to 'int32_t' in `ret`.

This leads to problems for example with calls like this:

`z_clock_set_timeout(next_timeout(), false);`

A negative number of ticks would be passed to the driver, which may or may not deal with it correctly. And if it so happen that `to->dticks - ticks_elapsed` is -1, an infinite/large timeout would be set given it corresponds to K_FOREVER.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/tests/kernel/fifo/fifo_timeout
2. west build -b cc1352r1_launchxl
3. west flash
4. Sometimes you would see the example hang/pause forever:

```
*** Booting Zephyr OS build v2.4.0-rc1-46-g840db3ab6ab6  ***
Running test suite test_fifo_timeout
===================================================================
START - test_timeout_empty_fifo
PASS - test_timeout_empty_fifo
===================================================================
START - test_timeout_non_empty_fifo
PASS - test_timeout_non_empty_fifo
===================================================================
START - test_timeout_fifo_thread
PASS - test_timeout_fifo_thread
===================================================================
START - test_timeout_threads_pend_on_fifo
thread (q order: 2, t/o: 0, fifo 0x20000978)
thread (q order: 3, t/o: 10, fifo 0x20000978)
thread (q order: 0, t/o: 20, fifo 0x20000978)
thread (q order: 4, t/o: 30, fifo 0x20000978)
thread (q order: 1, t/o: 40, fifo 0x20000978)
PASS - test_timeout_threads_pend_on_fifo
===================================================================
START - test_timeout_threads_pend_on_dual_fifos
thread (q order: 0, t/o: 0, fifo 0x2000098c)
```

**Expected behavior**
I'd expect next_timeout() to return a non-negative number.

**Impact**
Could cause infinite/large timeouts to be set at the driver level when a 0 timeout should be seen instead.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.11.3
 - Commit SHA or Version used: 840db3ab6ab60e8427fca9eb6b545610d6766fbf

pabigot, 2020-09-15:
`to->dticks` was an unsigned number, but this was reverted in #28034 to resolve #27874.

vanti, 2020-09-15:
Closing due to changes from #28034.

closed 2020-09-15 by vanti